genodelabs/genode #4077: platform_drv: enrich device XML info with physical resource names

skalk, 2021-04-16:
For driver ports like DDE Linux device drivers, it is more easy to relate physical resources given by the platform driver, e.g. with information provided by a device-tree-blob. Therefore, it would be nice to export these additional information via the devices XML info. For simple-and-clean, written-from-scratch drivers to not "infect" them with these physical, global names we can make the whole device info to be optional. Therefore, the client policy in the platfor_drv configuration has to include a `info=yes` attribute.

nfeske, 2021-04-20:
@skalk can you please point me to the most recent commit / branch to merge?

skalk, 2021-04-20:
@nfeske sorry, I mistakenly deleted the corresponding branch. I re-newed my merge_to_staging branch now. It is commit id 37b3035

skalk, 2021-04-21:
A problem arised: the recent addition of utilities for platform devices on ARM* (see issue 4075) uses inherently the device XML information, when using the `device_by_index()` utility in `Platform::Connection`. When making the device information optional this utility breaks. To workaround the problem I would extend this issue in removing `device_by_index()`, which is primarily only used with index 0, and add a `acquire_single_device` RPC to the platform API instead that delivers the first device for clients that use a single device anyway.

skalk, 2021-04-21:
@nfeske I've pushed a fixup commit 5a9a44d8d3e0786c8f4d5cc7d8414f1b8d3feb7a to my merge_to_staging branch, which fixes the Virtio NIC driver on x86, and implements above described API change.

nfeske, 2021-04-21:
Thanks a lot. Merged to staging.

commit 67b81c89 (2021-04-20): os: turn device XML info into an option

* The device XML information dataspace is only provided,
  when the client's policy states `info="yes"`
* The device XM information gets changed to include the
  physical resource names (I/O memory and IRQ addresses)
  instead of virtual ids and page offset

Fix #4077
commit 421d2bed (2021-05-05): platform driver: make device info XML optional

* The device XML information dataspace is only provided,
  when the client's policy states `info="yes"`
* The device XM information gets changed to include the
  physical resource names (I/O memory and IRQ addresses)
  instead of virtual ids and page offset

Fix #4077
closed 2021-05-05 by nfeske