genodelabs/genode #4078: qemu-usb: add usb webcam model

alex-ab, 2021-04-19:


alex-ab, 2021-04-23:
The issue_4078 contains the commit for the feature and is ready for review.

cnuke, 2021-04-26:
Thanks for the commits, LGTM - I added my minor remarks to the commit in question.

alex-ab, 2021-04-27:
> Thanks for the commits, LGTM - I added my minor remarks to the commit in question.

Thanks for review, I added the fixup e0a1e636d944467c9dacc2dd9818626559dc1d7c addressing your comments and removing some debug artefacts.

alex-ab, 2021-04-27:
> Thanks for review, I added the fixup [e0a1e63](https://github.com/genodelabs/genode/commit/e0a1e636d944467c9dacc2dd9818626559dc1d7c) addressing your comments and removing some debug artefacts.

Sorry, the commit message was wrong, dc81ec37ed167875ed0d6022fe2af9521d5c3a1c contains the right commit message.

chelmuth, 2021-04-30:
@alex-ab as @nfeske already mentioned to you the build with `-O0` fails. This is also true for `-O2 -fno-omit-frame-pointer` which I use in all tests. The register constraints just seem impossible to fulfill.
```
.../genode.git/contrib/libyuv-7af40ea6affe479d911ef879622b3358f2c4e490/libyuv/source/row_gcc.cc: In function ‘void libyuv::SplitRGBRow_SSSE3(const uint8_t*, uint8_t*, uint8_t*, uint8_t*, int)’:
.../genode.git/contrib/libyuv-7af40ea6affe479d911ef879622b3358f2c4e490/libyuv/source/row_gcc.cc:5129:48: error: ‘asm’ operand has impossible constraints
```

alex-ab, 2021-04-30:
@chelmuth, @nfeske: bb0be6c1b9ee93295fb4be69dbdba6c2b0ac1a73 fixes the compiler issues

chelmuth, 2021-04-30:
Great! Merged the fix to staging.

cproc, 2021-04-30:
I get compile errors on arm_v8a:

```
  Library libyuv                                                                                                                                             
    COMPILE  row_neon64.o                                                                                                                                    
    COMPILE  scale_neon64.o                                                                                                                                  
/tmp/ccf6RIF3.s: Assembler messages:                                                                                                                         
/tmp/ccf6RIF3.s:597: Error: operand mismatch -- `mov v0.8h,v4.8h'                                                                                            
/tmp/ccf6RIF3.s:597: Info:    did you mean this?                                                                                                             
/tmp/ccf6RIF3.s:597: Info:      mov v0.8b, v4.8b                                                                                                             
/tmp/ccf6RIF3.s:597: Info:    other valid variant(s):                                                                                                        
/tmp/ccf6RIF3.s:597: Info:      mov v0.16b, v4.16b                                                                                                           
/tmp/ccf6RIF3.s:598: Error: operand mismatch -- `mov v1.8h,v5.8h'                                                                                            
/tmp/ccf6RIF3.s:598: Info:    did you mean this?                                                                                                             
/tmp/ccf6RIF3.s:598: Info:      mov v1.8b, v5.8b                                                                                                             
/tmp/ccf6RIF3.s:598: Info:    other valid variant(s):                                                                                                        
/tmp/ccf6RIF3.s:598: Info:      mov v1.16b, v5.16b                                                                                                           
/tmp/ccf6RIF3.s:643: Error: operand mismatch -- `mov v2.8h,v0.8h'                                                                                            
/tmp/ccf6RIF3.s:643: Info:    did you mean this?                                                                                                             
/tmp/ccf6RIF3.s:643: Info:      mov v2.8b, v0.8b                                                                                                             
/tmp/ccf6RIF3.s:643: Info:    other valid variant(s):                                                                                                        
/tmp/ccf6RIF3.s:643: Info:      mov v2.16b, v0.16b                                                                                                           
/tmp/ccf6RIF3.s:694: Error: operand mismatch -- `mov v0.8h,v2.8h'
/tmp/ccf6RIF3.s:694: Info:    did you mean this?
/tmp/ccf6RIF3.s:694: Info:      mov v0.8b, v2.8b
/tmp/ccf6RIF3.s:694: Info:    other valid variant(s):
/tmp/ccf6RIF3.s:694: Info:      mov v0.16b, v2.16b
/tmp/ccf6RIF3.s:700: Error: operand mismatch -- `mov v0.8h,v4.8h'
/tmp/ccf6RIF3.s:700: Info:    did you mean this?
/tmp/ccf6RIF3.s:700: Info:      mov v0.8b, v4.8b
/tmp/ccf6RIF3.s:700: Info:    other valid variant(s):
/tmp/ccf6RIF3.s:700: Info:      mov v0.16b, v4.16b
/tmp/ccf6RIF3.s:703: Error: operand mismatch -- `mov v0.8h,v4.8h'
/tmp/ccf6RIF3.s:703: Info:    did you mean this?
/tmp/ccf6RIF3.s:703: Info:      mov v0.8b, v4.8b
/tmp/ccf6RIF3.s:703: Info:    other valid variant(s):
/tmp/ccf6RIF3.s:703: Info:      mov v0.16b, v4.16b
/tmp/ccf6RIF3.s:704: Error: operand mismatch -- `mov v1.8h,v5.8h'
/tmp/ccf6RIF3.s:704: Info:    did you mean this?
/tmp/ccf6RIF3.s:704: Info:      mov v1.8b, v5.8b
/tmp/ccf6RIF3.s:704: Info:    other valid variant(s):
/tmp/ccf6RIF3.s:704: Info:      mov v1.16b, v5.16b
...
```

alex-ab, 2021-04-30:
@cproc: right, thanks, 56a72f7955dcc48c1cd468fadd04d73b82aea8fb helps

alex-ab, 2021-04-30:
@cproc: or do you have a user of the lib on armv8 ?

cproc, 2021-04-30:
@alex-ab: no, I just noticed the error while testing the new tool chain (and then checked if it also happens with the old tool chain). Now I see this error on arm_v8a:

```
make[1]: *** No rule to make target 'libyuv.lib', needed by 'qemu-usb.lib'.  Stop.1584
```

alex-ab, 2021-04-30:
Oh je, I guest qemu-usb will require it ... right :-(

alex-ab, 2021-05-03:
@chelmuth, @nfeske: please add 56a72f7955dcc48c1cd468fadd04d73b82aea8fb and 55bf88111c1730da47ac2ac603e31ce2131c6269 to staging.

nfeske, 2021-05-03:
Merged both commits to staging.

alex-ab, 2021-05-04:
@nfeske: please consider f09939f0217436e0b1ed9015b7555e5f93f402867 for staging to fix vbox* package errors

chelmuth, 2021-05-04:
Merged to staging.

closed 2021-05-05 by nfeske

alex-ab, 2021-05-21:
@chelmuth: please consider 0ff24e22b45b3f50ac4c00026be91591e21ce0fb for staging

chelmuth, 2021-05-21:
Merged.

alex-ab, 2021-07-23:
@chelmuth: please consider 0b3d92ecb0d4685d72c954823101d77b6893ed4e for staging, it is required for the webcam appliance done by @jschlatow

commit 5de43653 (2021-04-28): qemu-usb: add usb webcam model

Fixes #4078
commit 0538763e (2021-06-16): qemu-usb: reduce cpu load by webcam model

if no new frame is available for capturing

Fixes #4078 #4196
commit 4aa99fd1 (2021-06-25): qemu-usb: reduce cpu load by webcam model

if no new frame is available for capturing

Fixes #4078 #4196