genodelabs/genode #4079: platform_driver: let device driver decide whether to use MSI-X or MSI

jschlatow, 2021-04-19:
Currently, the platform driver always enables MSI-X if available and does not let the device driver any choice in "downgrading" to MSI (or even IRQ). On my T490s, this led to the device driver to fail during firmware load. More precisely, the iwlwifi driver calls  `pci_enable_msix_range()`, which is implemented as a dummy in dde_linux and returning `-1`. As the driver expects to get the number of available interrupts/vectors (at least two), it falls back to msi using `pci_enable_msi()`. Yet, at this point, the platform driver had already decided to enable MSI-X and keep MSI disabled. Note, according to the spec, enabling both MSI and MSI-X leads to unspecified behaviour. Therefore, there is no way to allow this kind of dynamic fallback at the moment.

A quickfix is to add a `nomsix` option to the platform driver in addition to the already existing `nomsi`.

By the way, this should also be relevant for the xhci driver which calls the dummy function `pci_enable_msix_exact()`.

alex-ab, 2021-04-22:
Thanks for investigations. For me it sounds, that on the one hand the principal support in our wifi driver for msi-x needs adjustments and on the other hand the question most be solved, which interrupt style is appropriate for a pci device. According to the our experience and the quirk lists in the ported driver, just looking at the PCI capabilities of a device is not sufficient for the platform driver. Either the preferred interrupt mode must be negotiated by the device, e.g. telling on irq cap retrieval from the platform driver, or a quirk list must be maintained with the platform driver.

m-stein, 2021-04-22:
Thanks also from my side! I just rebased my sculpt_t490 branch onto the t490s branch of @jschlatow. Apparently this fixed the regression with my T490 as well.

alex-ab, 2021-04-22:
After a local discussion, it is preferred to have explicit msi msix xml attributes instead of no_msi no_msix as temporary fix. I will work on it.

alex-ab, 2021-04-22:
Please review and re-test 0addbd14127db92496697bfd8cd8617cc2674b45

m-stein, 2021-04-22:
@alex-ab Commit 0addbd1 gives me merge conflicts on jschlatow/t490s and it seems not to be connected to any branch. Can it be used as replacement for 74cdb03b05, f8d55c6f84, 98b4c5c631 on jschlatow/t490s?

alex-ab, 2021-04-22:
> @alex-ab Commit [0addbd1](https://github.com/genodelabs/genode/commit/0addbd14127db92496697bfd8cd8617cc2674b45) gives me merge conflicts on jschlatow/t490s and it seems not to be connected to any branch. Can it be used as replacement for [74cdb03](https://github.com/genodelabs/genode/commit/74cdb03b056c3b5fbdc68e2c146abe6f10f94ec2), [f8d55c6](https://github.com/genodelabs/genode/commit/f8d55c6f84a6f15b7b54aea625f7996be5b6f58d), [98b4c5c](https://github.com/genodelabs/genode/commit/98b4c5c6316d25d598bd8535e848e0f4af9c4706) on jschlatow/t490s?

The commit is part of the issue_4079 branch which is based on staging. cb891f0e793e0acd14deb9499b42459f20d5ae3a is a "backport" on the t490s branch.

m-stein, 2021-04-22:
@alex-ab Thanks for the backport! With commit cb891f0, the Wifi works fine but Now I can't access the partitions on my NVME-0 disk (Samsung mzvlb51, 500G) anymore. When clicking on it in the Storage dialog, instead of listing the different partitions (one of them my Genode partition) it just shows "Format device". Component "nvme-0.part_block" is started, however and I can't see anything suspicious in the log:
[image]
[image]

m-stein, 2021-04-22:
I'll come to the office tomorrow so we can debug it more easily.

jschlatow, 2021-04-22:
@m-stein @alex-ab am I mistaken or is `try_msi_msix` used as 'notry_msi_msix' [see here](https://github.com/genodelabs/genode/commit/cb891f0e793e0acd14deb9499b42459f20d5ae3a#diff-defe9b978805c2af49f5e1e8f4ed58ee7929cf8ddc351f1b70c9b9decdc15746R183)? I mean, with `try_msi_msix=true` the `_config_space` should be passed.

jschlatow, 2021-04-22:
> @m-stein @alex-ab am I mistaken or is `try_msi_msix` used as 'notry_msi_msix' [see [...]

I can confirm that commit cb891f0 works if I swap the `_config_space` and `~0UL` in the ternary operator.

alex-ab, 2021-04-22:
@m-stein, @jschlatow : thanks for testing and review/finding the mistake. I updated the both commits (staging/sculpt) accordingly.

m-stein, 2021-04-23:
@alex-ab Can confirm that with 8ec5b30e23b580f74596b92765dabf3b7dc2405d Wifi and the rest of my Sculpt work fine.

m-stein, 2021-04-23:
@jschlatow @alex-ab Thanks for dealing so fast with this problem.

nfeske, 2021-05-05:
I think the pressing concern of devices not working is addressed by the commit https://github.com/genodelabs/genode/commit/2db94b843836bf9f68ddd154296910d96eef57de on master.

It arguably side-steps the actual issue. So we will have to address this topic down the road sometime. However, as there is no urgent need and no concrete plan, I propose to close this issue.

alex-ab, 2021-05-06:
From my side we may close it and reopen whenever someone is working on actively.

closed 2021-05-06 by chelmuth