genodelabs/genode #4083: wifi_drv: no reconnect once the connection is dropped

nfeske, 2021-04-21:
With the current version of Sculpt, my wifi connection at home has become pretty unreliable. Sooner or later, the connection is dropped and does never recover. I'm able to revive the connection be restarting the wifi driver manually - until the connection is dropped the next time. Previous versions of Sculpt did not have this problem.

Today, @chelmuth managed to reproduce the problem manually by temporarily switching off and on our wifi access point. Sculpt expectedly loses the connection and (unexpectedly) never tries to reconnect.

When kicking the wifi driver a bit by saving the /config/wifi file (with no modifications) manually in vim, the driver reconnects.
We also observed that the reconnect works when setting the `connected_scan_interval` to value other than 0. This is of course counter-intuitive to me.

I can only guess that this regression is a side effect of the change to the uplink-session interface?

@cnuke Could you please have a look at this problem?

m-stein, 2021-04-21:
Don't know if this is related but since I updated Sculpt, the Wifi driver produces a page fault while scanning on my T490. In Spain I hadn't the time to deal with it so I just switched to Nic. The driver first complains that it failed to load a firmware chunk and then dies. I'll send the log later as I'm working and don't want to crash anything.

@nfeske @cnuke I cannot say for sure that the issues are not related to the Uplink session but we tested Sculpt several times on different machines before we merged the commits.

jschlatow, 2021-04-22:
@m-stein I experienced the same behaviour on my T490s and investigated (#4079). The firmware loading fails because of a timeout while the driver is waiting for an interrupt from the device.

cnuke, 2021-04-22:
@nfeske I am going to look into the problem, @chelmuth already informed me about yesterday's findings.

m-stein, 2021-04-22:
@jschlatow Thanks a lot for hinting me to this one. I just tried the commit and it works just fine.

@nfeske I cannot confirm the unreliable connection issue with my T490 so far.

cnuke, 2021-04-23:
@nfeske thanks for the log.

I already prepared commit f811367 that re-arms the scan timer in case that network cannot be found. It should remedy your problem as now the `sculpt_manager` should be able to react upon the _network not found_ state. That being said, everytime I encountered the same behaviour during my tests, even without any further scanning, the driver or rather the wireless device picked the lost network up again eventually.

So it would be interesting to know why that is not happening in your case but I am afraid without more detailed logging that is not possible. I will check why my suggestion to make the driver more verbose by setting `stdout="/dev/log` in its `<libc/>` node
apparently did not work on Sculpt (I used the `wifi.run` run-script during testing, maybe there is more to be done for Sculpt).

m-stein, 2021-05-06:
Yesterday, with Sculpt 21.03b (that contains the fixes for #4079), I had, for the first time with this release a sporadic disconnect of the Wifi from my home WLAN (I'm not often in this WLAN with Sculpt and at work I never experienced disconnects). The disconnect occured after quite some time (half an hour, an hour) of actively doing stuff over this WLAN. I tried to switch to NIC (worked) and switch back to Wifi (didn't work). I've attached [the log](https://github.com/genodelabs/genode/files/6433301/wifi_crash_log.txt) and are now running with the inspection commits that also @nfeske uses to get a more verbose log next time.

cnuke, 2021-05-06:
@m-stein thanks for the LOG file. It appears that at some point - for whatever reason - the device itself becomes unresponsive. The driver already tries to reset the device and fails. Even restarting the driver, which involves some kind of power off/on of the device by the platform driver, does not help as the device is still in an unresponsive state.

Since there are apparently not that many iwl 9xxx users yet, I appreciate getting more detailed debug information.

chelmuth, 2021-05-06:
> Since there are apparently not that many iwl 9xxx users yet, I appreciate getting more detailed debug information.

From this perspective it makes sense @jschlatow also enables debug information in his scenario. Also, it may be good to monitor iwlwireless resources for according bug reports/fixes. Last but not least, I wonder if enabling 11n support may change the behavior of driver and firmware to a more widespread and robust pattern (see following log snippet).

```
[runtime -> wifi_drv] Disable 11n mode
```

jschlatow, 2021-05-06:
I use wifi regularly but haven't seen any fatal disconnects yet. I already made some larger downloads, file transfers and git checkouts. Nonetheless, I will enable debugging info and stress the wifi a bit more.

@m-stein, let me know if you find out how to reproduce this. Do you have any details on the wifi access point with which it occured?

m-stein, 2021-05-06:
It occured with a FritzBox! 7430 YB, frequency of 2417 mhz, WPA2.

m-stein, 2021-05-07:
@jschlatow @cnuke [This](https://github.com/genodelabs/genode/files/6441673/wifi_connect_log.txt) is the log of a successful connect to my home WLAN with the more chatty driver. Maybe you can get more details about my WLAN from that. Unfortunately, the disconnect didn't happen again since I posted the first log.

cnuke, 2021-05-17:
@m-stein thanks for the LOG, I cannot spot anything suspicious. I fear we have to wait until the erroneous disconnect happens again.

jschlatow, 2021-05-17:
@cnuke since replacing my tp-link router with a new Fritz!Box 7520BH and using the 2,4GHz network by default, I occasionally experience a reconnect issue as well. It just happened for a second time. 
[Here](https://github.com/genodelabs/genode/files/6493417/wifi_disconnect.log) is the log file. I had cherry-picked your scan timer commit but, still, I had to switch to wired once and back to wifi to trigger the reconnect.

cnuke, 2021-05-17:
@jschlatow thank you very much for the LOG. It seems to be in line with a LOG @nfeske gave me to today (some EAPOL interaction and beacon loss followed by a local disconnect from which the driver apparently never recovers) and points to a general issue that is not specific to newer devices.

cnuke, 2021-05-25:
FWIW, I am using the `wifi_drv` at home today and I am connected to a Fritz!Box (in my case a 3490 via 5GHz) as well. So far I could not observe a disconnect. I am going to switch to 2.4GHz later. Currently I look for a way to make the driver itself more chatty in case it misses beacons.

chelmuth, 2021-05-25:
I'm still wondering if enabling 11n would make any difference...

cnuke, 2021-05-25:
@chelmuth that is still a question worth being answered indeed :-).

jschlatow, 2021-06-18:
@cnuke, I tried to reproduce the problem with another router and attached my backup fritzbox 7412 as a second AP to my primary fritzbox 7520 BH. I couldn't reproduce this using the model 7412 though whereas I regularly lose my connection to the 7520BH 2.4GHz network after a few minutes. Both devices are running the latest firmware available. I don't know whether this provides any indication about the cause of this issue or how to reproduce it. Let me know if I can be of any further help.

nfeske, 2022-05-25:
I'm closing this issue because the wifi driver has been replaced in the meanwhile.

closed 2022-05-25 by nfeske