genodelabs/genode #4084: vbox: high CPU load in Linux VM caused by udev

jschlatow, 2021-04-23:
When I was setting up a Linux VM on my Sculpt system, I noticed a high CPU utilisation in the VM caused by the systemd-udevd process.

As I'm using Archlinux, I first compiled a know-to-work kernel (5.4) and tried again. The problem persisted, but I could narrow it down to the following 'corner' case. On bootup and after login, everything is running smoothly unless I resize the VM window. Restoring the orignal size still leaves me with a rogue udevd. (Note, that I'm using i3 window manager and slim as login manager.)

A more precise way to see the effects in the VM is to start `udevadm monitor`. After login it reports a perfectly calm sea. Yet, once I resize the VM window, it reports a change event from `/device/pci0000:00/0000:00:02.0/drm/card0` every few milliseconds.

I took a cold dive into Virtualbox and noticed that the VM and VMM are apparently playing a very fast variant of ping-pong. Let me summarise what (I believe) is the sequence of events:

On resize, the VMM is calling `GenodeConsole::update_video_mode()`, which calls `SetVideoModeHint()`. The latter ends up in `vmmdevIPort_RequestDisplayChange()` where it detects that the resolution changed and the guest does not yet know about it, hence the guest is notified with a `VMMDEV_EVENT_DISPLAY_CHANGE_REQUEST`.
Normally, this event should be acknowledged by the guest so that the VMM updates the `lastReadDisplayChangeRequest` struct. Yet, in my case the event is never acknowledged with the consequence that `vmmdevIPort_RequestDisplayChange()` will always detect a changed resolution.
Yet, by some mechanism, the `get_modes` function of the `vboxvideo` module is invoked upon this event, which triggers the `VBVA_INFO_CAPS` guest-to-host command.
At some point in processing this command, the host calls `i_onAdditionsStateChange()`, which - in our implementation - calls `update_video_mode()` again and thereby closes the circle.

I can solve the issue for me by removing the `update_video_mode()` call in `i_onAdditionsStateChange()` but I assume it was put in this place for a reason. @alex-ab, @nfeske do you remember the intention?

alex-ab, 2021-04-23:
> I can solve the issue for me by removing the `update_video_mode()` call in `i_onAdditionsStateChange()` but I assume it was put in this place for a reason. @alex-ab, @nfeske do you remember the intention?

No, not really, the code is from 2016. It actually originated out of vbox4, if I am not mistaken, with commit
93f0cde72f159857dd4bdb93405c1c8074137d6b an reaction on i_onAdditionsStateChange was introduced. There is a issue connected, which may help to reason about.

alex-ab, 2021-04-23:
According to the original commit, the hint was not ever set, but depending on getFacilitiyStatus. Maybe this is just missing in the vbox5 case ?

jschlatow, 2021-04-23:
`getFacilityStatus` was removed with commit 0a1839e3e230f360e9154c0fda08ed52c7080adf.

chelmuth, 2021-04-23:
I'd suggest we give @jschlatow's change a shot in our own ZOO of Linux guests on developer machines and Windows test cases. From my experience all GUI elements of the vbox port where implemented with limited knowledge about the complex inner working of the original VirtualBox implementation. This fact also applies to my most recent "improvements" in dd4a3b02631a87d66536f9f10d9031d7e5a22867.

alex-ab, 2021-04-23:
> I'd suggest we give @jschlatow's change a shot in our own ZOO of Linux guests on developer machines and Windows test cases. From my experience all GUI elements of the vbox port where implemented [...]

I am fine with it.

nfeske, 2021-05-05:
I'm closing the issue since the fix entered the master branch today.

closed 2021-05-05 by nfeske