genodelabs/genode #4086: nic_router: avoid reusing the same nat port

sidhussmann, 2021-04-23:
The nat feature of nic_router always started the outgoing src port with
a fixed value of 49152. Consider following scenario:

```
client -> nic_router -> external_nat -> external_server
^                  ^
|                  |
+------Genode------+
```

If a `client` establishes a tcp connection to the `external_server`, and we
restart the Genode system, the client will connect to the server again.
However, from the server's perspective, the tcp session has not been
properly closed. Since the new connect will come from the same
src_ip/src_port combination, the server will forcefully reset the
connection. If that happens a lot, the server may even start to ignore
further connections from this ip/port combination for a while as a
mitigation. Since the `nic_router` always starts with the same port, even
the `external_nat` won't help, because its nat-table will still contain
the entry with src_ip/src_port combination. And from the `external_nat`'s
perspective, for each forwarded packet it will always refresh the
timeout for that entry.

sidhussmann, 2021-04-23:
670107dc1383ce925b246fc2d1d7535a2e1a8b33 adds a window of entropy to the dynamic port start, so the probability of running into this situation is minimized.

m-stein, 2021-04-23:
@sidhussmann Don't get me wrong, I see your problem here. But I'm hesitant regarding your solution. Isn't there a way to gracefully shut down your connections? Or asked the other way around - is your way of often restarting without maintaining a friendly communication towards the server a common use case?

sidhussmann, 2021-04-23:
@m-stein I understand your hesitation. Deciding to add a dependency to a third-party library should not be taken lightly. Maybe there is another source of entropy we could use?

> Or asked the other way around - is your way of often restarting without maintaining a friendly communication towards the server a common use case?

Most prominently, the issues shows up on our CI/CD-pipeline where we extensively test the reconfiguration of a dynamic runtime. We have dozens of tests with clients maintaining a connection to external servers. If one of the test fails, we run into this problem. This resulted in many false negatives.

But this issue also happens in productive systems. I believe the TCP session handling is a bit at odds with the desired-state configuration mechanism. We have a dynamic runtime (similar to the one in sculpt) which is controlled by adding/removing <start>-nodes to a deploy configuration.  We stress-test the scenario with re-configuring the dynamic runtime by adding/removing  TCP-client applications (or whole sub-systems). In order to gracefully shut down the connection in such a sculpt-like scenario, a component needs to be able to close the connection before it gets inexistent. As far as I understand, such a "shutdown"-protocol doesn't exist or did I miss(-understand) anything?

m-stein, 2021-04-23:
I would try not to introduce entropy at all. Apart from that, I don't know about any other way than using jitterentropy so far. 

AFAIK you are right that there is yet no general protocol for triggering a component shutdown from outside. Nonetheless, it could be locally implemented without much effort. You could re-configure the component to start the shutdown. Regarding an acknowledgement you could either wait for feedback from the component through a report that it writes and then destruct it. Or, even better, the component exits itself and you listen to the Init report that then indicates the exit.

chelmuth, 2021-04-27:
> [670107d](https://github.com/genodelabs/genode/commit/670107dc1383ce925b246fc2d1d7535a2e1a8b33) adds a window of entropy to the dynamic port start, so the probability of running into this situation is minimized.

A light-weight reliable source of entropy for components has been a topic for discussion since we integrated jitterentropy into more and more scenarios. We already sketched a solution that combines an initial random value for components with a PRNG (e.g., https://en.wikipedia.org/wiki/Xoroshiro128%2B) but need more time to settle on a final design.

But, I'm still uncertain about the original issue of *reused port numbers in NAPT scenarios*. Do I understand correctly that routers are completely free to map new connections to the same outer port mapping used before? Doesn't this mean that all implementations require port randomization not only for security considerations but also to function properly? Did you find some online resources dealing with this issue and also containing standard solution or best practices at least?

I'm also interested in your opinion about your test case in the light of the information above. Can you estimate which behavior of the client regarding port reuse after *not properly closing connections* is tolerated by the server? I'm convinced that "minimizing a probability" is rarely a robust solution and must be justified by hard information.

sidhussmann, 2021-04-27:
> Nonetheless, it could be locally implemented without much effort.

@m-stein your suggestion would introduce a set of implicit rules about the order on what configuration has to be written when. The management component should not need to know which configurations need to be written at what time. In our use-case this adds unreasonable amount of complexity to the handling of the dynamic configuration. This path leads to a slippery slope. 

Thinking a bit more about this issue, the problem at hand can be split in two:
1. Restarting the device (or the `nic_router` for that matter)
2. Restarting clients while `nic_router` keeps running

In https://github.com/genodelabs/genode/issues/4086#issue-865911752 I was really only talking about problem number 1. We haven't talked about problem number 2. 

In problem number 2, the way `nic_router` currently works is less than optimal. IMHO if a client dies and restarts, it most likely receives the worst possible NAT port: The same as before. 

> Do I understand correctly that routers are completely free to map new connections to the same outer port mapping used before? Doesn't this mean that all implementations require port randomization not only for security considerations but also to function properly? Did you find some online resources dealing with this issue and also containing standard solutions or best practices at least?

@chelmuth yes, there is the RFC 6056 that describes how the ephemeral ports should be selected. [Chapter 4](https://tools.ietf.org/html/rfc6056#section-4) describes how a NAPT should act. The RFC suggests four different algorithms for selecting ports. [Chapter 2.2](https://tools.ietf.org/html/rfc6056#section-2.2) mentions a simple increasing port allocation algorithm: "Traditional BSD Port Selection Algorithm".

I hesitate to implement one of the proposed algorithms, because they involve either entropy or hashing. But, we should at least have a "Traditional BSD Port Selection Algorithm", like it is described in section [2.2](https://tools.ietf.org/html/rfc6056#section-2.2).

Commit 1850f4d5cd8ae21aa7a5dd55402cd9cddc96d860 adds some safeguards, while commit e7ef23555a633f8e1a4155c142e33a2106af1f8b adds the "Traditional BSD Port Selection Algorithm".

m-stein, 2021-04-27:
@sidhussmann First of all, thanks for providing all the references
regarding this topic!

> @m-stein <https://github.com/m-stein> your suggestion would introduce a
> set of implicit rules about the order on what configuration has to be
> written when. The management component should not need [...]

Of course, I don't know your scenario in detail, but to me it didn't
seem much extra state. The transitions in the manager to kill a program
would be the same for each program, independent from its purpose/function.

> Thinking a bit more about this issue, the problem at hand can be split
> in two:
> 
>  1. Restarting the device (or the |nic_router| for that matter)
In this case we have only two options, either a "random" algorithm or
persistant storage. I wouldn't opt for the latter, so, we're left with
the former. RFC 6056 says "Could be random" in "2.2.  Ephemeral Port
Selection" about the initialization of the start port on system startup.
I don't know if this is enough reason for adding entropy to the router.
Any opinions?

>  2. Restarting clients while |nic_router| keeps running
> 
> In #4086 (comment)
> <https://github.com/genodelabs/genode/issues/4086#issue-865911752> I was
> really only talking about problem number 1. We haven't talked about
> problem number 2.
> 
> In problem number 2, the way |nic_router| currently works is less than
> optimal. IMHO if a client dies and restarts, it most likely receives the
> worst possible NAT port: The same as before.

You are completely right here, and you also surprised me as I wasn't
aware of this semantic. My understanding is, that I misinterpreted
the semantics of the ``_next`` member in the Bit_allocator or maybe it
changed over time.

> I hesitate to implement one of the proposed algorithms, because they
> involve either entropy or hashing. But, we should at least have a
> "Traditional BSD Port Selection Algorithm", like it is described in
> section 2.2 <https://tools.ietf.org/html/rfc6056#section-2.2>.
> 
> Commit 1850f4d
> <https://github.com/genodelabs/genode/commit/1850f4d5cd8ae21aa7a5dd55402cd9cddc96d860>
> adds some safeguards, while commit e7ef235
> <https://github.com/genodelabs/genode/commit/e7ef23555a633f8e1a4155c142e33a2106af1f8b>
> adds the "Traditional BSD Port Selection Algorithm".

Thanks for creating these commits! I'm definitely for adapting the behavior of the port allocation in the way that you depicted. However, I would refrain from adding another layer of abstraction. Therfore I added a fixup b82a2f7eb7 and ontop a general clean-up 2ffc6ce1f9 not directly related to these issues.

Furthermore, I would vote for getting rid of the try-and-error semantic on Bit_allocator::alloc. Maybe we shouldn't rely on the Bit_allocator at all as its semantic doesn't fit well. We could instead implement a local variant inside of Port_allocator. I just didn't have enough time today to do it.

What do you think?

chelmuth, 2021-04-28:
@sidhussmann thanks for the reference. I agree with your conclusion and like the approach of the increasing port number allocator. I'm also with @m-stein's fixup commit as it adds the feature to the port allocator without another artificial layer. I'd keep the bit allocator for now.

If you both agree, I'd merge the series to staging.

sidhussmann, 2021-04-28:
@m-stein and @chelmuth thank you for this productive discussion. 

> My understanding is, that I misinterpreted the semantics of the _next member in the Bit_allocator or maybe it changed over time.

Yes, it did change. My first commit (https://github.com/genodelabs/genode/commit/670107dc1383ce925b246fc2d1d7535a2e1a8b33) only solved problem number 1 (restart of the device/nic_router). 

> However, I would refrain from adding another layer of abstraction. Therfore I added a fixup b82a2f7 and ontop a general clean-up 2ffc6ce not directly related to these issues.

These commits look perfect! I added the additional layer of abstraction because I (probably falsely) assumed the patch would be easier for you to review. 
Maybe the C-style cast in https://github.com/genodelabs/genode/commit/b82a2f7eb71fe46b2ce4623cfcbebe65fbfd7d16#diff-4ecf214dc398126ba9ae7e0035abb9fa9287d68c8e157116ce62a25847ad9d63R42 could be changed to `static_cast`.

> Furthermore, I would vote for getting rid of the try-and-error semantic on Bit_allocator::alloc. Maybe we shouldn't rely on the Bit_allocator at all as its semantic doesn't fit well. We could instead implement a local variant inside of Port_allocator. I just didn't have enough time today to do it.

I agree that the way we currently interact with the Bit_allocator, and that we use exceptions in non-error cases, is a bit ugly. However, the way it currently is implemented, we'd hardly ever reach a case in productive environments where we have to catch `Bit_allocator::Range_conflict` in `Port_allocator::alloc()`. The "collision" only happens when `_next_port_offset` oversteps `NR_OF_PORTS`, and there have been connections alive for a very long time. The "collision" can also happen, when we call `Port_allocator::alloc(Port const port)` before. The way I understand however, this only happens during configuration changes. So, I'm with @chelmuth, that refactoring the `Bit_allocator` is not that pressing.

This brings me to a remaining open question. We haven't agreed how we deal with problem number 1 (restart of the device/nic_router). I understand that adding the complexity of jitterentropy is not a desired solution. I very much like the approach of having a lightweight pseudo-random generator as @chelmuth suggested. Maybe we can adapt `nic_router` to the way `lwip` is choosing their ephemeral ports? 

As @chelmuth pointed out, there are also security considerations. Implementing one of the four algorithms proposed in RFC 6056 would solve these issues, as well as the remaining problem number 1 (restart of the device/nic_router). 

> If you both agree, I'd merge the series to staging.

@chelmuth I'm happy with these commits. I believe we should add them to staging, even though one problem remains open. For the time being (in our repository), I will initialize `_next_port_offset` with a number generated by jitterentropy. This will be an easy patch that we can revert at a later time. I understand that this approach is not an option for you. Please let me know if I should publish this patch here as well for reference.

m-stein, 2021-04-28:
I updated my branch in a way that it is ready to be merged:
9c206591d6 nic_router: more descriptive port allocator code
bdd38fc3dc nic_router: use increasing src port for new nat

I'd love to also add an automated test that validates the allocation pattern, but not right now.

I merged the first 3 commits and slightly cleaned up the commit message. I hope everyone is fine with this. Furthermore, I, personally (although supporting the idea of randomizing the NAT ports) would wait whether the mentioned lightweight randomization feature becomes reality in the near future to jump right to it, instead of introducing jitterentropy now as an interim solution. I think it's a good idea to you keep it as local fix until then @sidhussmann.

chelmuth, 2021-04-28:
I merged both commits to staging.

m-stein, 2021-04-29:
@chelmuth Thanks!

closed 2021-05-05 by nfeske

sidhussmann, 2021-05-05:
@nfeske, I'm not sure if we should close this issue as it's only partially solved. @m-stein and @chelmuth what's your take on this?

nfeske, 2021-05-05:
It was closed automatically, triggered by the 'fixed' notice in the commit message.

I'd like to leave the answer to your question to @m-stein and @chelmuth. ;-)

chelmuth, 2021-05-05:
I'm uncertain if the issue should be left open as it can't be fixed without addressing the issue of initial component entropy (without using jitterentropy) and a light-weight PRNG first. Unfortunately, we did not agree on a plan when to address the prerequisites.

m-stein, 2021-05-05:
I don't have an opinion regarding this, we can leave it open as well.

sidhussmann, 2021-05-06:
I can't estimate how involving introducing a a light-weight PRNG is. Is there some preliminarily work with e.g.  https://en.wikipedia.org/wiki/Xoroshiro128%2B ?

Nonetheless, I believe there is value in the discussion of this issue regarding RFC 6056 and I think we should leave the issue open.

cnuke, 2021-05-06:
@sidhussmann FWIW, there is [lxip/random.cc](https://github.com/genodelabs/genode/blob/master/repos/dde_linux/src/lib/lxip/random.cc).

chelmuth, 2021-05-06:
I also value the information gathered in this issue and it is definitely preserved regardless of whether the issue is closed or open. However, if a *feature* issue stays open it becomes a liability and deserves planning how and when it will be solved. As I wrote before, we do not have a plan to address the component-entropy topic currently and I do not see it coming in the next months. As you reported you have a working intermediate fix on your branch, I think it's best to leave this issue closed.

sidhussmann, 2021-05-06:
> I also value the information gathered in this issue and it is definitely preserved regardless of whether the issue is closed or open. However, if a _feature_ issue stays open it becomes a liability [...]

Fair enough.

sidhussmann, 2021-05-06:
> @sidhussmann FWIW, there is [lxip/random.cc](https://github.com/genodelabs/genode/blob/master/repos/dde_linux/src/lib/lxip/random.cc).

@cnuke awesome! Thanks!

sidhussmann, 2021-05-06:
@cnuke since the seed is 42, this will always return the same sequence of numbers, right? What would be a good source for the seed in `nic_router`? We wouldn't want to depend on a RTC session, would we?

chelmuth, 2021-05-06:
Currently, *jitterentropy* is the entropy source of choice that should be used for seeding a PRNG.

For clarification, lxip uses `42` as an arbitrary starting point for pseudo-randomness in any ID selection, may it be ports, transaction IDs or sequence numbers. This approach fit the use case well up to now.

sidhussmann, 2021-05-06:
@chelmuth thank you for clarifying. This approach sounds reasonable for lxip. I can see now that it's not so straight forward to implement RFC 6056, if we don't want to depend on jitterentropy.

commit 34b1d07a (2021-04-28): nic_router: more descriptive port allocator code

This renames some members and local variables in the port allocator in order to
make the code more descriptive.

Fixes #4086