genodelabs/genode #4087: Apparently reversed capslock state in VM

m-stein, 2021-04-23:
I just accidentically pressed [ctrl]+[capslock] in my Linux VM in Sculpt and it apparently reversed my capslock state (i.e. LED off -> upper case, LED on -> lower case). This can be reverted by again pressing [ctrl]+[capslock]. Under native Linux [ctrl]+[capslock] also switches the case but together with the capslock LED. So, I assume that [ctrl]+[capslock] simply doesn't trigger LED handling in Sculpt while only [capslock] does.

alex-ab, 2021-04-23:
The capslock state is consumed by vbox reading a capslock ROM. In Sculpt the global_keys_handler (sculpt.run) generates the source report. It seems that there is no configuration for [ctrl]+[capslock] to toggle the capslock state.

chelmuth, 2021-04-27:
The background is that global keys are specified as the first key pressed, i.e., lose their *globalness* if pressed as secondary key in a combination. The best way to fix the issue would be the extension the special handling of capslock (as global state) in vbox. [capslock] could be filtered completely from the key event stream and only sent to the VM if the global state changes. But, this has high probability to bring more surprises for current configuration I'm afraid.

m-stein, 2021-04-27:
To me, the issue is not that annoying. I just wanted to make it public, so it doesn't run under cover. Shall I prioritize it over other work anyway right now?

chelmuth, 2021-04-27:
> To me, the issue is not that annoying. I just wanted to make it public, so it doesn't run under cover. Shall I prioritize it over other work anyway right now?

I don't think it deserves high priority currently. I just wanted to add my knowledge to the database ;-)

jschlatow, 2021-05-14:
For me, the capslock state doesn't seem to be consumed by vbox at all. Hitting capslock when the VM is not in focus will lead to an inconsistent state between host and guest.

jschlatow, 2021-05-17:
> For me, the capslock state doesn't seem to be consumed by vbox at all. Hitting capslock when the VM is not in focus will lead to an inconsistent state between host and guest.

Unfortunately, I cannot reproduce this anymore. Today, the capslock state is passed through as expected.

chelmuth, 2021-06-14:
With bdb91e77e758b0540d889a2c21845aa0ce754938 vbox5 filters `KEY_CAPSLOCK` from the input-event stream in ROM mode. Any feedback appreciated.

Commit f6e7c306e66fc369fb8faccc2dd4caf2396855c0 adds the `capslock="rom"` feature to vbox6. Please note, it's actually `rom` not `ROM`.

m-stein, 2021-06-14:
@chelmuth Thanks, I just updated my sculpt-installation and bdb91e7 works like a charm!

commit bdb91e77 (2021-06-07): vbox5: drop KEY_CAPSLOCK events on capslock="rom"

Guests are easily confused by spurious KEY_CAPSLOCK input events in
caps="rom" mode. These spurious events may reach the VMM if KEY_CAPSLOCK
is not pressed as first key in a combination and, therefore, is not
filtered as global key. Now, we filter KEY_CAPSLOCK in ROM mode in the
VMM explicitly, but let it pass in non-ROM mode.

Fixes #4087
commit 37f1873f (2021-06-25): vbox5: drop KEY_CAPSLOCK events on capslock="rom"

Guests are easily confused by spurious KEY_CAPSLOCK input events in
caps="rom" mode. These spurious events may reach the VMM if KEY_CAPSLOCK
is not pressed as first key in a combination and, therefore, is not
filtered as global key. Now, we filter KEY_CAPSLOCK in ROM mode in the
VMM explicitly, but let it pass in non-ROM mode.

Fixes #4087
closed 2021-06-25 by chelmuth