genodelabs/genode #4088: Pathname pointer check in the libc

cnuke, 2021-04-23:
Due to a configuration mistake on my side and a missing check in some contrib code I encountered a page-fault because `open(NULL, …)` was called. I traced the page-fault down to https://github.com/genodelabs/genode/blob/efbed6f7bfd2879b28ba209e8bbd2a7ae40ba0b4/repos/os/include/os/path.h#L36-L39
where this method is called by `Path_base::import()` with an invalid path pointer and path is unconditionally dereferenced.

Now, passing such a pointer to a libc function in this manner points probably to a programmer's mistake (or the fact that one relies on that being handled gracefully, i.e. returning _EFAULT_) and might only exist on the theoretical side of things.

cnuke, 2021-04-23:
Commit 2b9e289 (more of a band-aid) and 4afa399 lessen the problem - again I am indifferent to the practical relevenace in our situation but want to document the behaviour nonetheless.

nfeske, 2021-04-28:
Those changes are definitely for the better. Thanks for the commits, which went to staging just now.

commit 1bc439ac (2021-04-28): libc: check path pointer in file I/O functions

Fixes #4088
commit 8f30fc99 (2021-05-05): libc: check path pointer in file I/O functions

Fixes #4088
closed 2021-05-05 by nfeske