genodelabs/genode #4090: Sculpt inspect: hang up when inspecting passthrough USB

m-stein, 2021-04-26:
I have a SanDisk 32G USB stick that I passed through to my Linux VM by copying the corresponding device tag from report/drivers/usb_devices to config/launcher/usb_devices_rom. The USB stick shows up correctly in my VM and I could copy a certain amount of data to it. Now, when I clicked on "inspect" in the Leitzentrale Components view in the "USB -> usb-x-y (Sandisk) ..." node, the usb-x-y.drv gives me an ``Error: Device already claimed`` and shortly afterwards I see ``kernel [ 2] overmap attempt MEM - tree PD: 0x... -> 0x...``.

From that point, the inspect view shows only a black (transparent) background without any cursor/content. The components view still works except the sub-menus of the "USB" node (the node can however, still be opened and closed).

Ejecting the USB stick from inside the VM and then removing it from config/launcher/usb_devices_rom (using the system shell) doesn't solve the problem. I don't know whether it is possible to restart usb-x-y.drv with the Sculpt UI somehow.

nfeske, 2021-04-26:
You should always be able to recover inspect view by untoggling the inspect button in the USB device.

I wonder if the USB host driver is still in a healthy state or not. If not, it should be improved to deal more gracefully with the situation like your's where one USB device is requested by multiple sessions.

It seems to me that the Sculpt manager should also better not present the option to inspect the USB file system as long as the USB session is assigned to a VM. Right now, however, the Sculpt manager does not track USB-session routes. I think this issue will be addressed once we add proper USB device management for Sculpt (e.g., assigning a USB device to a VM via the GUI). There exists no concrete time frame for this feature though.

m-stein, 2021-04-26:
> You should always be able to recover inspect view by untoggling the inspect button in the USB device.
This doesn't work as the "usb-x-y ..." submenu of the "USB" dialog wont react to hovers or clicks anymore.

> I wonder if the USB host driver is still in a healthy state or not
How can I test this? I can still accces the files on the USB stick inside the VM at least.

nfeske, 2021-04-26:
@m-stein May you give https://github.com/genodelabs/genode/commit/ac8877624c23c5dc1afbf9f64ca9a87dcdadec04 a try?

m-stein, 2021-04-26:
Of course.

m-stein, 2021-04-26:
Now I see that the usb-x-y.drv and part_block etc. components are removed again as soon as it recognizes that the device can't be inspected. However, the inspect view and the "usb-x-y" submenu are still rendered dead afterwards.

nfeske, 2021-04-26:
But now you should be able to toggle the inspect button of another fs to regain control over the inspect view. Can you try?

nfeske, 2021-04-26:
The non-hoverable button of the USB device should also state "unsupported". Does it?

m-stein, 2021-04-26:
> The non-hoverable button of the USB device should also state "unsupported". Does it?

Yep, it does say "usb-x-y (Sandisk) unsupported".

In the log I can see ```[runtime] Warning: inspect: no route to service File_system (label="inspect -> vfs -> usb-x-y")```

m-stein, 2021-04-30:
I'll close this as it seems fixed.

nfeske, 2021-04-30:
Let's better keep it open until the fix enters the master branch.

commit ac887762 (2021-04-26): Sculpt: un-inspect failed USB storage

With this patch, the user is able to re-gain access to the inspect view
by toggling the inspect button of a file system unrelated to the failed
USB storage device.

Fixes #4090
closed 2021-05-05 by nfeske