genodelabs/genode #4092: VFS watch handlers should be executed at the application level, not I/O level

nfeske, 2021-04-26:
While analyzing issue #4091, I found that the code of watch handlers is executed at the I/O level as a direct side effect of the fs packet-stream handling. This is risky because such code may call application-level code that ends up executing `wait_and_dispatch_one_io_signal`. Depending on the application code, this can result in deadlock situations.

The problem should be fixed by replacing the direct call of `watch_response` (https://github.com/genodelabs/genode/blob/master/repos/os/src/lib/vfs/fs_file_system.h#L599) by merely marking the watch handler as triggered, and issuing a local signal to an application-level signal handler. This signal handler can then safely call all triggered watch handler functions.

cproc, 2024-02-07:
Commit 8d42d14 implements the suggested application-level signal handler solution.

nfeske, 2024-02-07:
Thank you @cproc.

I'm afraid that there is a second piece of the puzzle, which is the adaptation of the existing watch handlers used by VFS plugins, which - I suppose - use to rely on the I/O signal level.

nfeske, 2024-02-07:
Also the libc's `_terminal_resize_handler` and `_user_interrupt_handler` deserve a close look.

cproc, 2024-02-09:
> Also the libc's `_terminal_resize_handler` and `_user_interrupt_handler` deserve a close look.

I tried `vim.run` with the change and terminal resizing does not work anymore. So, if the libc can't execute application-level signal handlers, maybe the watch handlers need to detect and handle recursion instead if they call `wait_and_dispatch_one_io_signal()`? Maybe a generic helper struct like this could hide the complexity of that in the affected watch handlers?

```
struct Repeat_on_recursion
{
	bool in_progress { false };
	bool repeat { false };

	void operator() (auto const &fn)
	{
		if (in_progress) {
			repeat = true;
			return;
		}

		in_progress = true;

		do {
			repeat = false;
			fn();
		} while (repeat);

		in_progress = false;
	}
};

/* watch handler, might also be called directly at program start */
void _handle_file_update()
{
	static Repeat_on_recursion repeat_on_recursion;
	repeat_on_recursion([&] () {
		...
		/* might call _handle_file_update() again */
		wait_and_dispatch_one_io_signal();
		...
	});
}
```

nfeske, 2024-02-09:
> maybe the watch handlers need to detect and handle recursion [...]

I would prefer the straight-forward approach to offer two variants of the `Watch_handler`. The `Genode::Watch_handler` works at application level (as intended by this issue). An `Genode::Io::Watch_handler` would work on the I/O signal level. So at each use, one can intentionally select the right variant.

nfeske, 2024-02-09:
All VFS plugins would use Io::Watch_handler, which corresponds exactly how they are currently using the Genode::Watch_handler (operating on the I/O signal level). So all Watch_handler instances inside VFS plugins and the libc would become Io::Watch_handers.

Only applications would use the new Genode::Watch_handler (operating at app signal level).

cproc, 2024-02-09:
Yes, and in order to make sure that a `Genode::Watch_handler` always operates on application signal level and a `Genode::Io::Watch_handler` always operates on I/O signal level I would add two signal handlers (one app level and one I/O level) to each VFS plugin that produces watch notifications and instead of calling `Vfs_watch_handle::watch_response()` directly, the plugins would then send a local signal to both signal handlers and each signal handler would then go through the plugin's `Vfs_watch_handle` registry and call `watch_response()` on the handles with the matching desired signal level that is defined in `Vfs_watch_handle` depending on its creation via the `Watch_handler` constructor or the `Io::Watch_handler` constructor, correct?

cproc, 2024-02-09:
In some VFS plugins like the `Fs_file_system` one additional (app level) signal handler might suffice, because `_handle_ack()` already is an I/O-level signal handler which could call I/O level watch handlers directly, but on the other hand, adding another I/O level signal handler there as well could prevent recursive calls of `_handle_ack()` if the watch handler calls `wait_and_dispatch_one_io_signal()`. In other VFS plugin file systems like `Value_file_system` it is not known if `write()` was called from an application level signal handler or from an I/O level signal handler, so I would add signal handlers of both levels there to ensure that each watch handler is called on the correct level.

nfeske, 2024-02-12:
> the plugins would then send a local signal to both signal handlers

I have trouble following. I'm afraid you are intermixing the mechanisms for the detection of file changes (that's what the Watch_handler is all about) and the notifier-side about the file changes. The watch handlers inside the VFS plugins are only there for detecting changes of files the plugin depends on. E.g., the vfs_ttf plugin wants to know when the used ttf-file changes. When a plugin wants to notify the VFS user about changes of a file offered the plugin, the `watch_response` comes into play. It is all too easy to mix up these mechanisms. When contemplating changes of the `watch_response` side, the immediate use of signals is not feasible anyway because VFS users (specifically the VFS server) may need to translate file-change event into other forms of notifications (watch-response packets of the file-system session).

For the scope of this issue, however, the latter part `watch_response` is not of interest. Let us concentrate only on the `Watch_handler` side. All that needs to be done it replacing `Watch_handler` by `Io::Watch_handler` in the VFS plugins (and the libc).

cproc, 2024-02-12:
So, would the new application-level `Genode::Watch_handler` implementation contain an application-level signal handler to move from I/O-level to app-level? The current `Genode::Watch_handler` just calls the user-provided handler method on the signal level that `Watch_handler::watch_response()` was called on.

https://github.com/genodelabs/genode/blob/c4679e7af644f2ff8d07b9b48bc2c79eb57a79cc/repos/os/include/os/vfs.h#L957-L982

nfeske, 2024-02-12:
> would the new application-level Genode::Watch_handler implementation contain an application-level signal handler to move from I/O-level to app-level?

Yes.

The implementation may even _be_ an application-level signal handler (derived from `Genode::Signal_handler`). Privately, it could host an `Io::Watch_handler`, which - when called - issues a `local_submit` to the App-level signal handler to elevate the signal level from I/O to app-level. Then the user-provided handler function is always executed at app-level.

cproc, 2024-02-14:
Commit c6cdbcd implements the two `Watch_handler` variants, but there is a problem with the `terminal` component when an application-level watch handler is used, where `_handle_config()` (called from `Main()`) causes a watch notification (via `_root_dir.apply_config()`) and the `_handle_glyphs_changed()` watch handler triggers another `_handle_config()`, and so on. The problem did not occur with the I/O-level watch handler because when the watch handler was called immediately, the `_font.constructed()` condition was false and the watch handler did not send a config signal, but with the delayed execution of the application-level watch handler, `_font.constructed()` is true and the config signal is sent.

nfeske, 2024-02-14:
> variants, but there is a problem with the terminal component [...]

Thank you for the detailed description. This looks like the former use of the I/O-level watch handler merely masked a bug that already slumbered there.

As an interim fix, you may consider keeping the I/O-level watch handler in the terminal - just to avoid it being a show-stopper. Does the use of an `Io::Watch_handler` for the `_glyphs_changed_handler` restore the original behavior?

I'll look into changing the terminal/glyphs-fs/root-dir mechanics later, specifically looking for a way to break the cycle by detecting whether `apply_config` has any effect at all (e.g., by comparing the new config against a copy of the previous config). But that should better not be your concern in the scope of this issue.

cproc, 2024-02-14:
Yes, the `Io::Watch_handler` restores the original behavior. Commit 3ef6bd1 contains the updated version.

nfeske, 2024-02-14:
Thanks a lot.

While reviewing commit https://github.com/genodelabs/genode/commit/3ef6bd17b994f4f682de480c762b7d809b6da74e, I wondered if could we live without the second `Watch_handler` constructor (the one taking a `Vfs::File_system &` argument)? I'm unable to spot any use of it. For the `Io::Watch_handler`, the second constructor makes sense because the VFS plugins operate on VFS-internal data structures. But that is not expected by applications merely using the os/vfs.h API.

cproc, 2024-02-14:
Yes. Commit 1b1ce59 has the second constructor removed.

nfeske, 2024-02-14:
Merged to staging. Thank you for your patience.

commit dbc90839 (2024-02-14): vfs: support watch handlers on application signal level

Fixes #4092
closed 2024-02-29 by chelmuth