genodelabs/genode #4093: last column in sculpt's log_terminal shows the last printable character of a line

jschlatow, 2021-04-26:
The Log side bar in Sculpt's Leitzentrale uses the terminal implementation in non-wrapping mode, i.e. only a limited number of characters are printed of each line. A small bug in the terminal implementation, however, causes a subtle deviation from the expected behaviour:
Let's assume the side bar has 80 column. When printing a text line of 100 columns, I would expect the first 80 characters of the line to be shown. However, the implementation prints the first 79 characters + the 100th character (assuming the 101st character is `\n`).

Although not critical, this can be quite annoying at times. I tracked the issue down to the implementation of `Char_cell_array_character_screen::output()`, which always resets the cursor position to the last column once the width of the terminal was exceeded. As a result, the last column will be overwritten with the next character.

jschlatow, 2021-04-26:
As far as I can see, commit 6bf7e9cf1c056e89e713b3806147d53cc1a96edf fixes this issue.

nfeske, 2021-04-26:
I guess you referenced the wrong commit in the message but I can see the correct one.

The patch only leaves the question of what happens if `_boundary.width` is zero. Or can't it be zero?

jschlatow, 2021-04-26:
`_boundary.width=0` is fine, it means that no character will ever be printed.

On the other hand, I just recognised the missing lower bounds checking of `_current_pos.x` in the current implementation. An out-of-bounds access could potentially be triggered with multiple backspaces. I will have a thorough look at it tomorrow.

jschlatow, 2021-04-27:
Actually, I must revise my last statement about the backspace part. There is actually a lower bounds check that I must have overlooked yesterday. It appears the assumption is that the cursor position never leaves the defined boundaries to emulate terminal behaviour and so that character cells can be accessed without any bounds checking. I'm still unsure about whether all escape sequences adhere to this pattern though.

This is why `_current_pos.x` is currently reset to `_boundary.width -1`. My proposed fix would actually break this assumption. At the moment, I tend towards introducing a state variable to omit any further output of characters once we hit the last column instead.

nfeske, 2021-04-27:
> At the moment, I tend towards introducing a state variable to omit any further output of characters once we hit the last column instead.

I agree. An explicit state is much better for capturing the condition than a special cursor position.

jschlatow, 2021-04-27:
Commit 0c255c4 now contains the explicit state to handle the overflow condition. I haven't found any test case for the terminal implementation, though. So far, I've only tested it in Sculpt's log_terminal.

nfeske, 2021-04-28:
From a maintenance point of view, the patch looks quite invasive and repetitive for the arguably little gain it achieves.

I wonder, couldn't the patch become much smaller by adding the overflow handling to the `Cursor_guard`? I think of updating the overflowed state in the `Cursor_guard` destructor. I haven't looked closely though.

jschlatow, 2021-04-28:
I also thought about that but discarded the idea since this would disable the if statement at the end of the `output()` method. Yet, on a second thought, I might have an idea for that.

jschlatow, 2021-04-30:
@nfeske is commit a19d683 like what you had in mind?

nfeske, 2021-04-30:
Yes, that's beautiful! I left only one question inline.

BTW, the 50-character limit for commit messages applies only to the first line, the title of the message so to say. That's because this text is shown by various tools (also GitHub) limited to 50 characters by convention. You can keep the usual 80-character limit for the remainder of the commit message.

jschlatow, 2021-04-30:
Thanks for the review and the explanation about the character limit. I just pushed the updated commit 8090c9b, which also addresses your question.

nfeske, 2021-04-30:
That commit looks very nice. Merged to staging.

closed 2021-05-05 by nfeske