genodelabs/genode #4095: ssh_exec_terminal: scenario get stuck on various platforms

alex-ab, 2021-04-28:


alex-ab, 2021-04-28:
The both commits on the issue_4095 branch fixes one case where the test got stuck on my local testings.

nfeske, 2021-04-28:
Thanks a lot Alex for the fixes, which I merged to staging right away.

I'm probably not the perfect person to review the change of the ssh_terminal. Maybe @cnuke could have a look?

cnuke, 2021-04-29:
Commit 82943d6 LGTM :+1:.

nfeske, 2021-04-29:
I'm happy to report that your fixes had been very effective, @alex-ab! The ssh_exec_channel test works now in most cases, which is amazing.

There are a few timeouts (e.g., x86_32.pc.fiasco, x86_32.pc.sel4 on Qemu).

On seL4 (x86_32 on real hardware), the nic_drv issues a resource request, demanding 4 more caps.

On qemu/x86_32.pc.okl4, I can see a page fault that may be worth investigating:
```
[init -> nic_router] [downlink] NIC sessions: 1
no RM attachment (READ pf_addr=0xc pf_ip=0x8b493 from pager_object: pd='init -> ssh_terminal' thread='ep')
Warning: invalid signal-context capability
Warning: page-fault, pager_object: pd='init -> ssh_terminal' thread='ep' ip=0x8b493 pf-addr=0xc
```

On x86_64.pc.nova, a strange message occurs that I'm unable to interpret:
```
Connection to 10.0.0.9 closed by remote host.
Error: Spawned process died unexpectedly
```

alex-ab, 2021-04-29:
Nice ! I will investigate.

alex-ab, 2021-04-30:
> On x86_64.pc.nova, a strange message occurs that I'm unable to interpret:
> 
> ```
> Connection to 10.0.0.9 closed by remote host.
> Error: Spawned process died unexpectedly
> ```

It took me a while, to understand what goes on. It turns out that the ssh session is closed by the Genode client (via the ssh_terminal server), before the expected output is send via network. Because of that the sshpass process exits and the run script aborts. It seems that the ssh_terminal does not feature to flush internal buffers to network (which may delay the closing of the session for a specific time if it would), therefore the buffered internal data is dropped and never send.

I think without such a builtin flush feature, the test may fail arbitrarily, depending on scheduling of the threads on the used kernel. Sometimes the pthread becomes scheduled to send the data and succeeds before closing the session (by the EP thread), and sometimes the pthread does not get scheduled before the EP thread. Commit ebae3313be880f5cbc1faaba13f4ca44eb97e38d is a quirk which helps, but is also not 100% reliable. ba8c5df6a9399c1d64ed36187e777749024beb7d is required for seL4.

alex-ab, 2021-04-30:
> On qemu/x86_32.pc.okl4, I can see a page fault that may be worth investigating:
>
> ```
> [init -> nic_router] [downlink] NIC sessions: 1
> no RM attachment (READ pf_addr=0xc pf_ip=0x8b493 from [...]

This seems to be a more serious issue. The fault migrate and partially goes away, depending on instrumentation. I got the same fault also 2x times in core in the same code regions (slab.cc), as the one above in the ssh_terminal component. Because of that, I assume there is some memory corruption ongoing. Currently, I have no idea what is the underlying reason.

nfeske, 2021-04-30:
Thank you for the thorough and interesting analysis! I merged the sel4 quota fix without hesitation.

About the quirk I'm not so sure. It is quite impossible to draw the connection between those scheduling parameters to the problem that they solve. So the risk of accidentally re-introducing the effect later on is high without keeping your analysis back in our heads. Of course, fixing the limitation would be preferred in general. But as the component and this particular issue is not overly important for us compared to other topics right now, its hard to justify the effort.

Alternatively, would it be feasible to add an artificial delay as an ugly hack with comment that points to this issue? At least, this would keep the limitation in (distant) eye-sight while silencing the test.

alex-ab, 2021-04-30:
> Alternatively, would it be feasible to add an artificial delay as an ugly hack with comment that points to this issue? At least, this would keep the limitation in (distant) eye-sight while silencing the test.

In e49fa12d70e58ea7837dbdf4bc65c6fcbde07e12 in the bash client an additional sleep is added. sleep is not part of coreutils-minimal, because of that the coreutils package is used.

nfeske, 2021-04-30:
Very good! That's exactly what I had in mind. Merged to staging.

cnuke, 2021-04-30:
Briefly checking the implementation, commit 5ac97da could do the trick. Mind you I am not sure how that plays out when multiple ssh clients point to the same terminal, which in hindsight is a "misfeature".

nfeske, 2021-04-30:
Should I replace @alex-ab's quirk patch by @cnuke's commit at staging, for a test?

cnuke, 2021-04-30:
I would vote for giving 5ac97da a try over the weekend and fall-back to e49fa12 in case it does not work out or introduces other regressions to mitigate the problem until somebody has enough time to tackle it properly.

nfeske, 2021-04-30:
Aye aye.

cnuke, 2021-04-30:
@nfeske my bad, could you please remove commit 5e4e94c as it will trigger a compilation error - I'll provide another commit shortly. Sorry for inconvenience.

nfeske, 2021-04-30:
No problem.

cnuke, 2021-04-30:
@nfeske  Okay, second time is hopefully the charm - commit 01e4399 would be the way to go.

I checked on the x86_64 test machine as well as in Qemu/x86_64 where I could not trigger the _connection reset_ probleme anymore during my testing.

nfeske, 2021-04-30:
Thank @cnuke. I added your commit to staging.

alex-ab, 2021-05-07:
5fc098794076247323ebe9dd175074e4bb44f924 uses pthread_mutex instead of Genode::Mutex to avoid the sporadic deadlocks between ep and pthread

chelmuth, 2021-05-07:
Splendid, I merged the commit instantly. The mutex utility looks like a candidate for a *libc/mutex.h* because it makes the code much more readable. Also, it could then be merged with https://github.com/genodelabs/genode/blob/516a9a69253b650e256972c8433226dd5718db65/repos/libports/src/test/pthread/main.cc#L460-L492

tomga, 2021-05-08:
Not sure if it will help but I managed to make a run script that starts `ssh_terminal` with `gdb_monitor` and was able to get SIGTRAP under debugger few times. Sample output was:

```
Reading symbols from debug/ssh_terminal...done.
warning: remote target does not support file transfer, attempting to access files from local filesystem.
Reading symbols from /misc/tomga/projects/genode/genode/build/x86_64/var/libcache/ld-nova/ld-nova.lib.so...done.
Reading symbols from /misc/tomga/projects/genode/genode/build/x86_64/var/libcache/libc/libc.lib.so...done.
Reading symbols from /misc/tomga/projects/genode/genode/build/x86_64/var/libcache/vfs/vfs.lib.so...done.
Reading symbols from /misc/tomga/projects/genode/genode/build/x86_64/var/libcache/libcrypto/libcrypto.lib.so...done.
Reading symbols from /misc/tomga/projects/genode/genode/build/x86_64/var/libcache/libssh/libssh.lib.so...done.
Reading symbols from /misc/tomga/projects/genode/genode/build/x86_64/var/libcache/zlib/zlib.lib.so...done.
Continuing.
[New Thread 4]
[New Thread 6]
[New Thread 5]

Thread 2 received signal SIGTRAP, Trace/breakpoint trap.
Genode::Rpc_entrypoint::_activation_entry () at /misc/tomga/projects/genode/genode/repos/base-nova/src/lib/base/rpc_entrypoint.cc:124
124		addr_t id_pt; asm volatile ("" : "=D" (id_pt));
(gdb) bt
#0  Genode::Rpc_entrypoint::_activation_entry () at /misc/tomga/projects/genode/genode/repos/base-nova/src/lib/base/rpc_entrypoint.cc:124
#1  0x0000000000000000 in ?? ()
(gdb) info thread
  Id   Target Id         Frame 
  1    Thread 1          Genode::ipc_call (dst=..., snd_msg=..., rcv_msg=..., rcv_caps=rcv_caps@entry=0) at /misc/tomga/projects/genode/genode/repos/base-nova/src/lib/base/ipc.cc:82
* 2    Thread 2          Genode::Rpc_entrypoint::_activation_entry () at /misc/tomga/projects/genode/genode/repos/base-nova/src/lib/base/rpc_entrypoint.cc:124
  3    Thread 3          0x00000000000c108f in Nova::syscall_5 (p3=18446744073709551615, p2=<synthetic pointer>: <optimized out>, p1=<synthetic pointer>: <optimized out>, sel=<optimized out>, flags=1 '\001', s=Nova::NOVA_SM_CTRL) at /misc/tomga/projects/genode/genode/repos/base-nova/include/spec/64bit/nova/syscalls.h:314
  4    Thread 4          thread_stop_myself (myself=<optimized out>) at /misc/tomga/projects/genode/genode/repos/base-nova/include/spec/64bit/nova/syscalls.h:89
  5    Thread 6          thread_stop_myself (myself=<optimized out>) at /misc/tomga/projects/genode/genode/repos/base-nova/include/spec/64bit/nova/syscalls.h:89
  6    Thread 5          Genode::error<char const (&) [49]> () at /misc/tomga/projects/genode/genode/repos/base-nova/include/spec/64bit/nova/syscalls.h:154
(gdb) 
```
When I tried to compile `base-nova/src/lib/base/rpc_entrypoint.cc` with `-O0` then crash was not reproducible. Unfortunately currently I can't reproduce it anymore at all.

My run script is available on my [gdb_ssh_terminal](/tomga/genode/commits/gdb_ssh_terminal) branch. For some (unknown to me) reason Terminal sessions are not created in `Terminal` service announced by `ssh_terminal` and it causes: `PTY allocation request failed on channel 0` after connecting with ssh. Anyway when crash happened debugger properly stopped so maybe this script will be useful for someone. And if anyone sees some simple service routing mistake I made that causes this misbehavior then please let me know.

I also tried to run `ssh_terminal` with `SANITIZE_UNDEFINED = yes` but I think that currently it is not possible to use it with binaries using `libc` becaue it fails internally in `ubsan` after some errors about types not matching: `Env` and `Libc::Env_implementation` (I also tried to enable sanitizer on another component using `libc` main: `rump_fat` run script and similar crash occured).

tomga, 2021-05-13:
I found a root of the problem with crashes I experienced with `ssh_terminal`.

Scenario is following:
1. `echo-terminal` attaches to `Terminal` session which causes call to `ssh_poll_ctx_resize` which allocates `pollfd` array
2. in a second thread call to `poll` is performed with array allocated in point 1
3. before call to `poll` is finished `/bin/bash` attaches to `Terminal in thread from point 1 and calls to `ssh_poll_ctx_resize` again which moves array in a different memory location
4. later some allocations (happily in my reproducible testcase) caused reusing memory reallocated/freed in point 3 and overwriting it
5. after connecting with ssh call to `select` (used internally in `poll`) finished and some checks on already freed data was performed which caused page fault

Relevant stack traces are:
```
Thread 2 (LWP 3812608):
#0  pseudo_end () at <genode>/repos/base-linux/src/lib/syscall/spec/x86_64/lx_syscall.S:29
#1  0x00000000500451cb in wait_for_continue () at <genode>/repos/base-linux/src/lib/base/debug.cc:27
#2  0x000000000141b99f in ssh_poll_ctx_resize (ctx=0x7fffeee694c0, new_size=4) at <genode>/contrib/libssh-a475cccf96aa9126ed5fd135f108996c02f448e8/src/lib/libssh/src/poll.c:491
#3  0x000000000141ba9b in ssh_poll_ctx_add (ctx=0x7fffeee694c0, p=0x7fffeee69900) at <genode>/contrib/libssh-a475cccf96aa9126ed5fd135f108996c02f448e8/src/lib/libssh/src/poll.c:525
#4  0x000000000141c32b in ssh_event_add_fd (event=0x7fffeee27bb0, fd=16, events=1, cb=0x1029504 <write_avail_cb(socket_t, int, void*)>, userdata=0x7fffeee09020) at <genode>/contrib/libssh-a475cccf96aa9126ed5fd135f108996c02f448e8/src/lib/libssh/src/poll.c:779
#5  0x0000000001025a42 in Ssh::Terminal_session::Terminal_session (this=0x7fffeee09020, reg=..., conn=..., event_loop=0x7fffeee27bb0) at <genode>/repos/gems/src/server/ssh_terminal/server.cc:52
#6  0x0000000001027cc5 in Ssh::Server::attach_terminal (this=0x10501a8 <Libc::Component::construct(Libc::Env&)::main+9160>, conn=...) at <genode>/repos/gems/src/server/ssh_terminal/server.cc:346
#7  0x000000000100c392 in Terminal::Root_component::_create_session(char const*)::{lambda()#1}::operator()() const (__closure=0x401ff310) at <genode>/repos/gems/src/server/ssh_terminal/root_component.h:92
#8  0x0000000001015ec1 in Libc::Application_code::_execute<Terminal::Root_component::_create_session(char const*)::{lambda()#1}>(Genode::Meta::Empty&, Terminal::Root_component::_create_session(char const*)::{lambda()#1} const&) (this=0x401ff110, func=...) at <genode>/repos/libports/include/libc/component.h:92
#9  0x00000000010129b5 in Libc::with_libc<Terminal::Root_component::_create_session(char const*)::{lambda()#1}>(Genode::Functor const&)::Application_code_func::execute() (this=0x401ff110) at <genode>/repos/libports/include/libc/component.h:130
#10 0x0000000010f212db in Libc::Kernel::_user_entry (kernel=0x10ff7578 <unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space>) at <genode>/repos/libports/src/lib/libc/internal/kernel.h:308
#11 0x0000000010ee04f5 in call_func (sp=0x403fff68, func=0x10f2127b <Libc::Kernel::_user_entry(Libc::Kernel*)>, arg=0x10ff7578 <unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space>) at <genode>/repos/libports/src/lib/libc/spec/x86_64/internal/call_func.h:34
#12 0x0000000000000000 in ?? ()
```
and
```
Thread 4 (LWP 3812634):
#0  pseudo_end () at <genode>/repos/base-linux/src/lib/syscall/spec/x86_64/lx_syscall.S:29
#1  0x000000005004f6aa in lx_futex (val=0, op=0, uaddr=<optimized out>) at <genode>/repos/base-linux/src/lib/syscall/linux_syscalls.h:434
#2  thread_stop_myself (myself=<optimized out>) at <genode>/repos/base-linux/src/include/base/internal/lock_helper.h:62
#3  Genode::Lock::lock (this=0x404ff480, myself=...) at <genode>/repos/base/src/lib/base/lock.cc:128
#4  0x000000005004f783 in Genode::Lock::lock (this=0x404ff480) at <genode>/repos/base/src/lib/base/lock.cc:66
#5  0x0000000010f14e3d in Genode::Blockade::block (this=0x404ff480) at <genode>/repos/base/include/base/blockade.h:41
#6  0x0000000010f1fa2f in Libc::Pthread_blockade::block (this=0x404ff468) at <genode>/repos/libports/src/lib/libc/internal/pthread.h:324
#7  0x0000000010f1f483 in Libc::Monitor::Job::wait_for_completion (this=0x404ff450) at <genode>/repos/libports/src/lib/libc/internal/monitor.h:111
#8  0x0000000010f1f672 in Libc::Monitor::Pool::monitor (this=0x10ffb750 <unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space+16856>, job=...) at <genode>/repos/libports/src/lib/libc/internal/monitor.h:134
#9  0x0000000010f231df in Libc::Kernel::_monitor (this=0x10ff7578 <unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space>, fn=..., timeout_ms=0) at <genode>/repos/libports/src/lib/libc/internal/kernel.h:565
#10 0x0000000010f4bc52 in Libc::Monitor::monitor<select(int, fd_set*, fd_set*, fd_set*, timeval*)::<lambda()> >(const <lambda()> &, Libc::uint64_t) (this=0x10ff75a0 <unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space+40>, fn=..., timeout_ms=0) at <genode>/repos/libports/src/lib/libc/internal/monitor.h:82
#11 0x0000000010f4ab4e in select (nfds=13, readfds=0x404ffb20, writefds=0x404ffaa0, exceptfds=0x404ffa20, tv=0x0) at <genode>/repos/libports/src/lib/libc/select.cc:291
[... cut ...]
#20 0x00000000500aa3da in thread_start () at <genode>/repos/base-linux/src/lib/syscall/spec/x86_64/lx_clone.S:62
```

Currently I don't know what should be the way to fix this issue. Can somebody propose something? Should it be:
* avoid calling `ssh_*` functions from different threads
* attempt to force over-allocating this structure to not break with memory reallocation
* somehow break call to `poll/select`, do reallocation and force this listening on ssh connections thread to `poll/select` again

I feel all those options have their flaws. Other options?

chelmuth, 2021-05-14:
From what I understand, the ssh-terminal implementation misuses the API of libssh. The following is a short note from the libssh [documentation](https://api.libssh.org/stable/libssh_tutor_threads.html).

> At all times, you may use different sessions inside threads, make parallel connections, read/write on different sessions and so on. You cannot use a single session (or channels for a single session) in several threads at the same time. This will most likely lead to internal state corruption.

I'm far from a libssh expert, but I would read this as: If you use multi-threading, use one libssh context per thread or expect corruption.

tomga, 2021-05-18:
Branch [issue_4095](/tomga/genode/commits/issue_4095) contains fixes for page faults during execution of `ssh_terminal` run script. I checked it on Nova with qemu and on base-linux (with custom run script not provided here).

chelmuth, 2021-05-18:
Thanks @tomga, I merged both commits to staging.

alex-ab, 2021-05-18:
bd5c7872e560fe6f29ccc638e83de3ab969bdee6 avoids the deadlock which we see still nightly on old fiasco.

nfeske, 2021-05-28:
All tests are green now. Massive thanks to @alex-ab and everyone involved in fixing/improving the scenario!

May be close this issue?

alex-ab, 2021-05-28:
Yes, we may reopen if necessary.

closed 2021-05-28 by alex-ab