genodelabs/genode #4097: seoul: build errors with gcc 10

cproc, 2021-04-28:
When building seoul with gcc 10, the following errors occur:

```
    COMPILE  executor/vbios_reset.o
/.../contrib/seoul-e4c5e41a5d3b5398a310dbc39b52a5393526c816/src/app/seoul/executor/vbios_reset.cc: In member function ‘bool VirtualBiosReset::reset_helper(MessageBios&)’:
/.../contrib/seoul-e4c5e41a5d3b5398a310dbc39b52a5393526c816/src/app/seoul/executor/vbios_reset.cc:139:45: error: ‘void* memset(void*, int, size_t)’ clearing an object of non-trivial type ‘struct VirtualBiosReset::Resource’; use assignment or value-initialization instead [-Werror=class-memaccess]
  139 |     memset(_resources, 0, sizeof(_resources));
      |                                             ^
/.../contrib/seoul-e4c5e41a5d3b5398a310dbc39b52a5393526c816/src/app/seoul/executor/vbios_reset.cc:57:10: note: ‘struct VirtualBiosReset::Resource’ declared here
   57 |   struct Resource {
      |          ^~~~~~~~
```
```
    COMPILE  model/intel82576vf.o
/.../contrib/seoul-e4c5e41a5d3b5398a310dbc39b52a5393526c816/src/app/seoul/model/intel82576vf.cc: In member function ‘void Model82576vf::tx_queue::handle_ctx(uint64, tx_desc&)’:
/.../contrib/seoul-e4c5e41a5d3b5398a310dbc39b52a5393526c816/src/app/seoul/model/intel82576vf.cc:170:25: error: implicitly-declared ‘constexpr tx_desc& tx_desc::operator=(const tx_desc&)’ is deprecated [-Werror=deprecated-copy]
  170 |       ctx[desc.idx()] = desc;
      |                         ^~~~
In file included from /.../contrib/seoul-e4c5e41a5d3b5398a310dbc39b52a5393526c816/src/app/seoul/model/intel82576vf.cc:29:
/.../contrib/seoul-e4c5e41a5d3b5398a310dbc39b52a5393526c816/src/app/seoul/include/nul/net.h:53:3: note: because ‘tx_desc’ has user-provided ‘tx_desc::tx_desc(const tx_desc&)’
   53 |   tx_desc(const tx_desc &d) {
      |   ^~~~~~~
```
```
    COMPILE  model/vbios.o
/.../contrib/seoul-e4c5e41a5d3b5398a310dbc39b52a5393526c816/src/app/seoul/model/vbios.cc: In static member function ‘static bool StaticReceiver<Y>::receive_static(Device*, M&) [with M = CpuMessage; Y = VBios]’:
/.../contrib/seoul-e4c5e41a5d3b5398a310dbc39b52a5393526c816/src/app/seoul/model/vbios.cc:72:30: error: array subscript 512 is outside array bounds of ‘char [1]’ [-Werror=array-bounds]
   72 |  cpu->eip = BIOS_CODE_OFFSET + &end_vbios - &start_vbios - 1;
      |                              ^
/.../contrib/seoul-e4c5e41a5d3b5398a310dbc39b52a5393526c816/src/app/seoul/model/vbios.cc:25:26: note: while referencing ‘end_vbios’
   25 | extern char start_vbios, end_vbios;
      |                          ^~~~~~~~~
```
```
    COMPILE  executor/halifax.o
In file included from /.../contrib/seoul-e4c5e41a5d3b5398a310dbc39b52a5393526c816/src/app/seoul/executor/instcache.h:267,
                 from /.../contrib/seoul-e4c5e41a5d3b5398a310dbc39b52a5393526c816/src/app/seoul/executor/halifax.cc:22:
/.../contrib/seoul-e4c5e41a5d3b5398a310dbc39b52a5393526c816/src/app/seoul/executor/insthelper.h: In static member function ‘static void InstructionCache::exec_0f0018_ltr_0(InstructionCache*, void*, void*)’:
/.../contrib/seoul-e4c5e41a5d3b5398a310dbc39b52a5393526c816/src/app/seoul/executor/insthelper.h:489:9: error: ‘res’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  489 |   void *res;
      |         ^~~
/.../contrib/seoul-e4c5e41a5d3b5398a310dbc39b52a5393526c816/src/app/seoul/executor/insthelper.h: In member function ‘int InstructionCache::load_gdt_descriptor(InstructionCache::Descriptor&, short unsigned int, bool)’:
/.../contrib/seoul-e4c5e41a5d3b5398a310dbc39b52a5393526c816/src/app/seoul/executor/insthelper.h:489:9: error: ‘res’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
In file included from /.../contrib/seoul-e4c5e41a5d3b5398a310dbc39b52a5393526c816/src/app/seoul/genode/include/sys/utcb.h:23,
                 from /.../contrib/seoul-e4c5e41a5d3b5398a310dbc39b52a5393526c816/src/app/seoul/genode/include/service/string.h:58,
                 from /.../contrib/seoul-e4c5e41a5d3b5398a310dbc39b52a5393526c816/src/app/seoul/genode/include/executor/archcpustate.h:24,
                 from /.../contrib/seoul-e4c5e41a5d3b5398a310dbc39b52a5393526c816/src/app/seoul/include/executor/cpustate.h:19,
                 from /.../contrib/seoul-e4c5e41a5d3b5398a310dbc39b52a5393526c816/src/app/seoul/executor/halifax.cc:19:
/.../contrib/seoul-e4c5e41a5d3b5398a310dbc39b52a5393526c816/src/app/seoul/include/service/cpu.h: In static member function ‘static void InstructionCache::exec_0f0110_lgdt_2(InstructionCache*, void*, void*)’:
/.../contrib/seoul-e4c5e41a5d3b5398a310dbc39b52a5393526c816/src/app/seoul/include/service/cpu.h:132:71: error: ‘addr’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  132 |     if (operand_size == 2) *reinterpret_cast<unsigned int *>(tmp_dst) = *reinterpret_cast<unsigned int *>(tmp_src);
      |                            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /.../contrib/seoul-e4c5e41a5d3b5398a310dbc39b52a5393526c816/src/app/seoul/executor/instcache.h:267,
                 from /.../contrib/seoul-e4c5e41a5d3b5398a310dbc39b52a5393526c816/src/app/seoul/executor/halifax.cc:22:
/.../contrib/seoul-e4c5e41a5d3b5398a310dbc39b52a5393526c816/src/app/seoul/executor/insthelper.h:233:11: note: ‘addr’ was declared here
  233 |     void *addr;        \
      |           ^~~~
/.../contrib/seoul-e4c5e41a5d3b5398a310dbc39b52a5393526c816/src/app/seoul/executor/insthelper.h:245:1: note: in expansion of macro ‘helper_LDT’
  245 | helper_LDT(LGDT, gd, MTD_GDTR)
      | ^~~~~~~~~~

... and more errors of the same type
```

alex-ab, 2021-04-29:
Seoul works with dc6d05069a4d6177bc4231c8d7850478783533ab and the gcc 10 toolchain.

nfeske, 2021-04-29:
Thanks @alex-ab for the update. Merged to staging.

closed 2021-05-05 by nfeske

alex-ab, 2021-05-17:
@chelmuth, @nfeske: please add 068c2d43c2e4ce057e86c6d38adedecc915f6088 to staging

nfeske, 2021-05-18:
Merged to staging.

commit 755e3f2e (2021-04-29): seoul: adjust to GCC 10.03

Fixes #4097