genodelabs/genode #4098: text_area: build error with gcc 10

cproc, 2021-04-28:
When building `app/text_area` with gcc 10, the following error occurs:

```
  Program app/text_area/text_area
    COMPILE  dialog.o
/.../repos/gems/src/app/text_area/dialog.cc: In lambda function:
/.../repos/gems/src/app/text_area/dialog.cc:231:31: error: implicitly-declared ‘constexpr Text_area::Dialog::Position& Text_area::Dialog::Position::operator=(const Text_area::Dialog::Position&)’ is deprecated [-Werror=deprecated-copy]
  231 |                  .y = first_y };
      |                               ^
In file included from /.../repos/gems/src/app/text_area/dialog.cc:15:
/.../repos/gems/src/app/text_area/dialog.h:90:4: note: because ‘Text_area::Dialog::Position’ has user-provided ‘Text_area::Dialog::Position::Position(const Text_area::Dialog::Position&)’
   90 |    Position(Position const &other) : x(other.x), y(other.y) { }
      |    ^~~~~~~~
```

nfeske, 2021-04-29:
Does https://github.com/nfeske/genode/commit/6ead8476844c45728fb41296257459bd39008e55 help as a band-aid?

cproc, 2021-04-29:
@nfeske: yes, that fixes the build error

nfeske, 2021-04-30:
Ok, I merged my fix to staging then.

commit 6ead8476 (2021-04-30): app/text_area: fix build error with gcc 10

Fixes #4098
closed 2021-05-05 by nfeske