genodelabs/genode #4100: base: handle 'unpack_string()' error in 'Arg::string()'

cproc, 2021-04-28:
When building `drivers/ahci` for x86_64 with gcc 10, the following error occurs:

```
  Program drivers/ahci/spec/x86/ahci_drv
    COMPILE  main.o
In file included from /.../repos/base/include/util/token.h:17,
                 from /.../repos/base/include/util/xml_node.h:18,
                 from /.../repos/base/include/base/attached_rom_dataspace.h:18,
                 from /.../repos/os/src/drivers/ahci/main.cc:15:
/.../repos/base/include/util/string.h: In member function ‘virtual Genode::Session_capability Ahci::Main::session(const Session_args&, const Genode::Affinity&)’:
/.../repos/base/include/util/string.h:82:15: error: ‘buf’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
   82 |   for (; s && *s; s++, res++);
      |               ^~
```

The error message appears somewhat misleading (there is no `buf` variable at the indicated location). At the end it turned out that the actual error location is in the `Genode::label_from_args()` function:

```
inline Session_label label_from_args(char const *args)
{
	char buf[Session_label::capacity()];
	Arg_string::find_arg(args, "label").string(buf, sizeof(buf), "");
 
	return Session_label(Cstring(buf));
}
```

The `Genode::Arg::string()` function does not initialize 'buf' with the default string if `Genode::unpack_string()` returns with an error code.

cproc, 2021-04-28:
Fixed in 98bbf42

chelmuth, 2021-04-28:
Merged to staging.

commit 304fa037 (2021-04-28): base: handle 'unpack_string()' error in 'Arg::string()'

Fixes #4100
commit 190eafea (2021-05-05): base: handle 'unpack_string()' error in 'Arg::string()'

Fixes #4100
closed 2021-05-05 by nfeske