genodelabs/genode #4101: dde_ipxe: build error with gcc 10

cproc, 2021-04-28:
When building `drivers/nic` with gcc 10, the following error occurs:

```
  Program drivers/nic/ipxe_nic_drv
    LINK     ipxe_nic_drv
/.../x86_64-pc-elf/bin/ld: /.../build/x86_64/var/libcache/dde_ipxe_nic/dde_ipxe_nic.lib.a(x86_string.o):/.../contrib/dde_ipxe-621c4d00e447dccb654ee495b51c6accf79f541f/src/lib/dde_ipxe/src/include/compiler.h:749: multiple definition of `__licence_gpl2_or_later'; /.../build/x86_64/var/libcache/dde_ipxe_nic/dde_ipxe_nic.lib.a(rdtsc_timer.o):/.../contrib/dde_ipxe-621c4d00e447dccb654ee495b51c6accf79f541f/src/lib/dde_ipxe/src/include/compiler.h:749: first defined here
/.../x86_64-pc-elf/bin/ld: /.../build/x86_64/var/libcache/dde_ipxe_nic/dde_ipxe_nic.lib.a(bitops.o):/.../contrib/dde_ipxe-621c4d00e447dccb654ee495b51c6accf79f541f/src/lib/dde_ipxe/src/include/string.h:14: multiple definition of `__licence_gpl2_only'; /.../build/x86_64/var/libcache/dde_ipxe_nic/dde_ipxe_nic.lib.a(x86_string.o):/.../contrib/dde_ipxe-621c4d00e447dccb654ee495b51c6accf79f541f/src/lib/dde_ipxe/src/include/string.h:14: first defined here

... and more errors of the same type
```

The error occurs with gcc 10, because it has the '-fno-common' option set by default.

cproc, 2021-04-28:
Commit 8fbe6d5 fixes the problem by adding the '-fcommon' compiler option, which is also the upstream solution [1].

[1] https://github.com/ipxe/ipxe/commit/f982a712979619dbae2c6e0d741757e2ce94be11

chelmuth, 2021-04-28:
Merged to staging.

ssumpf, 2021-04-29:
> Commit [8fbe6d5](https://github.com/genodelabs/genode/commit/8fbe6d544574c058bd0bee560191a0bb8e6828d7) fixes the problem by adding the '-fcommon' compiler option, which is also the upstream solution [1].
> 
> [1] [ipxe/ipxe@f982a71](https://github.com/ipxe/ipxe/commit/f982a712979619dbae2c6e0d741757e2ce94be11)

This we saw in rump already (#3960). The default got changed from `-fcommon` to `-fno-common`. There might be many more places like this.

commit 86a8f1bc (2021-04-28): dde_ipxe: fix build error with gcc 10

Fixes #4101
closed 2021-05-05 by nfeske