genodelabs/genode #4103: readline: build error with gcc 10

cproc, 2021-04-28:
When building `LIB=readline` with gcc 10, the following error occurs:

```
  Library readline
    MERGE    readline.lib.so
genode-x86-ld: terminal.o:/.../contrib/readline-a0bfb07e4ee36edffd6eb908e2f57be2bb119fc8/src/lib/readline/terminal.c:103: multiple definition of `PC'; genode.o:/.../repos/libports/src/lib/readline/genode.cc:26: first defined here
genode-x86-ld: terminal.o:/.../contrib/readline-a0bfb07e4ee36edffd6eb908e2f57be2bb119fc8/src/lib/readline/terminal.c:103: multiple definition of `BC'; genode.o:/.../repos/libports/src/lib/readline/genode.cc:27: first defined here
genode-x86-ld: terminal.o:/.../contrib/readline-a0bfb07e4ee36edffd6eb908e2f57be2bb119fc8/src/lib/readline/terminal.c:103: multiple definition of `UP'; genode.o:/.../repos/libports/src/lib/readline/genode.cc:28: first defined here
```

cproc, 2021-04-28:
Fixed in 43746a8

chelmuth, 2021-04-30:
Merged to staging.

commit 58df465b (2021-04-30): readline: fix build error with gcc 10

Fixes #4103
closed 2021-05-05 by nfeske