genodelabs/genode #4105: virtualbox6: build errors with gcc 10

cproc, 2021-04-29:
When building virtualbox6 with gcc 10, the following errors occur:

```
  Library virtualbox6-client
    COMPILE  Main/src-client/ConsoleImpl2.o
In file included from /.../contrib/virtualbox6-aac8fe9dc30f5bc1488916af29e03d62199ac4ff/src/virtualbox6/include/VBox/HostServices/GuestPropertySvc.h:35,
                 from /.../contrib/virtualbox6-aac8fe9dc30f5bc1488916af29e03d62199ac4ff/src/virtualbox6/src/VBox/Main/include/ConsoleImpl.h:72,
                 from /.../contrib/virtualbox6-aac8fe9dc30f5bc1488916af29e03d62199ac4ff/src/virtualbox6/src/VBox/Main/src-client/ConsoleImpl2.cpp:36:
/.../contrib/virtualbox6-aac8fe9dc30f5bc1488916af29e03d62199ac4ff/src/virtualbox6/src/VBox/Main/src-client/ConsoleImpl2.cpp: In member function ‘int Console::i_configNetwork(const char*, unsigned int, unsigned int, INetworkAdapter*, PCFGMNODE, PCFGMNODE, PCFGMNODE, bool, bool)’:
/.../contrib/virtualbox6-aac8fe9dc30f5bc1488916af29e03d62199ac4ff/src/virtualbox6/include/iprt/err.h:152:49: error: narrowing conversion of ‘-38’ from ‘int’ to ‘unsigned int’ [-Wnarrowing]
  152 | #define VERR_ACCESS_DENIED                  (-38)
      |                                                 ^
/.../contrib/virtualbox6-aac8fe9dc30f5bc1488916af29e03d62199ac4ff/src/virtualbox6/src/VBox/Main/src-client/ConsoleImpl2.cpp:5265:30: note: in expansion of macro ‘VERR_ACCESS_DENIED’
 5265 |                         case VERR_ACCESS_DENIED:
      |                              ^~~~~~~~~~~~~~~~~~
```

```
  Program virtualbox6/virtualbox6
    COMPILE  dummies.o
In file included from /.../repos/ports/src/virtualbox6/dummies.cc:213:
/.../contrib/virtualbox6-aac8fe9dc30f5bc1488916af29e03d62199ac4ff/src/virtualbox6/src/VBox/Main/src-server/HostDnsService.h:44:22: error: ‘string’ is not a member of ‘std’
   44 |     std::vector<std::string> servers;
      |                      ^~~~~~
/.../contrib/virtualbox6-aac8fe9dc30f5bc1488916af29e03d62199ac4ff/src/virtualbox6/src/VBox/Main/src-server/HostDnsService.h:1:1: note: ‘std::string’ is defined in header ‘<string>’; did you forget to ‘#include <string>’?
  +++ |+#include <string>
    1 | /* $Id: HostDnsService.h $ */
```

I'm not sure what to do about the 'narrowing' error. I could not reproduce it when building the same VirtualBox version for Linux (with gcc 10.3.0 built from source), so not sure if this indicates a real bug or if for some reason the mismatch is just not detected when building for Linux and does not cause any problems in practice.

cproc, 2021-04-29:
Commit f1ab3be fixes the missing include error.

chelmuth, 2021-04-29:
I consider the code in *ConsoleImpl2.cc* sloppy and believe this is concealed by compiler flags. The code should not be
```c++
case VERR_ACCESS_DENIED:
```
but
```c++
case E_ACCESSDENIED:
```
as can be found in *src/VBox/Main/src-client/ConsoleImpl.cpp:9585*. @cproc could you give an according patch a try?

cproc, 2021-04-29:
@chelmuth: yes, that worked. There were also a few more 'narrowing conversion' errors:

```
  Program virtualbox6/virtualbox6
    COMPILE  nem.o
/.../repos/ports/src/virtualbox6/nem.cc: In function ‘void nemHCNativeNotifyPhysPageProtChanged(PVMCC, RTGCPHYS, RTHCPHYS, uint32_t, PGMPAGETYPE, uint8_t*)’:
/.../repos/ports/src/virtualbox6/nem.cc:475:27: error: narrowing conversion of ‘(fPageProt & (1 << 0))’ from ‘uint32_t’ {aka ‘unsigned int’} to ‘bool’ [-Werror=narrowing]
  475 |   .readable   = fPageProt & NEM_PAGE_PROT_READ,
/.../repos/ports/src/virtualbox6/nem.cc:476:27: error: narrowing conversion of ‘(fPageProt & (1 << 2))’ from ‘uint32_t’ {aka ‘unsigned int’} to ‘bool’ [-Werror=narrowing]
  476 |   .writeable  = fPageProt & NEM_PAGE_PROT_WRITE,
/.../repos/ports/src/virtualbox6/nem.cc:477:27: error: narrowing conversion of ‘(fPageProt & (1 << 1))’ from ‘uint32_t’ {aka ‘unsigned int’} to ‘bool’ [-Werror=narrowing]
  477 |   .executable = fPageProt & NEM_PAGE_PROT_EXECUTE,
```

I combined all fixes in commit  06d8b7b

chelmuth, 2021-04-29:
Great! Merged to staging.

commit ded72460 (2021-04-29): virtualbox6: fix build errors with gcc 10

Fixes #4105
closed 2021-05-05 by nfeske