genodelabs/genode #4106: virtualbox5: build errors with gcc 10

cproc, 2021-04-29:
When building virtualbox5 with gcc 10, the following errors occur:

```
  Library virtualbox5-main
    COMPILE  Main/src-client/ConsoleImpl2.o
In file included from /.../contrib/virtualbox5-e60d9717f4833b0cba86ddb3b461925d974c5fa1/src/app/virtualbox/include/iprt/string.h:33,
                 from /.../contrib/virtualbox5-e60d9717f4833b0cba86ddb3b461925d974c5fa1/src/app/virtualbox/include/iprt/cpp/ministring.h:30,
                 from /.../contrib/virtualbox5-e60d9717f4833b0cba86ddb3b461925d974c5fa1/src/app/virtualbox/include/VBox/com/string.h:46,
                 from /.../contrib/virtualbox5-e60d9717f4833b0cba86ddb3b461925d974c5fa1/src/app/virtualbox/include/VBox/com/ErrorInfo.h:30,
                 from /.../contrib/virtualbox5-e60d9717f4833b0cba86ddb3b461925d974c5fa1/src/app/virtualbox/src/VBox/Main/include/ObjectState.h:24,
                 from /.../repos/ports/src/virtualbox5/frontend/VirtualBoxBase.h:28,
                 from /.../contrib/virtualbox5-e60d9717f4833b0cba86ddb3b461925d974c5fa1/src/app/virtualbox/src/VBox/Main/include/ConsoleImpl.h:21,
                 from /.../contrib/virtualbox5-e60d9717f4833b0cba86ddb3b461925d974c5fa1/src/app/virtualbox/src/VBox/Main/src-client/ConsoleImpl2.cpp:34:
/.../contrib/virtualbox5-e60d9717f4833b0cba86ddb3b461925d974c5fa1/src/app/virtualbox/src/VBox/Main/src-client/ConsoleImpl2.cpp: In member function ‘int Console::i_configNetwork(const char*, unsigned int, unsigned int, INetworkAdapter*, PCFGMNODE, PCFGMNODE, PCFGMNODE, bool, bool)’:
/.../contrib/virtualbox5-e60d9717f4833b0cba86ddb3b461925d974c5fa1/src/app/virtualbox/include/iprt/err.h:745:49: error: narrowing conversion of ‘-38’ from ‘int’ to ‘unsigned int’ [-Wnarrowing]
  745 | #define VERR_ACCESS_DENIED                  (-38)
      |                                                 ^
/.../contrib/virtualbox5-e60d9717f4833b0cba86ddb3b461925d974c5fa1/src/app/virtualbox/src/VBox/Main/src-client/ConsoleImpl2.cpp:4977:30: note: in expansion of macro ‘VERR_ACCESS_DENIED’
 4977 |                         case VERR_ACCESS_DENIED:
      |                              ^~~~~~~~~~~~~~~~~~
```

```
  Program virtualbox5/virtualbox5
    COMPILE  frontend/console.o
/.../repos/ports/src/virtualbox5/frontend/console.cc: In function ‘void vboxClipboardDestroy()’:
/.../repos/ports/src/virtualbox5/frontend/console.cc:368:2: error: ‘free’ was not declared in this scope
  368 |  free(decoded_clipboard_content);
      |  ^~~~
/.../repos/ports/src/virtualbox5/frontend/console.cc:32:1: note: ‘free’ is defined in header ‘<cstdlib>’; did you forget to ‘#include <cstdlib>’?
   31 | #include "vmm.h"
  +++ |+#include <cstdlib>
   32 | 
/.../repos/ports/src/virtualbox5/frontend/console.cc: In function ‘int vboxClipboardReadData(VBOXCLIPBOARDCLIENTDATA*, uint32_t, void*, uint32_t, uint32_t*)’:
/.../repos/ports/src/virtualbox5/frontend/console.cc:436:3: error: ‘free’ was not declared in this scope
  436 |   free(decoded_clipboard_content);
      |   ^~~~
/.../repos/ports/src/virtualbox5/frontend/console.cc:436:3: note: ‘free’ is defined in header ‘<cstdlib>’; did you forget to ‘#include <cstdlib>’?
/.../repos/ports/src/virtualbox5/frontend/console.cc:438:38: error: ‘malloc’ was not declared in this scope
  438 |   decoded_clipboard_content = (char*)malloc(node.content_size());
      |                                      ^~~~~~
/.../repos/ports/src/virtualbox5/frontend/console.cc:438:38: note: ‘malloc’ is defined in header ‘<cstdlib>’; did you forget to ‘#include <cstdlib>’?
```

```
  Library virtualbox5-nova
    COMPILE  sup_old.o
/.../repos/ports/src/virtualbox5/spec/nova/sup_old.cc: In constructor ‘Periodic_gip::Thread::Thread()’:
/.../repos/ports/src/virtualbox5/spec/nova/sup_old.cc:59:5: error: ‘printf’ was not declared in this scope
   59 |     printf("Error: RTThreadCreate() %s\n", RTErrGetShort(rc));
      |     ^~~~~~
/.../repos/ports/src/virtualbox5/spec/nova/sup_old.cc:28:1: note: ‘printf’ is defined in header ‘<cstdio>’; did you forget to ‘#include <cstdio>’?
   27 | #include <iprt/uint128.h>
  +++ |+#include <cstdio>
   28 | #include <VBox/err.h>
/.../repos/ports/src/virtualbox5/spec/nova/sup_old.cc:60:5: error: ‘exit’ was not declared in this scope
   60 |     exit(-1);
      |     ^~~~
```

cproc, 2021-04-29:
Fixed in 9929b8e

chelmuth, 2021-04-30:
Merged to staging.

commit 2a529827 (2021-04-30): virtualbox5: fix build errors with gcc 10

Fixes #4106
closed 2021-05-05 by nfeske