genodelabs/genode #4107: bash: build error with gcc 10

cproc, 2021-04-29:
When building `noux-pkg/bash` with gcc 10, the following error occurs:

```
      [bash]  /.../x86_64-pc-elf/bin/ld: ./lib/termcap/libtermcap.a(termcap.o):/.../contrib/bash-c54af4376c78eb9e3d1463bd015ca4cd75770954/src/noux-pkg/bash/lib/termcap/termcap.c:295: multiple definition of `PC'; ./lib/readline/libreadline.a(terminal.o):/.../contrib/bash-c54af4376c78eb9e3d1463bd015ca4cd75770954/src/noux-pkg/bash/lib/readline/terminal.c:109: first defined here
      [bash]  /.../x86_64-pc-elf/bin/ld: ./lib/termcap/libtermcap.a(tparam.o):/.../contrib/bash-c54af4376c78eb9e3d1463bd015ca4cd75770954/src/noux-pkg/bash/lib/termcap/tparam.c:126: multiple definition of `BC'; ./lib/readline/libreadline.a(terminal.o):/.../contrib/bash-c54af4376c78eb9e3d1463bd015ca4cd75770954/src/noux-pkg/bash/lib/readline/terminal.c:109: first defined here
      [bash]  /.../x86_64-pc-elf/bin/ld: ./lib/termcap/libtermcap.a(tparam.o):/.../contrib/bash-c54af4376c78eb9e3d1463bd015ca4cd75770954/src/noux-pkg/bash/lib/termcap/tparam.c:127: multiple definition of `UP'; ./lib/readline/libreadline.a(terminal.o):/.../contrib/bash-c54af4376c78eb9e3d1463bd015ca4cd75770954/src/noux-pkg/bash/lib/readline/terminal.c:109: first defined here
```

Another '-fno-common' issue.

cproc, 2021-04-29:
Commit 99b9af2 fixes the problem by adding the '-fcommon' compiler option.

chelmuth, 2021-04-30:
Merged to staging.

commit 45eeeb1b (2021-04-30): bash: fix build error with gcc 10

Fixes #4107
closed 2021-05-05 by nfeske