genodelabs/genode #4108: block_cache: build error with gcc 10

cproc, 2021-04-29:
When building `server/block_cache` with gcc 10, the following error occurs:

```
  Program server/block_cache/block_cache
    LINK     block_cache
/.../x86_64-pc-elf/bin/ld: lru.o: in function `Cache::Chunk<4096u, Driver<Lru_policy>::Policy>::sync(unsigned long long, unsigned long long)':
/.../repos/os/src/server/block_cache/chunk.h:191: undefined reference to `Driver<Lru_policy>::Policy::sync(Lru_policy::Element const*, char*)'
```

cproc, 2021-05-06:
This problem is still a mystery.

chelmuth, 2021-05-07:
Commit be5a2f65c619f9c04a87b06ae57e60336ca9c11f fixes the issue by explicit instantiation of the template function. It seems GCC10 did not add the weak instance as GCC8 did.

commit be5a2f65 (2021-05-07): block_cache: explicitly instantiate template function

The explicit instantiation is required because the function is defined
in the compilation unit only but referenced externally.

Fixes #4108
commit a5385ceb (2021-05-10): block_cache: explicitly instantiate template function

The explicit instantiation is required because the function is defined
in the compilation unit only but referenced externally.

Fixes #4108
closed 2021-05-10 by chelmuth