genodelabs/genode #4109: nic_router: build errors with gcc 10

cproc, 2021-04-29:
When building `server/nic_router` with gcc 10, the following errors occur:

```
  Program server/nic_router/nic_router
    COMPILE  interface.o
In file included from /.../repos/os/include/net/ethernet.h:19,
                 from /.../repos/os/include/net/tcp.h:21,
                 from /.../repos/os/src/server/nic_router/interface.cc:15:
In function ‘void* Genode::memcpy(void*, const void*, Genode::size_t)’,
    inlined from ‘void Net::Network_address<<anonymous>, <anonymous>, <anonymous> >::copy(void*) [with unsigned int LEN = 6; char DELIM = ':'; bool HEX = true]’ at /.../repos/os/include/net/netaddress.h:55:39,
    inlined from ‘void Net::Arp_packet::dst_mac(Net::Mac_address)’ at /.../repos/os/include/net/arp.h:107:58,
    inlined from ‘void Net::Interface::_send_arp_reply(Net::Ethernet_frame&, Net::Size_guard&, Net::Arp_packet&)’ at /.../repos/os/src/server/nic_router/interface.cc:1351:13:
/.../repos/base/include/util/string.h:162:43: error: writing 6 bytes into a region of size 4 [-Werror=stringop-overflow=]
  162 |   for (i = 0; i < (size & 0x7); i++, *d++ = *s++);
      |                                      ~~~~~^~~~~~
In function ‘void* Genode::memmove(void*, const void*, Genode::size_t)’,
    inlined from ‘void* Genode::memcpy(void*, const void*, Genode::size_t)’ at /.../repos/base/include/util/string.h:143:18,
    inlined from ‘void Net::Network_address<<anonymous>, <anonymous>, <anonymous> >::copy(void*) [with unsigned int LEN = 6; char DELIM = ':'; bool HEX = true]’ at /.../repos/os/include/net/netaddress.h:55:39,
    inlined from ‘void Net::Arp_packet::dst_mac(Net::Mac_address)’ at /.../repos/os/include/net/arp.h:107:58,
    inlined from ‘void Net::Interface::_send_arp_reply(Net::Ethernet_frame&, Net::Size_guard&, Net::Arp_packet&)’ at /.../repos/os/src/server/nic_router/interface.cc:1351:13:
/.../repos/base/include/util/string.h:119:36: error: writing 6 bytes into a region of size 4 [-Werror=stringop-overflow=]
  119 |    for (i = 0; i < size; i++, *d++ = *s++);
      |                               ~~~~~^~~~~~
In function ‘void* Genode::memcpy(void*, const void*, Genode::size_t)’,
    inlined from ‘void Net::Network_address<<anonymous>, <anonymous>, <anonymous> >::copy(void*) [with unsigned int LEN = 6; char DELIM = ':'; bool HEX = true]’ at /.../repos/os/include/net/netaddress.h:55:39,
    inlined from ‘void Net::Arp_packet::dst_mac(Net::Mac_address)’ at /.../repos/os/include/net/arp.h:107:58,
    inlined from ‘Net::Interface::_broadcast_arp_request(const Net::Ipv4_address&, const Net::Ipv4_address&)::<lambda(void*, Net::Size_guard&)>’ at /.../repos/os/src/server/nic_router/interface.cc:1294:14,
    inlined from ‘void Net::Interface::send(Genode::size_t, FUNC&&) [with FUNC = Net::Interface::_broadcast_arp_request(const Net::Ipv4_address&, const Net::Ipv4_address&)::<lambda(void*, Net::Size_guard&)>]’ at /.../repos/os/src/server/nic_router/interface.h:414:17,
    inlined from ‘void Net::Interface::_broadcast_arp_request(const Net::Ipv4_address&, const Net::Ipv4_address&)’ at /.../repos/os/src/server/nic_router/interface.cc:1277:6:
/.../repos/base/include/util/string.h:162:43: error: writing 6 bytes into a region of size 4 [-Werror=stringop-overflow=]
  162 |   for (i = 0; i < (size & 0x7); i++, *d++ = *s++);
      |                                      ~~~~~^~~~~~
In function ‘void* Genode::memmove(void*, const void*, Genode::size_t)’,
    inlined from ‘void* Genode::memcpy(void*, const void*, Genode::size_t)’ at /.../repos/base/include/util/string.h:143:18,
    inlined from ‘void Net::Network_address<<anonymous>, <anonymous>, <anonymous> >::copy(void*) [with unsigned int LEN = 6; char DELIM = ':'; bool HEX = true]’ at /.../repos/os/include/net/netaddress.h:55:39,
    inlined from ‘void Net::Arp_packet::dst_mac(Net::Mac_address)’ at /.../repos/os/include/net/arp.h:107:58,
    inlined from ‘Net::Interface::_broadcast_arp_request(const Net::Ipv4_address&, const Net::Ipv4_address&)::<lambda(void*, Net::Size_guard&)>’ at /.../repos/os/src/server/nic_router/interface.cc:1294:14,
    inlined from ‘void Net::Interface::send(Genode::size_t, FUNC&&) [with FUNC = Net::Interface::_broadcast_arp_request(const Net::Ipv4_address&, const Net::Ipv4_address&)::<lambda(void*, Net::Size_guard&)>]’ at /.../repos/os/src/server/nic_router/interface.h:414:17,
    inlined from ‘void Net::Interface::_broadcast_arp_request(const Net::Ipv4_address&, const Net::Ipv4_address&)’ at /.../repos/os/src/server/nic_router/interface.cc:1277:6:
/.../repos/base/include/util/string.h:119:36: error: writing 6 bytes into a region of size 4 [-Werror=stringop-overflow=]
  119 |    for (i = 0; i < size; i++, *d++ = *s++);
      |                               ~~~~~^~~~~~
```

m-stein, 2021-04-30:
I'll look into it.

m-stein, 2021-05-03:
With these, building the NIC router should finish without warnings/errors:
89cb7bc98f net: circumvent stringop-overflow error
65881b0691 net: fix packed-conversion compiler warning

nfeske, 2021-05-03:
Thanks @m-stein for your thorough investigation. I merged both fixes.

commit 05f6d18f (2021-05-03): net: circumvent stringop-overflow error

With the update to GCC 10, the compiler stopped with an error when compiling
places where a MAC address is copied from outside into a packed object using
the Net::Netaddress::copy method (e.g. in
Net::Arp_packet::dst_mac(Mac_address)):

! error: writing 6 bytes into a region of size 4 [-Werror=stringop-overflow=]

While trying to find a clean solution for this error, I found posts on
gcc.gnu.org and github that stated that the size calculations that cause these
errors are incorrect. Indeed, I could verify that the actual size of the two
regions was stat [... cut]
closed 2021-05-05 by nfeske