genodelabs/genode #4110: nitpicker: build error with gcc 10

cproc, 2021-04-29:
When building `server/nitpicker` with gcc 10, the following error occurs:

```
  Program server/nitpicker/nitpicker
    COMPILE  user_state.o
/.../repos/os/src/server/nitpicker/user_state.cc: In member function ‘Nitpicker::User_state::Handle_forget_result Nitpicker::User_state::forget(const Nitpicker::View_owner&)’:
/.../repos/os/src/server/nitpicker/user_state.cc:420:18: error: narrowing conversion of ‘(((int)update_hover_result.Nitpicker::User_state::Update_hover_result::hover_changed) | ((int)hover_vanished))’ from ‘int’ to ‘bool’ [-Werror=narrowing]
  419 |   .hover_changed = update_hover_result.hover_changed
      |                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  420 |                  | hover_vanished,
      |                  ^~~~~~~~~~~~~~~~
```

nfeske, 2021-04-29:
Does it go away when replacing the bit-wise OR (`|`) by a logical OR (`||`)?

cproc, 2021-04-29:
@nfeske: yes it does. Fixed in b1295f9.

nfeske, 2021-04-29:
Thanks. Merged to staging.

commit cb3f08b9 (2021-04-29): nitpicker: fix build error with gcc 10

Fixes #4110
closed 2021-05-05 by nfeske