genodelabs/genode #4112: vfs_stress: build errors with gcc 10

cproc, 2021-04-29:
When building `test/vfs_stress` with gcc 10, the following errors occur:

```
Program test/vfs_stress/vfs_stress
    COMPILE  main.o
In file included from /.../repos/base/include/util/token.h:17,
                 from /.../repos/base/include/util/xml_node.h:18,
                 from /.../repos/os/include/vfs/types.h:19,
                 from /.../repos/os/include/vfs/directory_service.h:17,
                 from /.../repos/os/include/vfs/file_system.h:17,
                 from /.../repos/os/include/vfs/env.h:17,
                 from /.../repos/os/include/vfs/file_system_factory.h:17,
                 from /.../repos/os/include/vfs/simple_env.h:17,
                 from /.../repos/os/src/test/vfs_stress/main.cc:36:
/.../repos/base/include/util/string.h: In member function ‘void Populate_test::populate(int)’:
/.../repos/base/include/util/string.h:550:18: error: array subscript 1 is outside array bounds of ‘char [1]’ [-Werror=array-bounds]
  550 |    for (; str && *str && res < max_len; str++, res++);
      |                  ^~~~
/.../repos/os/src/test/vfs_stress/main.cc:226:8: note: while referencing ‘dir_type’
  226 |   char dir_type = *(path.base()+(path_len-2));
      |        ^~~~~~~~
In file included from /.../repos/base/include/util/token.h:17,
                 from /.../repos/base/include/util/xml_node.h:18,
                 from /.../repos/os/include/vfs/types.h:19,
                 from /.../repos/os/include/vfs/directory_service.h:17,
                 from /.../repos/os/include/vfs/file_system.h:17,
                 from /.../repos/os/include/vfs/env.h:17,
                 from /.../repos/os/include/vfs/file_system_factory.h:17,
                 from /.../repos/os/include/vfs/simple_env.h:17,
                 from /.../repos/os/src/test/vfs_stress/main.cc:36:
/.../repos/base/include/util/string.h: In member function ‘void Write_test::write(int)’:
/.../repos/base/include/util/string.h:550:18: error: array subscript 1 is outside array bounds of ‘char [1]’ [-Werror=array-bounds]
  550 |    for (; str && *str && res < max_len; str++, res++);
      |                  ^~~~
/.../repos/os/src/test/vfs_stress/main.cc:289:8: note: while referencing ‘dir_type’
  289 |   char dir_type = *(path.base()+(path_len-2));
      |        ^~~~~~~~
In file included from /.../repos/base/include/util/token.h:17,
                 from /.../repos/base/include/util/xml_node.h:18,
                 from /.../repos/os/include/vfs/types.h:19,
                 from /.../repos/os/include/vfs/directory_service.h:17,
                 from /.../repos/os/include/vfs/file_system.h:17,
                 from /.../repos/os/include/vfs/env.h:17,
                 from /.../repos/os/include/vfs/file_system_factory.h:17,
                 from /.../repos/os/include/vfs/simple_env.h:17,
                 from /.../repos/os/src/test/vfs_stress/main.cc:36:
/.../repos/base/include/util/string.h: In member function ‘void Read_test::read(int)’:
/.../repos/base/include/util/string.h:550:18: error: array subscript 1 is outside array bounds of ‘char [1]’ [-Werror=array-bounds]
  550 |    for (; str && *str && res < max_len; str++, res++);
      |                  ^~~~
/.../repos/os/src/test/vfs_stress/main.cc:363:8: note: while referencing ‘dir_type’
  363 |   char dir_type = *(path.base()+(path_len-2));
      |        ^~~~~~~~
```

chelmuth, 2021-05-05:
@cproc could you please check if b315454217d3980ead8c5e0cb322f8867c75f1d3 alone fixes the issue? I still have no GCC10 available. Nevertheless, I'll also keep 88792571b5b3597e782a22c9a881c5ce9dd8961f as a style improvement for *vfs_stress*.

cproc, 2021-05-06:
@chelmuth: yes, b315454 alone fixes the issue as well

chelmuth, 2021-05-06:
That's great, thanks.

commit b3154542 (2021-05-05): Check max_len before dereferencing pointer in Cstring constructor

Fixes #4112
commit 718f44ae (2021-05-10): Check max_len before dereferencing pointer in Cstring constructor

Fixes #4112
closed 2021-05-10 by chelmuth