genodelabs/genode #4113: cbe_tester: build error with gcc 10

cproc, 2021-04-29:
When building `app/cbe_tester` with gcc 10, the following error occurs:

```
  Program app/cbe_tester/cbe_tester
    COMPILE  main.o
/.../repos/gems/src/app/cbe_tester/main.cc: In member function ‘void Main::_cmd_pool_handle_pending_list_snapshots_cmds(bool&)’:
/.../repos/gems/src/app/cbe_tester/main.cc:2237:70: error: cannot bind packed field ‘ids.Cbe::Active_snapshot_ids::values[idx]’ to ‘long long unsigned int&’
 2237 |       log("   Snapshot #", snap_nr, " is generation ", ids.values[idx]);
      |                                                        ~~~~~~~~~~~~~~^
```

m-stein, 2021-04-30:
I will look into this today.

m-stein, 2021-04-30:
It seems that I need an older glibc:
```
  Library spark
    COMPILE  a-except.o
/usr/local/genode/tool/21.05/bin/../libexec/gcc/x86_64-pc-elf/10.3.0/gnat1: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /usr/local/genode/tool/21.05/bin/../libexec/gcc/x86_64-pc-elf/10.3.0/gnat1)
```
Is this correct?

m-stein, 2021-04-30:
Oh sorry, meant a newer one:
```
> ldd --version
ldd (Ubuntu GLIBC 2.27-3ubuntu1.4) 2.27
```

cproc, 2021-04-30:
Was the tool chain perhaps built on a newer system? Then building it (the tool chain) from source should work.

m-stein, 2021-04-30:
Ah, that's a good hint. Yeah, I built it on another machine because it takes unbelievably long :)

m-stein, 2021-04-30:
I'll do this on Monday and invest the rest of the day in the nic_router build problem instead.

nfeske, 2021-05-05:
@m-stein Please keep an eye on this issue because it is critical for switching to GCC 10.

m-stein, 2021-05-05:
Ok, I will prioritize it.

m-stein, 2021-05-05:
With 127b5e37d5, the test finishes successfully and without any compiler warnings/errors with GCC 10.

nfeske, 2021-05-05:
Cool! Merged to staging.

commit 88b19af8 (2021-05-05): cbe_tester: compile errors/warnings with gcc 10

Fixes #4113
commit 66feb939 (2021-05-10): cbe_tester: compile errors/warnings with gcc 10

Fixes #4113
closed 2021-05-10 by chelmuth