genodelabs/genode #4114: nova: build errors with gcc 10

cproc, 2021-04-29:
When building nova with gcc 10, the following errors occur:

```
  Program kernel/nova/hypervisor
    COMPILE  acpi_dmar.o
In file included from /.../contrib/nova-0316f8810c665d115fb860399d824a6531aa5aad/src/kernel/nova/include/pte.hpp:24,
                 from /.../contrib/nova-0316f8810c665d115fb860399d824a6531aa5aad/src/kernel/nova/include/dpt.hpp:23,
                 from /.../contrib/nova-0316f8810c665d115fb860399d824a6531aa5aad/src/kernel/nova/src/acpi_dmar.cpp:25:
/.../contrib/nova-0316f8810c665d115fb860399d824a6531aa5aad/src/kernel/nova/include/atomic.hpp: In member function ‘void Acpi_table_dmar::parse() const’:
/.../contrib/nova-0316f8810c665d115fb860399d824a6531aa5aad/src/kernel/nova/include/atomic.hpp:40:73: error: array subscript 4 is outside array bounds of ‘char [1]’ [-Werror=array-bounds]
   40 |         static inline void set_mask (T &ptr, T v) { __sync_or_and_fetch (&ptr, v); }
      |                                                     ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~
In file included from /.../contrib/nova-0316f8810c665d115fb860399d824a6531aa5aad/src/kernel/nova/include/buddy.hpp:23,
                 from /.../contrib/nova-0316f8810c665d115fb860399d824a6531aa5aad/src/kernel/nova/include/slab.hpp:23,
                 from /.../contrib/nova-0316f8810c665d115fb860399d824a6531aa5aad/src/kernel/nova/include/iommu_intel.hpp:24,
                 from /.../contrib/nova-0316f8810c665d115fb860399d824a6531aa5aad/src/kernel/nova/src/acpi_dmar.cpp:24:
/.../contrib/nova-0316f8810c665d115fb860399d824a6531aa5aad/src/kernel/nova/include/extern.hpp:29:13: note: while referencing ‘PAGE_H’
   29 | extern char PAGE_H;
      |             ^~~~~~
```
```
    COMPILE  cpu.o
/.../contrib/nova-63929e7d6c463df355f0e9326f06a2b8d2c0243c/src/kernel/nova/src/cpu.cpp: In static member function ‘static void Cpu::check_features()’:
/.../contrib/nova-63929e7d6c463df355f0e9326f06a2b8d2c0243c/src/kernel/nova/src/cpu.cpp:165:32: error: attribute ‘fallthrough’ not preceding a case label or default label [-Werror]
  165 |                 [[fallthrough]];
      |                                ^
```

cproc, 2021-04-29:
@alex-ab: I have a fix commit on my r10 nova branch [1]

[1] https://github.com/cproc/NOVA/commit/6f77dfc231c75a3ff325959d34f2bbde0a9d5b10

alex-ab, 2021-04-29:
@cproc: Thanks, I prepared the commit in your name.

chelmuth, 2021-04-29:
Merged to staging.

commit 0d69e71d (2021-04-29): nova: update kernel for gcc 10

Fixes #4114
closed 2021-05-05 by nfeske