genodelabs/genode #4116: Unconditional URB sglist access in usb_host_drv

cnuke, 2021-04-29:
Under certain conditions, namely when using a bounce-buffer, the `usb_host_drv`'s xHCI driver accesses the _URB_'s scatter-gather list unconditionally. The driver calls `sg_pcopy_to_buffer` and `sg_pcopy_from_buffer` which both are not implemented and lead to the component stopping its execution.

cnuke, 2021-04-29:
* 93e1b28 backports a fix from upstream where the sglist access depends on `urb->num_sgs` being set. AFAICT this field only gets set in `usb_sg_init` that apparently is not called in our port.
* ea83b91 changes the dummy implementations to mere `TRACE` dummies that return a size of _0_. This return value will trigger the check introduced in the previous commit so we will see a warning in the unlikely event that should happen.

Thanks to Peter for reporting the issue and will add the fixed label as soon as he confirms the fix.

nfeske, 2021-05-05:
I merged both commits to staging.

commit 908f2289 (2021-05-05): usb_host: backport xhci bounce-buffer fix

This commit backports the Linux upstream commit (d4a6106) [*]
that introduces a check to prevent the sglist from being used
unconditionally.

[*] 'xhci: fix bounce buffer usage for non-sg list case'

Fixes #4116.
commit d0f084d4 (2021-05-10): usb_host: backport xhci bounce-buffer fix

This commit backports the Linux upstream commit (d4a6106) [*]
that introduces a check to prevent the sglist from being used
unconditionally.

[*] 'xhci: fix bounce buffer usage for non-sg list case'

Fixes #4116.
closed 2021-05-10 by chelmuth