genodelabs/genode #4118: foc: errors with gcc 10 / binutils 2.36

cproc, 2021-04-29:
When building Fiasco.OC with gcc 10 and binutils 2.36, the following errors occur:

```
      [l4build]  gcc-10 is not supported.
      [l4build]  Please use a gcc of the following versions: 4.7 4.8 4.9 5 6 7 8 9
```
```
      [bootstrap]    [bootstrap] ... Compiling startup.o
      [bootstrap]  In file included from /.../contrib/foc-2e8cbb44d4c009238d96b66a9fc085e038f22e61/src/kernel/foc/l4/pkg/bootstrap/server/src/startup.cc:45:
      [bootstrap]  /.../contrib/foc-2e8cbb44d4c009238d96b66a9fc085e038f22e61/src/kernel/foc/l4/pkg/bootstrap/server/src/support.h:24:10: fatal error: cstring: No such file or directory
      [bootstrap]     24 | #include <cstring>
      [bootstrap]        |          ^~~~~~~~~
```
```
      [bootstrap]    [bootstrap] ==> Linking bootstrap32.elf
      [bootstrap]  /.../x86_64-pc-elf/bin/ld: error: linker script file '/.../contrib/foc-2e8cbb44d4c009238d96b66a9fc085e038f22e61/src/kernel/foc/l4/pkg/bootstrap/server/src/ARCH-amd64/boot32/bootstrap32.ld' appears multiple times
```
```
      [fiasco.oc]  /.../contrib/foc-2e8cbb44d4c009238d96b66a9fc085e038f22e61/src/kernel/foc/kernel/fiasco/src/Makeconf:260: *** gcc-10 is not supported. Please use one of the following versions: 4.7 4.8 4.9 5 6 7 8 9.  Stop.
```

When building for ARM platforms:

```
      [fiasco.oc]    ==> Linking fiasco.image
      [fiasco.oc]  genode-arm-ld: cannot use executable file 'bootstrap.fiasco.o' as input to a link
```

On x86_64, a page fault occurs on kernel start:

```
KERNEL: Warning: No page-fault handler for 0xfffffffffff32004, error 0x0, pc fffffffff020aa1f
RAX fffffffffff32004    RBX 0000007fff000000
RCX 00000000001036e7    RDX 0000000000005001
RSI 0000000000000000    RDI 0000007fff000018
RBP 0000000000107fa0    RSP ffffffff11d53fe0
R8  0000000000000000    R9  000000000000000f
R10 000000000000000f    R11 0000000000000282
R12 00000000001043c7    R13 000000000000000f
R14 0000007fff000010    R15 0000000000000000
RIP fffffffff020aa1f RFLAGS 0000000000000002
CS 0008 SS 0010

trapno 14, error 0, from kernel mode
page fault linear address fffffffffff32004

KERNEL: Warning: Sigma0 raised an exception --> HALT
```

Most of the errors already have upstream fixes, except for the ARM link problem, which is currently being worked on [1].

[1] https://github.com/kernkonzept/fiasco/issues/4

cproc, 2021-04-29:
Commit e0bfff4 fixes the errors except ARM link.

nfeske, 2021-05-05:
Merged to staging.

cproc, 2021-05-06:
The ARM link error has been fixed upstream. Commit 58b1ef5 adds the patch.

nfeske, 2021-05-06:
I merged the fixup to staging.

commit cb99d146 (2021-05-05): foc: fix errors with gcc 10 and binutils 2.36

Fixes #4118
commit c7b2314d (2021-05-10): foc: fix errors with gcc 10 and binutils 2.36

Fixes #4118
closed 2021-05-10 by chelmuth