genodelabs/genode #4122: cxx: add 'strchr()'

cproc, 2021-04-29:
When building `test/cap_integrity` for the foc/rpi3 platform with gcc 10, the following error occurs:

```
  Program test/cap_integrity/test-cap_integrity
    LINK     test-cap_integrity
/.../aarch64-none-elf/bin/ld: /.../build/arm_v8a/var/libcache/cxx/cxx.lib.a(supc++.o): in function `aarch64_cie_signed_with_b_key':
/.../build/tool_chain-21.05/aarch64/gcc/aarch64-none-elf/libgcc/./md-unwind-support.h:46: undefined reference to `strchr'
```

cproc, 2021-04-29:
Fixed in b7a7100

nfeske, 2021-04-29:
If this function is only needed for one test, wouldn't it be sensible to add the symbol local to the test instead of extending the cxx library?

cproc, 2021-04-29:
Apparently the function is only needed for this one test because this test statically links the cxx lib. From looking at the gcc patch which introduced the dependency [1] it seems unlkely that the function will be called on Genode, so I guess even a dummy function in the test will do. I'll open a new issue for that.

[1] https://gcc.gnu.org/legacy-ml/gcc-patches/2018-11/msg00105.html

cproc, 2021-04-29:
Unfortunately it turned out that run scripts (like log.run) don't start when the symbol is missing:

```
genode build completed
using 'core-foc-rpi3.a' as 'core.a'
using 'ld-foc.lib.so' as 'ld.lib.so'
/.../aarch64-none-elf/bin/ld: debug/core-foc-rpi3.a(supc++.o): in function `aarch64_cie_signed_with_b_key':
/.../build/tool_chain-21.05/aarch64/gcc/aarch64-none-elf/libgcc/./md-unwind-support.h:46: undefined reference to `strchr'
collect2: error: ld returned 1 exit status
    while executing
"exec [cross_dev_prefix]g++ {*}$arch -nostdlib {*}[core_ld_opts]  -Wl,-z -Wl,max-page-size=0x1000  -Wl,-Ttext=$link_address -Wl,-gc-sections  -Wl,-nost..."
    (procedure "build_core" line 18)
    invoked from within
"build_core debug/$core_obj {} [run_dir].core [core_link_address]"
    (procedure "build_core_image" line 16)
    invoked from within
"build_core_image $binaries"
    (procedure "run_boot_dir_arm" line 7)
    invoked from within
"run_boot_dir_arm $binaries"
    (procedure "run_boot_dir" line 5)
    invoked from within
"run_boot_dir $binaries"
    (procedure "build_boot_image" line 35)
    invoked from within
"build_boot_image "core ld.lib.so init test-log""
    (file "/.../repos/base/run/log.run" line 23)
    invoked from within
"source $include_name"
    ("foreach" body line 6)
    invoked from within
"foreach include_name [get_cmd_arg --include ""] {
	# first check if the include name is absolute
	if {[string first "/" $include_name] == 0} {
		puts ..."
    (file "/.../tool/run/run" line 1062)
Makefile:395: recipe for target 'run/log' failed
```

nfeske, 2021-04-30:
Given this turn of events, I added b7a7100e0528b5e0f9ed28a56fd175d3c3397af7 to staging now.

commit c3871ca9 (2021-04-30): cxx: add 'strchr()'

Fixes #4122
closed 2021-05-05 by nfeske