genodelabs/genode #4123: lx_fs: build errors with gcc 10

cproc, 2021-04-30:
When building `server/lx_fs` with gcc 10, the following errors occur:

```
  Program server/lx_fs/lx_fs
    COMPILE  main.o
In file included from /.../repos/os/src/server/lx_fs/directory.h:24,
                 from /.../repos/os/src/server/lx_fs/main.cc:24:
/.../repos/os/src/server/lx_fs/file.h: In member function ‘virtual File_system::Status Lx_fs::File::status()’:
/.../repos/os/src/server/lx_fs/file.h:162:42: error: narrowing conversion of ‘(st.stat::st_mode & 256)’ from ‘__mode_t’ {aka ‘unsigned int’} to ‘bool’ [-Werror=narrowing]
  162 |     .rwx   = { .readable   = (st.st_mode & S_IRUSR),
      |                              ~~~~~~~~~~~~^~~~~~~~~~
/.../repos/os/src/server/lx_fs/file.h:163:42: error: narrowing conversion of ‘(st.stat::st_mode & 128)’ from ‘__mode_t’ {aka ‘unsigned int’} to ‘bool’ [-Werror=narrowing]
  163 |                .writeable  = (st.st_mode & S_IWUSR),
      |                              ~~~~~~~~~~~~^~~~~~~~~~
/.../repos/os/src/server/lx_fs/file.h:164:42: error: narrowing conversion of ‘(st.stat::st_mode & 64)’ from ‘__mode_t’ {aka ‘unsigned int’} to ‘bool’ [-Werror=narrowing]
  164 |                .executable = (st.st_mode & S_IXUSR) },
      |                              ~~~~~~~~~~~~^~~~~~~~~~
In file included from /.../repos/os/src/server/lx_fs/main.cc:24:
/.../repos/os/src/server/lx_fs/directory.h: In member function ‘virtual Genode::size_t Lx_fs::Directory::read(char*, Genode::size_t, File_system::seek_off_t)’:
/.../repos/os/src/server/lx_fs/directory.h:225:42: error: narrowing conversion of ‘(st.stat::st_mode & 256)’ from ‘__mode_t’ {aka ‘unsigned int’} to ‘bool’ [-Werror=narrowing]
  225 |     .rwx   = { .readable   = (st.st_mode & S_IRUSR),
      |                              ~~~~~~~~~~~~^~~~~~~~~~
/.../repos/os/src/server/lx_fs/directory.h:226:42: error: narrowing conversion of ‘(st.stat::st_mode & 128)’ from ‘__mode_t’ {aka ‘unsigned int’} to ‘bool’ [-Werror=narrowing]
  226 |                .writeable  = (st.st_mode & S_IWUSR),
      |                              ~~~~~~~~~~~~^~~~~~~~~~
/.../repos/os/src/server/lx_fs/directory.h:227:42: error: narrowing conversion of ‘(st.stat::st_mode & 64)’ from ‘__mode_t’ {aka ‘unsigned int’} to ‘bool’ [-Werror=narrowing]
  227 |                .executable = (st.st_mode & S_IXUSR) },
      |                              ~~~~~~~~~~~~^~~~~~~~~~
/.../repos/os/src/server/lx_fs/directory.h: In member function ‘virtual File_system::Status Lx_fs::Directory::status()’:
/.../repos/os/src/server/lx_fs/directory.h:257:42: error: narrowing conversion of ‘(st.stat::st_mode & 256)’ from ‘__mode_t’ {aka ‘unsigned int’} to ‘bool’ [-Werror=narrowing]
  257 |     .rwx   = { .readable   = (st.st_mode & S_IRUSR),
      |                              ~~~~~~~~~~~~^~~~~~~~~~
/.../repos/os/src/server/lx_fs/directory.h:258:42: error: narrowing conversion of ‘(st.stat::st_mode & 128)’ from ‘__mode_t’ {aka ‘unsigned int’} to ‘bool’ [-Werror=narrowing]
  258 |                .writeable  = (st.st_mode & S_IWUSR),
      |                              ~~~~~~~~~~~~^~~~~~~~~~
/.../repos/os/src/server/lx_fs/directory.h:259:42: error: narrowing conversion of ‘(st.stat::st_mode & 64)’ from ‘__mode_t’ {aka ‘unsigned int’} to ‘bool’ [-Werror=narrowing]
  259 |                .executable = (st.st_mode & S_IXUSR) },
      |                              ~~~~~~~~~~~~^~~~~~~~~~
```

cproc, 2021-04-30:
Fixed in 169bf04

nfeske, 2021-04-30:
Merged to staging.

In principle, we may opt to avoid such hard casts be changing the condition to `(st.st_mode & S_IRUSR) != 0`. What do you think?

cproc, 2021-04-30:
> In principle, we may opt to avoid such hard casts be changing the condition to `(st.st_mode & S_IRUSR) != 0`. What do you think?

Sure, I changed it in fixup commit 979de76.

nfeske, 2021-04-30:
All the better. :-) Merged to staging.

commit 0b8ed9ab (2021-04-30): lx_fs: fix build errors with gcc 10

Fixes #4123
closed 2021-05-05 by nfeske