genodelabs/genode #4124: sculpt manager: cut arrow button

m-stein, 2021-04-30:
When I open ``config -> keyboard / launcher / managed`` in Sculpt in the Leitzentrale Files view, the left-oriented arrow button (the "back" button)  in front of keyboard / launcher / managed is shown only partially. The tip of the arrow is cut away. However, this doesn't have any functional implications AFAICT.

m-stein, 2021-04-30:
I just noticed that this is also affected by the set font size. With small fonts the effect becomes less visible. With large fonts, however, the arrows are cut more - even the arrows in the root directory  ("enter" buttons) - and also the radio buttons in the menu are stretched vertically with this font size.

nfeske, 2021-04-30:
Thanks for reporting. I can reproduce the issue.

nfeske, 2021-05-05:
Fixed by https://github.com/genodelabs/genode/commit/ef6c37791e6fb75a1b87394db79e9be260c1e6a5

m-stein, 2021-05-06:
@nfeske Thanks for the fix! I can confirm that ef6c377 fixes not only the cut arrows but also the stretched radio buttons and other effects of font size changes on my system.

nfeske, 2021-05-06:
Thank you for the feedback. It is a little strange that it fixes anything besides the back button though. The patch touches the back button only.

m-stein, 2021-05-06:
I checked it twice. All buttons now appear as desired and don't change not even in size with other font sizes.

commit ef6c3779 (2021-05-05): sculpt: fix glitch in fs-browser back arrow

Fixes #4124
commit ad847d05 (2021-05-10): sculpt: fix glitch in fs-browser back arrow

Fixes #4124