genodelabs/genode #4127: base: increase bss alignment to 8 bytes

cproc, 2021-04-30:
When executing a run script (for example log.run) on hw/rpi3 (qemu) built with gcc 10 and binutils 2.36, no log output appears.

The reason is that in the generated bootstrap binary `_bss_start` has an 8-byte aligned address and `_bss_end` does not and therefore the zero-out loop in bootstrap which works with 8-byte increments does not detect the end of the area correctly. Increasing the alignment of `_bss_start` and `_bss_end` from 4 to 8 bytes solves the problem.

cproc, 2021-04-30:
Fixed in dd0acb1

nfeske, 2021-04-30:
Merged to staging.

Thank you for documenting your findings so well. I'm fascinated with following your stream of issues. ;-)

commit 0725ac82 (2021-04-30): base: increase bss alignment to 8 bytes

Fixes #4127
closed 2021-05-05 by nfeske