genodelabs/genode #4128: ada/spark build errors with gcc 10

cproc, 2021-04-30:
When running depot_autopilot with gcc 10, the following errors occur:

```
  Program test/spark/test-spark
    COMPILE  machinery.o
    COMPILE  main.o
    COMPILE  print.o
    COMPILE  startup.o
error: "s-stalib.adb" and "main.adb" compiled with different GNAT versions
error: "system.ads" and "main.adb" compiled with different GNAT versions
error: "ada.ads" and "main.adb" compiled with different GNAT versions
error: "s-secsta.adb" and "main.adb" compiled with different GNAT versions
error: "componolit.ads" and "main.adb" compiled with different GNAT versions
error: "componolit-runtime.ads" and "main.adb" compiled with different GNAT versions
error: "componolit-runtime-platform.adb" and "main.adb" compiled with different GNAT versions
error: "componolit-runtime-debug.adb" and "main.adb" compiled with different GNAT versions
error: "componolit-runtime-exceptions.ads" and "main.adb" compiled with different GNAT versions
error: "s-unstyp.ads" and "main.adb" compiled with different GNAT versions
error: "s-parame.adb" and "main.adb" compiled with different GNAT versions
error: "s-stoele.adb" and "main.adb" compiled with different GNAT versions
```

The errors don't occur when running `make test/spark` instead of `make run/depot_autopilot`. I found a version of the 's-stalib.adb' file in the `ada-runtime` port, which might need to be updated, but I'm not really familiar with this topic.

nfeske, 2021-05-05:
@m-stein Will you take care of this issue?

m-stein, 2021-05-05:
Of course.

m-stein, 2021-05-05:
This 20a67b725ee55c87a9e38bbbfb261b85c7814430 should fix it.

nfeske, 2021-05-06:
I temporarily reverted the fix in https://github.com/genodelabs/genode/commit/a7fcf2693b46b2df2e50a2546faacd9032029b5b because it has a hard dependency on gcc-10.

m-stein, 2021-05-06:
@nfeske Sorry, I forgot to mention that this shouldn't be merged before gcc 10.

commit 35e44eb8 (2021-05-05): ports/ada-runtime: update to support gcc 10

Fixes #4128
commit f8b2fd52 (2021-05-28): ports/ada-runtime: update to support gcc 10

Fixes #4128
closed 2021-05-28 by chelmuth