genodelabs/genode #4129: depot_autopilot: store signal context capability as copy

cproc, 2021-04-30:
When running `depot_autopilot` built with gcc 10 on base-linux, the following error occurs:

```
Genode sculpt-21.03-147-ga21582f3a6 <local changes>
17592186044415 MiB RAM and 8997 caps assigned to init
[init -> depot_autopilot] 
[init -> depot_autopilot] --- Run "test-block" (max 20 sec) ---
[init -> depot_autopilot] 
[init -> depot_autopilot] 0.049 [init] Warning: test-block-client: configured RAM exceeds available RAM, proceed with 21916097
[init -> depot_autopilot] 0.068 [init -> test-block-client] block device with block size 512 sector count 1024 (testing 1024 sectors)
[init -> depot_autopilot] 0.071 [init -> test-block-client] reading block 0 - 1023, 246 per request
[init -> depot_autopilot] 0.118 [init -> test-block-client] reading block 0 - 1023, 1 per request
[init -> depot_autopilot] 0.212 [init -> test-block-client] reading block 0 - 1023, 1 per request
[init -> depot_autopilot] 0.304 [init -> test-block-client] read/write/compare block 0 - 1023, 8 per request
[init -> depot_autopilot] 0.794 [init -> test-block-client] --- all tests finished ---
[init -> depot_autopilot] 
[init -> depot_autopilot]  test-block                      ok         0.794  log
Warning: invalid signal-context capability
Kernel: Segmentation fault (signum=11), see Linux kernel log for details

ld.lib.so[21238]: segfault at 401efff8 ip 0000000050089ef1 sp 00000000401f0000 error 6 in ld.lib.so[50000000+c8000]
```

The reason is that a signal context capability which was obtained by the `Signal_handler` implicit conversion operator [1] is stored as reference, but the operator returns a copy on the stack which became invalid. Changing the operator to return a reference to the capability did not solve the problem for some reason, but storing a copy of the capability instead of a reference did.

[1] https://github.com/genodelabs/genode/blob/efbed6f7bfd2879b28ba209e8bbd2a7ae40ba0b4/repos/base/include/base/signal.h#L492

cproc, 2021-04-30:
Fixed in 147f18d

cproc, 2021-04-30:
Actually, changing the operator to return a reference might work (but then the compiler complained about the 'const' keyword). I'll check again.

chelmuth, 2021-04-30:
We follow the common practice to store capabilities as copies (not references) because the capability object itself is one reference to a capability index. Therefore, I'd vote to keep your initial fix.

cproc, 2021-04-30:
Ok. For completeness, the operator change originally did not work because I got the signature wrong. It worked when written as `operator Capability<Signal_context> const& () const { return _cap; }`.

chelmuth, 2021-04-30:
Merged c9626b830f75968ca8037bc244810641b008aeca to staging.

commit c9626b83 (2021-04-30): depot_autopilot: store signal context capability as copy

Fixes #4129
closed 2021-05-05 by nfeske