genodelabs/genode #4132: qt5: build errors with gcc 10

cproc, 2021-05-03:
When building qtwebengine with gcc 10, the following error occurs:

```
In file included from ../../../../../../../../contrib/qt5-35581cb853857df728ca88342347d633f9b3d304/src/lib/qt5/qtwebengine/src/3rdparty/chromium/base/trace_event/trace_event_memory_overhead.cc:5:
../../../../../../../../contrib/qt5-35581cb853857df728ca88342347d633f9b3d304/src/lib/qt5/qtwebengine/src/3rdparty/chromium/base/trace_event/trace_event_memory_overhead.h:60:29: error: ‘string’ in namespace ‘std’ does not name a type
   60 |   void AddString(const std::string& str);
      |                             ^~~~~~
../../../../../../../../contrib/qt5-35581cb853857df728ca88342347d633f9b3d304/src/lib/qt5/qtwebengine/src/3rdparty/chromium/base/trace_event/trace_event_memory_overhead.h:15:1: note: ‘std::string’ is defined in header ‘<string>’; did you forget to ‘#include <string>’?
   14 | #include "base/macros.h"
  +++ |+#include <string>
   15 | 
```
and a few more about similar missing includes.

cproc, 2021-05-03:
Fixed in 4512e1c

chelmuth, 2021-05-03:
Merged to staging.

commit c59dbe05 (2021-05-03): qt5: fix build errors with gcc 10

Fixes #4132
closed 2021-05-05 by nfeske