genodelabs/genode #4133: Report MAC address from NIC drivers

chelmuth, 2021-05-03:
Since the transformation of NIC drivers to Uplink session clients, the MAC address is not known to the system until the link state changes to UP and the NIC router reports the link information. As the router is not the right component to distribute this information anyway and the MAC address may be required in other components (e.g., a system GUI) also if the link is down, drivers should optionally report this information if configured to do so.

The configuration should follow our report-config pattern
```
<report mac_address="yes"/>
```
and the report should look consistent among all drivers.
```
<nic mac_address="00:00:00:00:00:00"/>
```

m-stein, 2021-05-04:
@chelmuth Is there already a plan how the report should look with multiple interfaces at the same driver?

chelmuth, 2021-05-05:
Good point in cases we can't avoid to drive two devices from one component. What do you think about the following proposal?

```xml
<devices>
  <nic mac_address="02:00:00:00:00:01"/>
  <nic mac_address="02:00:00:00:00:02"/>
</devices>
```

m-stein, 2021-05-05:
In the example of the FEC NIC driver, devices must have unique names to be distinguishable. Shouldn't we reflect these names at least in case of multiple devices?
```xml
<devices>
  <nic name="tom"   mac_address="02:00:00:00:00:01"/>
  <nic name="jerry" mac_address="02:00:00:00:00:02"/>
</devices>
```

m-stein, 2021-05-05:
As an alternative, I would also throw in the following style that is used e.g. by the NIC router (report is labelled "state"):
```xml
<state>
  <device name="tom"   mac_address="02:00:00:00:00:01"/>
  <device name="jerry" mac_address="02:00:00:00:00:02"/>
</state>
```

chelmuth, 2021-05-05:
After offline discussion we settled on the following scheme, which I'm going to implement for linux, ipxe, and wifi first.

```xml
<devices>
  <nic label="tom"   mac_address="02:00:00:00:00:01"/>
  <nic label="jerry" mac_address="02:00:00:00:00:02"/>
</devices>
```

The `label="..."` attribute is optional and can be used my multi-headed drivers to distinguish multiple network devices. The label will then also be used when creating the Uplink session.

closed 2023-08-24 by nfeske