genodelabs/genode #4134: problems in gdb_monitor_interactive/target_conifg run scripts

tomga, 2021-05-03:


tomga, 2021-05-03:
Commit 3ece2cd8a3dbd025f3510e7fae89087425ffe82c fixes some issues in current versions of `gdb_monitor*` run scripts (those that are not enabled for autopilot):

 * increased RAM and caps to similar values like in `gdb_monitor.run`
 * added `posix.lib.so` to `boot_modules` in `gdb_monitor_interactive.run` like in `gdb_monitor.run`
 * changed `-nographic` to `-display none` in `gdb_monitor_interactive.run` like in `gdb_monitor.run`
 * changed quoting in command starting terminal with `bash` that starts `gdb` - without it commands to `gdb` from parameters were not passed correctly

With those changes `gdb_monitor_interactive.run` and `genode_monitor_target_config.run` seem to work as expected on `x86_64` with `nova` kernel.

nfeske, 2021-05-04:
Thanks @tomga. The fixes are very much appreciated.

commit 386df6ae (2021-05-04): Fixes in gdb_monitor_interactive/target_config.run

Fixes #4134
closed 2021-05-05 by nfeske