genodelabs/genode #4135: vfs: fonts_fs does not handle config changes if <rom> changes

jschlatow, 2021-05-03:
First of all some background: When I set up my wifi at home, I added emojis to the SSID to test whether/how the clients deal with that. Currently, emojis are not shown in Sculpt because the Bitstream Vera font does not contain the corresponding glyphs. As there are a few fonts with simple (uncoloured) emoji glyphes, I had a quick try and got stuck in the following issue.

I added a new ttf file to the ttf-bitstream-vera-minimal package, created a new sculpt image and tried to change the fonts config. Yet, when adding a `<rom name="new.ttf"/>`, I got the following error:

```
[leitzentrale -> fonts_fs] Error: VFS config update failed (node type 'dir' != fs type 'rom')
```

Apparently, the vfs does not handle config changes of rom nodes.

The error can also be reproduced as follows:

1. start sculpt
2. copy `/config/managed/fonts` to `/config/fonts`
3. (optional) change all `<ttf>` nodes to reference `VeraMono.ttf`
4. remove the `<rom name="Vera.ttf"/>` node and save

nfeske, 2021-05-04:
The VFS does not support dynamic reconfiguration of its structure. E.g., replacing one ROM by another is not possible.

Sooner or later, we will address this limitation. But since the topic has a lot of ripple effects, it needs careful planning. So rather later than sooner.

What could work to solve your problem is placing the font files in a regular <fs> following a naming convention such as regular.ttf, mono.ttf. The <ttf> file system would refer to those file names and should be able to respond dynamically to changes of the font file. So you could overwrite the regular.ttf file by another one to trigger the font change dynamically.

Alternatively, you may opt to publish a separate fonts_fs pkg that readily includes your font. So as a user, one would deploy your fonts_fs over the package provided by genodelabs. From a user's perspective, this would probably most convenient.

jschlatow, 2021-05-04:
> The VFS does not support dynamic reconfiguration of its structure. E.g., replacing one ROM by another is not possible.
> 
> Sooner or later, we will address this limitation. But since the topic has a lot of ripple effects, it needs careful planning. So rather later than sooner.

Understood. From a pure user's perspective though, it appears that dynamic reconfiguration is supported because one can change the font size and path of the `<ttf>` nodes. Do you think there is a way to catch the unsupported changes and put out a clear error message like "unsupported config update"?

> What could work to solve your problem is placing the font files in a regular following a naming convention such as regular.ttf, mono.ttf. The file system would refer to those file names and should be able to respond dynamically to changes of the font file. So you could overwrite the regular.ttf file by another one to trigger the font change dynamically.
> 
> Alternatively, you may opt to publish a separate fonts_fs pkg that readily includes your font. So as a user, one would deploy your fonts_fs over the package provided by genodelabs. From a user's perspective, this would probably most convenient.

I also thought about building a separate fonts_fs pkg. Yet, if I understand it correctly, this would not change the system fonts used by the Leitzentrale. Or is there some routing magic that I overlooked?

nfeske, 2021-05-04:
> Do you think there is a way to catch the unsupported changes and put out a clear error message like "unsupported config update"?

I see where the wrong expectation comes from. The font size change is actually the only feature of the VFS that is able to respond to config changes. So it is really a special case for now.

> Yet, if I understand it correctly, this would not change the system fonts used by the Leitzentrale. Or is there some routing magic that I overlooked?

Yes. However, in a future version of Sculpt, those font files could actually be hosted in the config fs. Wouldn't that be cool? A user could just drop the preferred font in the config directory of the sculpt partition to customize it.

jschlatow, 2021-05-05:
@nfeske I just tried putting the fonts in the config fs. My approach is to modify the leitzentrale.config in a live system. More precisely, I added the following route to the fonts_fs child:

```
<service name="ROM" label="Vera.ttf">
   <parent label="config -> fonts/regular.ttf"/> </service>
```

The route appears to work since the child is requesting addititonal ram resource once I copy a emoji capable font to `/config/fonts/regular.ttf`. Unforunately, once I add the route any interaction with a view component (e.g. network_view, settings_view) will trigger a page fault in that component. This is also reproducable by using the original Vera.ttf as regular.ttf.

My line of thought was that by adding the above route, users could optionally host their font files in `/config/fonts`. Yet, if the corresponding files are not present, the default built-in fonts would be used.

nfeske, 2021-05-05:
> More precisely, I added the following route to the fonts_fs child

That route looks good to me. But I think it must be there initially and cannot be added later.

When changing a route that is currently in use (like the ROM route for "Vera.ttf"), the client component is restarted. In this case, this would be the fonts_fs. All file-system sessions to the fonts_fs that exist at this point are doomed. However, the fs clients (like the menu-view instances) cannot know that. They fault as soon as they access the packet-stream buffer of the file-system session. We have to be careful to prevent the fonts_fs from being restarted, e.g., by featuring the required routes in the initial leitzentrale.config, always referring the `regular.ttf` (and `monospace.ttf`) instead of `Vera.ttf`.

> My line of thought was that by adding the above route, users could optionally host their font files in /config/fonts. Yet, if the corresponding files are not present, the default built-in fonts would be used.

That was also my intention.

Have you added the default fonts to the initial content of the config fs specified in the `<import>` node (in sculpt.run)?

jschlatow, 2021-05-06:
@nfeske, please have a look at my branch [4135_user_defined_fonts](https://github.com/jschlatow/genode/tree/4135_user_defined_fonts), which contains a commit series for placing fonts in `/config/fontfiles`. I tested this with the Bitstream Vera font with Symbola merged in. The result is a ~3M ttf file, therefore some RAM quotas needed to be increased.

I found out that fonts were not updated if the file changed (fixed in d39f230). Moreover, to allow a conventient deployment of the fonts_fs package, I changed the fonts config to load the files from a File_system session.

jschlatow, 2021-05-06:
Unfortunately, there is still an issue that just came up when replacing the ttf file in a running Sculpt. I tested this with 
6bd46f1 and a fonts_fs pkg accessing `/config/fontfiles` via a chroot component. All clients updated their fonts smoothly, yet by changing the system fonts config to pull the fonts via a File_system session instead of ROM sessions, I broke this. Now, when I try to replace the ttf file in inspect view or system shell, the `cp` command does not finish (or only appears to as the failing font update might hinder the view update. Maybe just a ram or caps quota issue but the log is pretty silent.

jschlatow, 2021-05-06:
Another obervation is that the new `fontfiles` directory appears in the files view (but is not browsable). I am wondering why this is happening as the keyboard directory is not shown.

nfeske, 2021-05-07:
It is fantastic to see you navigating through Sculpt, diving into the guts, and removing stumbling blocks along the way! :-)

Even though I'm currently focused on Linux/DDE-related work, I couldn't resist peeking at your branch.

- The patches of vfs_ttf and os/vfs.h are very welcome. They nicely fill gaps that I was admittedly not fully aware of. The patches look excellent.

- Regarding the naming of the place where to keep the font files, let us better name the directory `style`. This can then become the designated place for visual tweaks of the leitzentrale, not just the font but also the menu-view style. The latter has already the form of a file structure (see https://github.com/genodelabs/genode/tree/master/repos/gems/src/app/menu_view/styles) but currently, this content is integrated as a mere tar archive (menu_view_styles.tar). Menu view does not yet support the dynamic reconfiguration of the style but this would be a natural addition. ...ultimately allowing us to re-skin the leitzentrale at runtime.

- I would keep the default fonts_fs package as is. If more features (like picking up font files from a file-system session) are needed, they should be provided as alternative packages. My reasoning is that we should keep the on-boarding steps as simple as possible. The manual session-routing labor described in the getting-started section of the documentation (https://genode.org/documentation/articles/sculpt-21-03#Getting_a_first_impression) is already a long-winded and mouse-click-heavy experience until one can see the glory of the first system-shell window. Right now, the fonts_fs is covered with only a few words:

   "The font used by the terminal is obtained from a font server. To create the font server, install and add the package genodelabs -> GUI -> fonts fs, and assign the system font configuration to its ROM (config)."

  If we follow the route suggested by your "consistently pull.." commit, we would need to cover the concept behind it in more detail. I presume that users discovering Sculpt for the first time could probably not care less about font-related technicalities. The font handling is arguably cool (I'm quite proud of it) but it does not deserve the spotlight. In contrary, I I sense that the additional bureaucratic step and cognitive load would taint the initial user experience.

- My suggestion to wrap both fonts in a a file-system session was not well thought trough. It was more an impulse to side-step the limitation of the VFS. It is probably better to stay with the two <rom> nodes but just name them differently. If the names of the <rom> nodes stay the same, we can still update the ROM content without the need to reconfigure the VFS. Regarding the naming of those ROMs, my suggestion was a bit sloppy as well. The names should better be consistent with the terminology found in the fonts.config. Here, "regular" stands for the font face (the same across monospace and text variants). The roles of the two ttf files are actually a proportional one (used for title, text, and annotation) and a monospaced one. So I would call the files `default.ttf` and monospace.ttf`. By leaving the <rom> nodes intact, we also keep the fonts_fs package simple. When I made the <fs> suggestion, I did not consider that the fonts.config it used there too, thereby imposing the need for a file-system session.

  Sorry for luring you down this path.

jschlatow, 2021-05-07:
Thanks for the feedback. I updated the branch following your suggestions.

The fonts_fs pkg now comes with its own fonts but can be used with a tailored launcher to use the system fonts. I also changed the `runtime_view` config to use the default.ttf. The only remaining component with a hard-coded font is `log_terminal` which I'd rather keep independent and, with the fixed tiny font size, changes would hardly be recognisable. 

>   Sorry for luring you down this path.

It was an exicting ride on the learning curve ;-)

jschlatow, 2021-05-07:
And, for future reference: When pulling the fonts from config_fs via a File_system session, I was able to update the monospace.ttf smoothly but not the default.ttf. In the latter case, the inspect_view and other views got stuck.

nfeske, 2021-05-17:
I just merged the two collateral commits about the ttf watching and the read loop.

Commit c6bc9113ae465f96cdb14537269285c33e3abd07 looks good to me. I'll merge it as soon as I get the full picture including the subsequent commits.

I'm a little bit afraid that dropping the configuration from the fonts_fs pkg archive in 77614e817457b1a9bf56b67188ec31d014d35d17 may break the usage of the fonts_fs pkg outside of Sculpt. Have you double-checked that none of the other users of pkg/fonts_fs relies on it? AFAIK, the pkg/terminal depends on it, which in turn is used by the Goa tutorials.

jschlatow, 2021-05-17:
> I'm a little bit afraid that dropping the configuration from the fonts_fs pkg archive in [77614e8](https://github.com/genodelabs/genode/commit/77614e817457b1a9bf56b67188ec31d014d35d17) may break the usage of the fonts_fs pkg outside of Sculpt. Have you double-checked that none of the other users of pkg/fonts_fs relies on it? AFAIK, the pkg/terminal depends on it, which in turn is used by the Goa tutorials.

Right. I haven't had Goa on my map when introducing this change and only grepped through the genode repo. The fonts_fs.config must be kept, I'll prepare a fixup commit. WIth the renaming of the ttf files, however, I'm afraid the Goa tutorials must be adapted nonetheless.

jschlatow, 2021-05-17:
Now I remember my reasoning: The fonts.config was not included as rom in the <content> section of the fonts_fs runtime. (Also notice the inconsistent naming: fonts.config vs fonts_fs_config.). I thus assumed that the fonts_fs.config has been ineffective for a while anyway.

nfeske, 2021-05-17:
That's very much understandable. The dual purpose of the fonts_fs pkg in the context of Sculpt (where its config is obtained via a session route) and outside of Sculpt (where the config is built-in to make the pkg easy it integrate) is not really obvious. Maybe it would have been cleaner to have two separate packages to begin with, e.g., using `pkg/fonts_fs` for Sculpt and a package named `pkg/vera_fonts_fs` for the stand-alone use?

Adjusting the Goa tutorial is not a problem in any case. I keep Goa in sync with the Sculpt releases, so I will just need to remember it until the next Sculpt release.

jschlatow, 2021-06-17:
@nfeske, I double-checked the usage of the fonts_fs pkg. The fonts_fs.config is actually used in text_area.run and scrcpy.run. All other 'users' I could find re-use the run/sculpt/fonts.config or create their own (e.g. menu_view.run). I added a fixup commit c85b975 that undeletes the fonts_fs.config from the raw pkg. I'd also suggest to adapt the 'config' attribute in fonts_fs' runtime for consistency and add fonts_fs.config to its rom content. To my understanding, the latter is a harmless change as Sculpt will create a route for the required 'config' rom which dominates over the route that depot_deploy generates as a consequence of the provided `config="fonts_fs.config`. With this modification, however, the fonts_fs pkg is also deployable with the built-in config if this route is omitted.

Regarding the Goa tutorial, there is no need for adaptation as it doesn't use the fonts_fs pkg but only the independent terminal pkg.

nfeske, 2023-08-24:
This issue is inactive since a long time. @jschlatow I wonder how we should go about it?

closed 2026-04-30 by jschlatow