genodelabs/genode #4136: base-linux: support for 64-bit ARM

nfeske, 2021-05-04:
Currently, base-linux supports x86_32 and x86_64. I'd like to extend the support to AARCH64 to ease the Genode development on 64-bit ARM hosts such as the MNT-Reform laptop.

nfeske, 2021-05-04:
The two commits linked above contain preparatory steps.

Commit https://github.com/genodelabs/genode/commit/edf905c45b361628f5ee9191d95a80d8004b6122 sidesteps the need to pass a TLS pointer through the clone system-call binding.

Commit https://github.com/genodelabs/genode/commit/d17d20bb34784d31dc9a04468cecfb712c28217e harmonizes the system calls used across x86 and AARCH64.

nfeske, 2021-05-05:
Commit https://github.com/genodelabs/genode/commit/dc68e3391997468427cac5c77c407c76798141ad introduces a way to build (parts of) core in an architecture-specific way.

nfeske, 2021-05-06:
Commit https://github.com/genodelabs/genode/commit/5b3b41b4a5f88cc7a7f4ce1f7c5c58b58d145920 adds the actual AARCH64 support. @skalk, I hope you are fine with the crt0 changes.

Commit https://github.com/genodelabs/genode/commit/12dee2c5fe7567a902fc4ba2a732f993368cc80b is needed for non-trivial scenarios that depend on the timeout framework (i.e., all scenarios that use the timer driver). I put this change into a separate commit because it is somewhat quirky. I'd love to eventually replace it with a way to access the `pmccntr_el0` register directly on top of Linux, but I haven't found a clean way yet. However, for practical means, short-circuiting the `Trace::timestamp` function works surprisingly well. One can even successfully execute scenarios as complex as leitzentrale.run.

trimpim, 2021-05-06:
@nfeske  while enabling `base-linux` for our marvell armhf_v7a target we also hit this problem in two places, trace/timestamp and lib/jitterentropy.

Our first approach was to return a constant value, which is in case of `jitterentropy` can't be a good solution. Therefore we opted to use a `get_time`  (CLOCK_MONOTONIC) system call which has a certain overhead of course.

closed 2021-05-10 by nfeske