genodelabs/genode #4137: user-defined dynamic remapping rules

jschlatow, 2021-05-04:
Some remapping rules in Sculpt hide certain keys - such as F12 and LEFTMETA - from applications. Particularly when interacting with virtual machines, it is sometimes necessary to use these keys in their original meaning nonetheless.

Since the remapping of these keys is performed by the event_filter, which supports inclusion of dynamically updated roms, these remapping rules can be toggled via a rarely-used key such as scrolllock by adding a global_keys_handler and rom_filter in a separate pkg.

nfeske, 2021-05-05:
That's very elegant! I love how you apply this customization via a complementary package, with no modifications of Sculpt required.

I'm not 100% sure about adding the package the Genode repository. As long as you are the sole user, it may be better hosted at Genode-world or as a Goa project independent from the mainline Genode repository, at least initially. Once we see its proliferated use, we can then decide to add it to the official Genode repository (and possibly add it to pkg/sculpt-installation). What do you think?

jschlatow, 2021-05-05:
Sure. I haven't used Goa yet, but this would probably be a good opportunity to give it a try ;-). I will combine this with a Genodians article for better visibility and documentation.

jschlatow, 2021-05-14:
@nfeske I just had a try with Goa but I'm afraid it's unable to demonstrate its strengths for user_keys. I'd therefore opt for genode-worl instead. Why? There are two main aspects in this regard: First, goa is intended for application development and packaging. In particular, `goa run` is a quite useful feature for interactive testing. However, user_keys seemlessly integrates with sculpt, hence `goa run` doesn't provide any useful preview. Second, user_keys requires `rom_to_file` which is not available in any depot. I must therefore build and publish `rom_to_file` in my own depot first. Since this already requires me to set up my genode repository for publishing, I could also publish `user_keys` right away from genode-world.

nfeske, 2021-05-17:
I suggested Goa primarily for the streamlined publish workflow, which is less bureaucratic than the recipes. But this was just a suggestion. I'm perfectly fine with hosting your package in genode-world.

Shall we close this issue here?

closed 2021-05-17 by jschlatow