genodelabs/genode #4138: gdb_monitor: number of preserved caps is fixed and can be too small

tomga, 2021-05-05:
I'm experimenting with using `gdb_monitor` and if it is used to debug more complex component than currently tested it can exhaust fixed `100` number of caps that it preserves for itself when starting child component. It took me quite some time to understand what is going on and I think this problem deserves some care.

Currently fixed value is hardcoded [here](/genodelabs/genode/blob/516a9a69253b650e256972c8433226dd5718db65/repos/ports/src/app/gdb_monitor/gdbserver/genode-low.cc#L511) (and changing it to a bigger value fixed a problem for me).

Few lines earlier there's code that handles RAM preservation configuration of the form:
```
<preserve name="RAM" quantum="32M"/>
```

I could implement similar configuration for `CAPS` but I'd like to have some doubts resolved:

1. Was the fixed limit expected to be enough and it just happened that currently it is too small (at least in a way I use `gdb_monitor`) or maybe it was good enough that nobody felt a need do add a configuration option for it?
2. If the configuration for it is desired would the form below be ok?
```
<preserve name="CAPS" quantum="150"/>
```
3. Should it become an obligatory parameter like it is for RAM?

tomga, 2021-05-06:
I just realized this topic was raised on mailing list recently in topic 'how to debug gdb?' and @cproc mentioned exactly the same line there. If read it carefully earlier I'd probably saved some time on debugging. Anyway questions stay valid.

nfeske, 2023-06-16:
This issue won't be fixed because we'll dedicate further efforts on the new debug monitor (#4917).

closed 2023-06-16 by nfeske