genodelabs/genode #4139: gdb: build errors with gcc 10

cproc, 2021-05-06:
When building `noux-pkg/gdb_arm` with gcc 10, the following errors occur:

```
  Program noux-pkg/gdb_arm/gdb_arm
      [gdb_arm]  /.../x86_64-pc-elf/bin/ld: libsim.a(maverick.o):/.../contrib/gdb-8eddf0d249a22993e57b3242112fb62f503101c9/src/noux-pkg/gdb/sim/arm/maverick.c:63: multiple definition of `DSPregs'; libsim.a(wrapper.o):/.../contrib/gdb-8eddf0d249a22993e57b3242112fb62f503101c9/src/noux-pkg/gdb/sim/arm/wrapper.c:132: first defined here
      [gdb_arm]  /.../x86_64-pc-elf/bin/ld: libsim.a(maverick.o):/.../contrib/gdb-8eddf0d249a22993e57b3242112fb62f503101c9/src/noux-pkg/gdb/sim/arm/maverick.c:65: multiple definition of `DSPsc'; libsim.a(wrapper.o):/.../contrib/gdb-8eddf0d249a22993e57b3242112fb62f503101c9/src/noux-pkg/gdb/sim/arm/wrapper.c:134: first defined here
      [gdb_arm]  /.../x86_64-pc-elf/bin/ld: libsim.a(maverick.o):/.../contrib/gdb-8eddf0d249a22993e57b3242112fb62f503101c9/src/noux-pkg/gdb/sim/arm/maverick.c:64: multiple definition of `DSPacc'; libsim.a(wrapper.o):/.../contrib/gdb-8eddf0d249a22993e57b3242112fb62f503101c9/src/noux-pkg/gdb/sim/arm/wrapper.c:133: first defined here
      [gdb_arm]  /.../x86_64-pc-elf/bin/ld: libsim.a(armemu32.o):/.../contrib/gdb-8eddf0d249a22993e57b3242112fb62f503101c9/src/noux-pkg/gdb/sim/arm/armemu.c:1145: multiple definition of `isize'; libsim.a(armemu26.o):/.../contrib/gdb-8eddf0d249a22993e57b3242112fb62f503101c9/src/noux-pkg/gdb/sim/arm/armemu.c:1145: first defined here
```

Another '-fno-common' issue.

cproc, 2021-05-06:
Fixed in 0bed7f6

nfeske, 2021-05-06:
The patch touches code that seem to be unrelated to Genode, about some specific DSP.

Wouldn't it be sensible to somehow exclude the support for those devices from your build of GDB? (I'm imagining a minor patch of a makefile)

cproc, 2021-05-06:
@nfeske: commit 45fa3d6 appears to fix it even without patching

commit 9a4f5fb1 (2021-05-06): gdb: fix build errors with gcc 10

Fixes #4139
commit ff160dec (2021-05-10): gdb: fix build errors with gcc 10

Fixes #4139
closed 2021-05-10 by chelmuth