genodelabs/genode #4140: cpu_load_display: build errors with gcc 10

cproc, 2021-05-06:
When building `app/cpu_load_display` with gcc 10, the following errors occur:

```
  Program app/cpu_load_display/cpu_load_display
    COMPILE  main.o
In file included from /.../repos/gems/src/app/cpu_load_display/main.cc:21:
/.../repos/gems/include/nano3d/scene.h: In instantiation of ‘void Nano3d::Scene< <template-parameter-1-1> >::Surface::_clear(Genode::Surface<T>&) [with T = Genode::Pixel_rgba<unsigned int, Genode::Surface_base::RGB888, 16711680, 16, 65280, 8, 255, 0, 0, 0>; PT = Genode::Pixel_rgba<unsigned int, Genode::Surface_base::RGB888, 16711680, 16, 65280, 8, 255, 0, 0, 0>]’:
/.../repos/gems/include/nano3d/scene.h:175:11:   required from ‘void Nano3d::Scene< <template-parameter-1-1> >::Surface::clear() [with PT = Genode::Pixel_rgba<unsigned int, Genode::Surface_base::RGB888, 16711680, 16, 65280, 8, 255, 0, 0, 0>]’
/.../repos/gems/include/nano3d/scene.h:235:24:   required from ‘void Nano3d::Scene< <template-parameter-1-1> >::_handle_period() [with PT = Genode::Pixel_rgba<unsigned int, Genode::Surface_base::RGB888, 16711680, 16, 65280, 8, 255, 0, 0, 0>]’
/.../repos/gems/include/nano3d/scene.h:246:22:   required from ‘Nano3d::Scene< <template-parameter-1-1> >::Scene(Genode::Env&, Genode::uint64_t, Gui::Point, Gui::Area) [with PT = Genode::Pixel_rgba<unsigned int, Genode::Surface_base::RGB888, 16711680, 16, 65280, 8, 255, 0, 0, 0>; Genode::uint64_t = long long unsigned int; Gui::Point = Genode::Point<>; Gui::Area = Genode::Area<>]’
/.../repos/gems/src/app/cpu_load_display/main.cc:326:75:   required from ‘Cpu_load_display::Scene< <template-parameter-1-1> >::Scene(Genode::Env&, Genode::uint64_t, Gui::Point, Gui::Area) [with PT = Genode::Pixel_rgba<unsigned int, Genode::Surface_base::RGB888, 16711680, 16, 65280, 8, 255, 0, 0, 0>; Genode::uint64_t = long long unsigned int; Gui::Point = Genode::Point<>; Gui::Area = Genode::Area<>]’
/.../repos/gems/src/app/cpu_load_display/main.cc:474:46:   required from here
/.../repos/gems/include/nano3d/scene.h:169:16: error: converting a packed ‘Genode::Pixel_rgba<unsigned int, Genode::Surface_base::RGB888, 16711680, 16, 65280, 8, 255, 0, 0, 0>’ pointer (alignment 1) to a ‘long int’ pointer (alignment 8) may result in an unaligned pointer value [-Werror=address-of-packed-member]
  169 |     for (long *dst = (long *)surface.addr(); n--; dst++)
      |                ^~~
In file included from /.../repos/os/include/os/pixel_rgb888.h:17,
                 from /.../repos/gems/src/app/cpu_load_display/main.cc:19:
/.../repos/os/include/os/pixel_rgba.h:38:15: note: defined here
   38 | class Genode::Pixel_rgba
      |               ^~~~~~~~~~
In file included from /.../repos/gems/src/app/cpu_load_display/main.cc:21:
/.../repos/gems/include/nano3d/scene.h: In instantiation of ‘void Nano3d::Scene< <template-parameter-1-1> >::Surface::_clear(Genode::Surface<T>&) [with T = Genode::Pixel_rgba<unsigned char, Genode::Surface_base::ALPHA8, 0, 0, 0, 0, 0, 0, 255, 0>; PT = Genode::Pixel_rgba<unsigned int, Genode::Surface_base::RGB888, 16711680, 16, 65280, 8, 255, 0, 0, 0>]’:
/.../repos/gems/include/nano3d/scene.h:176:11:   required from ‘void Nano3d::Scene< <template-parameter-1-1> >::Surface::clear() [with PT = Genode::Pixel_rgba<unsigned int, Genode::Surface_base::RGB888, 16711680, 16, 65280, 8, 255, 0, 0, 0>]’
/.../repos/gems/include/nano3d/scene.h:235:24:   required from ‘void Nano3d::Scene< <template-parameter-1-1> >::_handle_period() [with PT = Genode::Pixel_rgba<unsigned int, Genode::Surface_base::RGB888, 16711680, 16, 65280, 8, 255, 0, 0, 0>]’
/.../repos/gems/include/nano3d/scene.h:246:22:   required from ‘Nano3d::Scene< <template-parameter-1-1> >::Scene(Genode::Env&, Genode::uint64_t, Gui::Point, Gui::Area) [with PT = Genode::Pixel_rgba<unsigned int, Genode::Surface_base::RGB888, 16711680, 16, 65280, 8, 255, 0, 0, 0>; Genode::uint64_t = long long unsigned int; Gui::Point = Genode::Point<>; Gui::Area = Genode::Area<>]’
/.../repos/gems/src/app/cpu_load_display/main.cc:326:75:   required from ‘Cpu_load_display::Scene< <template-parameter-1-1> >::Scene(Genode::Env&, Genode::uint64_t, Gui::Point, Gui::Area) [with PT = Genode::Pixel_rgba<unsigned int, Genode::Surface_base::RGB888, 16711680, 16, 65280, 8, 255, 0, 0, 0>; Genode::uint64_t = long long unsigned int; Gui::Point = Genode::Point<>; Gui::Area = Genode::Area<>]’
/.../repos/gems/src/app/cpu_load_display/main.cc:474:46:   required from here
/.../repos/gems/include/nano3d/scene.h:169:16: error: converting a packed ‘Genode::Pixel_rgba<unsigned char, Genode::Surface_base::ALPHA8, 0, 0, 0, 0, 0, 0, 255, 0>’ pointer (alignment 1) to a ‘long int’ pointer (alignment 8) may result in an unaligned pointer value [-Werror=address-of-packed-member]
  169 |     for (long *dst = (long *)surface.addr(); n--; dst++)
      |                ^~~
In file included from /.../repos/os/include/os/pixel_rgb888.h:17,
                 from /.../repos/gems/src/app/cpu_load_display/main.cc:19:
/.../repos/os/include/os/pixel_rgba.h:38:15: note: defined here
   38 | class Genode::Pixel_rgba
      |               ^~~~~~~~~~
```

cproc, 2021-05-06:
The same errors occur for `app/nano3d`

nfeske, 2021-05-06:
@cproc Could you give https://github.com/genodelabs/genode/commit/62aaa3321046a5cf279fc4f2b4bebad98b5e4be8 a try for both programs?

I did some quick measurement of the effect using `GENODE_LOG_TSC` on my x250. The removal of the optimization slows down the code by factor 2. Even though I don't welcome this effect, it is arguably tolerable in this concrete case (the surface clearing takes only 1-2% CPU time after all). On the other hand, the difference shows the potential of the low-hanging fruit of an optimized bzero function.

However, I was very surprised to find out that using O3 instead of O2 slows down the code again by factor 2. I would not have expected the performance getting worse by using O3.

cproc, 2021-05-06:
@nfeske: yes, that fixed the errors

commit 62aaa332 (2021-05-06): nano3d: remove surface-clear optimization

This patch fixes a GCC-10 compile error. Even though the optimization
was quite effective - I measured a speedup of factor 2 - it is not all
that important for the overall application performance. In the nano3d
case, we are talking about 1 vs. 2 percent of CPU time.

Fixes #4140
commit d19b7516 (2021-05-10): nano3d: remove surface-clear optimization

This patch fixes a GCC-10 compile error. Even though the optimization
was quite effective - I measured a speedup of factor 2 - it is not all
that important for the overall application performance. In the nano3d
case, we are talking about 1 vs. 2 percent of CPU time.

Fixes #4140
closed 2021-05-10 by chelmuth