genodelabs/genode #4141: gpt_write: build errors with gcc 10

cproc, 2021-05-06:
When building `app/gpt_write` with gcc 10, the following errors occur:

```
  Program app/gpt_write/gpt_write
    COMPILE  main.o
In file included from /.../repos/gems/src/app/gpt_write/main.cc:21:
/.../repos/gems/src/app/gpt_write/gpt.h: In member function ‘bool Gpt::Entry::read_name(char*, Genode::size_t) const’:
/.../repos/gems/src/app/gpt_write/gpt.h:129:49: error: taking address of packed member of ‘Gpt::Entry’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
  129 |    return !!Util::extract_ascii(dest, dest_len, name, NAME_LEN);
      |                                                 ^~~~
/.../repos/gems/src/app/gpt_write/gpt.h: In member function ‘bool Gpt::Entry::write_name(const Label&)’:
/.../repos/gems/src/app/gpt_write/gpt.h:137:33: error: taking address of packed member of ‘Gpt::Entry’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
  137 |    return !!Util::convert_ascii(name, NAME_LEN,
      |                                 ^~~~
```

cnuke, 2021-05-06:
I will look into that.

cnuke, 2021-05-06:
Commit eb404bd fixes the error.

cnuke, 2021-05-06:
Commit bc03a32 fixes the error by turning the member into a byte array and adjusting the convert functions.

nfeske, 2021-05-06:
Thanks @cnuke for the rework. I merged the commit to staging.

commit 37a3ef1e (2021-05-06): gpt_write: address unaligned access for name

Turn member into byte array to prevent unaligned memory access.

Fixes #4141.
commit c5c3d7ca (2021-05-10): gpt_write: address unaligned access for name

Turn member into byte array to prevent unaligned memory access.

Fixes #4141.
closed 2021-05-10 by chelmuth