genodelabs/genode #4142: zynq_nic_drv: build errors with gcc 10

cproc, 2021-05-06:
When building `drivers/nic/spec/zynq` with gcc 10, the following errors occur:

```
  Program drivers/nic/spec/zynq/zynq_nic_drv
    COMPILE  main.o
In file included from /.../repos/os/src/drivers/nic/spec/zynq/main.cc:25:
/.../repos/os/src/drivers/nic/spec/zynq/cadence_gem.h: In member function ‘Nic::Mac_address Genode::Cadence_gem_base::read_mac_address()’:
/.../repos/os/src/drivers/nic/spec/zynq/cadence_gem.h:440:68: error: taking address of packed member of ‘Net::Mac_address’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
  440 |     uint32_t* const low_addr_pointer = reinterpret_cast<uint32_t*>(&mac.addr[0]);
      |                                                                    ^~~~~~~~~~~~
/.../repos/os/src/drivers/nic/spec/zynq/cadence_gem.h:441:69: error: taking address of packed member of ‘Net::Mac_address’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
  441 |     uint16_t* const high_addr_pointer = reinterpret_cast<uint16_t*>(&mac.addr[4]);
      |                                                                     ^~~~~~~~~~~~
/.../repos/os/src/drivers/nic/spec/zynq/cadence_gem.h: In member function ‘void Genode::Cadence_gem_base::write_mac_address(const Mac_address&)’:
/.../repos/os/src/drivers/nic/spec/zynq/cadence_gem.h:679:80: error: taking address of packed member of ‘Net::Mac_address’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
  679 |     const uint32_t* const low_addr_pointer = reinterpret_cast<const uint32_t*>(&mac.addr[0]);
      |                                                                                ^~~~~~~~~~~~
/.../repos/os/src/drivers/nic/spec/zynq/cadence_gem.h:680:81: error: taking address of packed member of ‘Net::Mac_address’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
  680 |     const uint16_t* const high_addr_pointer = reinterpret_cast<const uint16_t*>(&mac.addr[4]);
      |                                                                                 ^~~~~~~~~~~~
```

m-stein, 2021-05-06:
I can have a look into it.

m-stein, 2021-05-07:
@cproc Unfortunately, I'm not making any progress with the toolchain. I had to restart ``tool/toolchain arm`` twice, first because of no space left on device and second because some weird stuff happend with the re-used build dir. During the third run, with clean build dir, it got really far but stopped with:
```
/home/lypo/genodelabs/toolchain_21_05_build/bootstrap/install/lib/gcc/x86_64-pc-linux-gnu/10.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: libsim.a(maverick.o):/home/lypo/genodelabs/genode/contrib/gdb-8eddf0d249a22993e57b3242112fb62f503101c9/src/noux-pkg/gdb/sim/arm/maverick.c:63: multiple definition of `DSPregs'; libsim.a(wrapper.o):/home/lypo/genodelabs/genode/contrib/gdb-8eddf0d249a22993e57b3242112fb62f503101c9/src/noux-pkg/gdb/sim/arm/wrapper.c:132: first defined here
/home/lypo/genodelabs/toolchain_21_05_build/bootstrap/install/lib/gcc/x86_64-pc-linux-gnu/10.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: libsim.a(maverick.o):/home/lypo/genodelabs/genode/contrib/gdb-8eddf0d249a22993e57b3242112fb62f503101c9/src/noux-pkg/gdb/sim/arm/maverick.c:65: multiple definition of `DSPsc'; libsim.a(wrapper.o):/home/lypo/genodelabs/genode/contrib/gdb-8eddf0d249a22993e57b3242112fb62f503101c9/src/noux-pkg/gdb/sim/arm/wrapper.c:134: first defined here
/home/lypo/genodelabs/toolchain_21_05_build/bootstrap/install/lib/gcc/x86_64-pc-linux-gnu/10.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: libsim.a(maverick.o):/home/lypo/genodelabs/genode/contrib/gdb-8eddf0d249a22993e57b3242112fb62f503101c9/src/noux-pkg/gdb/sim/arm/maverick.c:64: multiple definition of `DSPacc'; libsim.a(wrapper.o):/home/lypo/genodelabs/genode/contrib/gdb-8eddf0d249a22993e57b3242112fb62f503101c9/src/noux-pkg/gdb/sim/arm/wrapper.c:133: first defined here
/home/lypo/genodelabs/toolchain_21_05_build/bootstrap/install/lib/gcc/x86_64-pc-linux-gnu/10.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: libsim.a(armemu32.o):/home/lypo/genodelabs/genode/contrib/gdb-8eddf0d249a22993e57b3242112fb62f503101c9/src/noux-pkg/gdb/sim/arm/armemu.c:1145: multiple definition of `isize'; libsim.a(armemu26.o):/home/lypo/genodelabs/genode/contrib/gdb-8eddf0d249a22993e57b3242112fb62f503101c9/src/noux-pkg/gdb/sim/arm/armemu.c:1145: first defined here
```
I'm starting to wonder whether it's worth it to spend another 3 hours with a busy machine. AFAIKT, the problem with the Zynq driver should be fixed by e955f4b026. Could you give it a try with your toolchain?

nfeske, 2021-05-07:
This build problem was discussed at #4139 and the fix should be at staging.

m-stein, 2021-05-07:
But I should cherry pick it, right? The new toolchain isn't on staging yet, or is it?

nfeske, 2021-05-07:
We will merge the actual tool chain on Monday to attain the pre-tool-chain CI results over the weekend. With this issue settled, we should be ready to go. Thank you for the fix!

I think you should be fine to cherry-pick the GDB build fix, but I don't urge you. I just found it worth hinting at the connection between the error message you posted and the other issue.

cproc, 2021-05-07:
@m-stein: with your commit a part of the errors disappeared. These two remained:

```
  Program drivers/nic/spec/zynq/zynq_nic_drv
    COMPILE  main.o
In file included from /.../repos/os/src/drivers/nic/spec/zynq/main.cc:25:
/.../repos/os/src/drivers/nic/spec/zynq/cadence_gem.h: In member function ‘void Genode::Cadence_gem_base::write_mac_address(const Mac_address&)’:
/.../repos/os/src/drivers/nic/spec/zynq/cadence_gem.h:691:80: error: taking address of packed member of ‘Net::Mac_address’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
  691 |     const uint32_t* const low_addr_pointer = reinterpret_cast<const uint32_t*>(&mac.addr[0]);
      |                                                                                ^~~~~~~~~~~~
/.../repos/os/src/drivers/nic/spec/zynq/cadence_gem.h:692:81: error: taking address of packed member of ‘Net::Mac_address’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
  692 |     const uint16_t* const high_addr_pointer = reinterpret_cast<const uint16_t*>(&mac.addr[4]);
      |                                                                                 ^~~~~~~~~~~~
```

chelmuth, 2021-05-07:
@cproc @m-stein have a patch for the remaining error almost ready

m-stein, 2021-05-07:
@chelmuth I'm already done with it.

m-stein, 2021-05-07:
d8e6a2556e fixup "nic/zynq: fix packed conversion error with gcc 10"

m-stein, 2021-05-07:
Sorry, had to adapt it, so, that it compiles:
732164a560 fixup "nic/zynq: fix packed conversion error with gcc 10"

I'd prefer this one as it documents the use of the structs.

chelmuth, 2021-05-07:
Anyway... please check if 41b8dedf54d60c64accc2afa670ce1d5a2b4736f reflects what you wanted to achieve.

m-stein, 2021-05-07:
I finished compiling the arm toolchain and can confirm that the commits that are now on staging fix the issue :)

commit e53a8a2a (2021-05-07): nic/zynq: fix packed conversion error with gcc 10

Fixes #4142
commit d2bf5655 (2021-05-10): nic/zynq: fix packed conversion error with gcc 10

Fixes #4142
closed 2021-05-10 by chelmuth