genodelabs/genode #4143: gdb_monitor problems

cproc, 2021-05-10:
As reported by @tomga on the mailing list [1], there are currently problems when using gdb_monitor to debug components which provide a service. While @tomga investigated a potential race related to thread creation and service announcement [2], I tried to find out if there was a state in the past when debugging the ssh_terminal with gdb_monitor worked, and with the help of git bisect found commit a89d61a to be the last version where connecting with an ssh client und using the echo test was possible in principle (except when some other sporadic errors occured, but it did not just hang without error message as with later versions). I created a debugging branch based on this commit for reference [3].

The next commit 36b6ebc ("libc: process VFS operations in monitor") caused an error message about "pthread_self() called from alien thread", which was resolved by cherry-picking commit 556a5c8 ("libc: clean up 'clock_gettime' handling"). With these two commits added, gdb_monitor gets stuck in `Cpu_thread_client::state()` addressed to the gdb_monitor-local Cpu_thread_component. The relation between the libc VFS change and this symptom is not clear yet, it could be possible that the change just triggered an existing bug in gdb_monitor. I'll investigate further. At least these findings confirm that the routing information in the test run script is correct and the bug is somewhere else.

[1] https://www.mail-archive.com/users@lists.genode.org/msg01602.html
[2] https://www.mail-archive.com/users@lists.genode.org/msg01605.html
[3] https://github.com/cproc/genode/commits/gdb_ssh_terminal-20210509

tomga, 2021-05-10:
> At least these findings confirm that the routing information in the test run script is correct and the bug is somewhere else.

Thank you for finding time and confirming that so quickly.

cproc, 2021-05-12:
So far I noticed two kinds of problems:

- the gdb_monitor-local session components call libc code (directly or indirectly via original gdb functions) and sometimes `pthread_self()` is called and fails because it is called from an alien thread (target_ep). Also, the session components don't currently use the `Libc::with_libc()` wrapper
- possibly related to the libc VFS change, sometimes when a `Cpu_thread_component` is created and tries to create a pipe, it looks like the monitor executes a different job (select() timeout) before the `pipe()` call returns and if this other job tries to use a `Cpu_thread client` addressed to a local `Cpu_thread component`, a deadlock seems to occur, possibly because the entrypoint is already busy creating the `Cpu_thread_component`

nfeske, 2023-06-16:
I'm closing this issue because further efforts with supporting GDB will be focused on the new debug monitor (#4917).

closed 2023-06-16 by nfeske