genodelabs/genode #4144: depot: fix locale influence on hash generation

m-stein, 2021-05-11:
I experienced the creation of many new hashes directly on tag sculpt-21.03b when creating package sculpt. Thanks to the guidance of @chelmuth, we found that this was related to my locale setting ``LC_ALL=en_US.UTF-8`` in ``.bashrc`` (I'm on a spanish system but want my bash to work entirely in English).

m-stein, 2021-05-11:
Fixed by 4527f6f80b.

m-stein, 2021-05-12:
Although in online discussions I read the contrary, I could verify that in Make ``$(sort ...)`` always sorts ``LC_ALL=C``-style independent of the environment settings. However, ``$(shell echo ... | sort)`` always sorts according to the environment settings.

m-stein, 2021-05-12:
@chelmuth I added a commit 028def622e to be squashed that should cover all tools.

chelmuth, 2021-05-12:
For completeness: GNU Make uses `strcmp()` (which is locale-independent) not `strcoll()` (which takes into account `LC_ALL`, `LC_COLLATE`, and `LANG` in this order).

https://github.com/mirror/make/blob/1ae90c030d5ffc79f74e3938880445a6b8b2412e/src/misc.c#L37-L50

BTW, this is also true for the ViM-internal sort function.

commit 3a680cdc (2021-05-18): squash "depot: fix locale influence on hash generation"

Please rewrite commit message:

tool: use local LC_ALL=C instead of LC_COLLATE=C

We used to export 'LC_COLLATE=C' for the environment of Genode tools. This was
meant to ensure that sorting is always done C-style and not dependent on the
users locale settings. This is required, for instance, to ensure that the same
archive always yields the same hash.

However, 'export LC_COLLATE=C' is not sufficient. It can be outruled by an
'LC_ALL' setting in the users environment. The manual of 'sort' recommends to
set 'LC_ALL=C' locally if you want  [... cut]
commit 33202495 (2021-05-18): squash "depot: fix locale influence on hash generation"

Please rewrite commit message:

tool: use local LC_ALL=C instead of LC_COLLATE=C

We used to export 'LC_COLLATE=C' for the environment of Genode tools. This was
meant to ensure that sorting is always done C-style and not dependent on the
users locale settings. This is required, for instance, to ensure that the same
archive always yields the same hash.

However, 'export LC_COLLATE=C' is not sufficient. It can be outruled by an
'LC_ALL' setting in the users environment. The manual of 'sort' recommends to
set 'LC_ALL=C' locally if you want  [... cut]
commit 139a2cfa (2021-05-28): tool: use local LC_ALL=C instead of LC_COLLATE=C

We used to export 'LC_COLLATE=C' for the environment of Genode tools. This was
meant to ensure that sorting is always done C-style and not dependent on the
users locale settings. This is required, for instance, to ensure that the same
archive always yields the same hash.

However, 'export LC_COLLATE=C' is not sufficient. It can be outruled by an
'LC_ALL' setting in the users environment. The manual of 'sort' recommends to
set 'LC_ALL=C' locally if you want reliable results and this is what this
commit does. Furthermore it removes the former 'ex [... cut]
closed 2021-05-28 by chelmuth