genodelabs/genode #4149: usb_host: Make USB robust against session destruction & lower CPU usage for ISOCH tranfsers

ssumpf, 2021-05-12:
1. Currently if the USB session is destructed during a synchronous operation (e.g., control message), the `Lx::Task` worker is destroyed and accessed by the Linux code when the operation is complete. Therefore, we want to bind the lifetime of the `Lx::Task` to the lifetime of the USB device and not the session. This way a finished device request has only to check if the session still exists and a session request if the device still exists.
2. Because of a quirk for Intel XHCI controllers in Linux, isochronous transfers cause an interrupt for every descriptor. This leads to a situation where under heavy isochronous load up to 8000 IRQs/s are generated causing high CPU load (~15%). We want to get this values closer to the one without the quirk (~4%).

ssumpf, 2021-05-12:
dcd8fea address the first point, while 9f57f27 addresses the second point. @chelmuth: Pleases consider for staging.

ssumpf, 2021-05-12:
@chelmuth: 86f2f73 remove the unnecessary `_udev` check.

chelmuth, 2021-05-12:
Thanks for the cleanup commit. Merged to staging.

nfeske, 2021-05-12:
Just watching from the sidelines, if the `_udev` pointer cannot become a `nullptr`, couldn't it be a reference to make this clear?

ssumpf, 2021-05-14:
6a83a8f makes `_udev` a reference.

chelmuth, 2021-05-14:
Thanks, merged to staging.

nfeske, 2021-05-14:
Maybe related, the usb_hid_raw test produces the page fault on staging (on all x86 platforms), here for nova_x86_64:

```
[2021-05-14 06:01:26] [init -> usb_drv] dev_info: USB disconnect, device number 2[0m[0m

[2021-05-14 06:01:26] no RM attachment (READ pf_addr=0x4 pf_ip=0x102986f from pager_object: pd='init -> usb_drv' thread='ep') [0m

[2021-05-14 06:01:26] page fault, pd='init -> usb_drv' thread='ep' cpu=0 ip=0x102986f address=0x4 stack pointer=0x401feaa0 qualifiers=0x4 irUwp reason=1[0m
```

chelmuth, 2021-05-14:
> Maybe related, the usb_hid_raw test produces the page fault on staging (on all x86 platforms), here for nova_x86_64:

This originates from the following change.

https://github.com/genodelabs/genode/commit/60521c30bc282541223cc6e9583ffdcc6c0b16cd#diff-6ca7278a5335344d569f690523edf9774837bbb56df7ba67c5816e012d0489e3L940-L950

It seems `_udev` is valid, but `_udev.actconfig` is not. @ssumpf may `ready()` only return `true` if `actconfig` is set?

ssumpf, 2021-05-14:
> > Maybe related, the usb_hid_raw test produces the page fault on staging (on all x86 platforms), here for nova_x86_64:
> 
> This originates from the following change.
> 
> [60521c3#diff-6ca7278a5335344d569f690523edf9774837bbb56df7ba67c5816e012d0489e3L940-L950](https://github.com/genodelabs/genode/commit/60521c30bc282541223cc6e9583ffdcc6c0b16cd#diff-6ca7278a5335344d569f690523edf9774837bbb56df7ba67c5816e012d0489e3L940-L950)
> 
> It seems `_udev` is valid, but `_udev.actconfig` is not. @ssumpf may `ready()` only return `true` if `actconfig` is set?

@chelmuth: Thanks for your findings. I am with you regarding `ready` and have implemented it in c8a0cbe. Could you try this out on mogli, because I cannot trigger the issues in Qemu so far.

chelmuth, 2021-05-14:
Commit  591e3e6 fixes the issue, thanks.

chelmuth, 2021-05-19:
Since I merged this commit series, autopilot reports sporadic failures like follows.

```
autopilot.checker.2021-05-13-00-49/x86_64/x86_64.pc.foc.usb_hid_raw.log:383:[2021-05-13 06:04:17] [init -> usb_drv] Warning: schedule() called without runnable tasks
autopilot.checker.2021-05-13-00-49/x86_64/x86_64.pc.nova.vbox5_genode_usb_hid_raw.log:414:[2021-05-13 06:12:09] [init -> usb_drv] Warning: schedule() called without runnable tasks
autopilot.checker.2021-05-13-00-49/x86_32/x86_32.pc.pistachio.usb_hid_raw.log:286:[2021-05-13 07:15:42] [init -> usb_drv] Warning: schedule() called without runnable tasks
autopilot.checker.2021-05-14-00-47/x86_64/x86_64.pc.sel4.usb_hid_raw.log:314:[2021-05-14 06:02:45] [init -> usb_drv] Warning: schedule() called without runnable tasks
autopilot.checker.2021-05-14-00-47/x86_64/x86_64.pc.nova.vbox5_genode_usb_hid_raw.log:414:[2021-05-14 06:06:27] [init -> usb_drv] Warning: schedule() called without runnable tasks
autopilot.checker.2021-05-15-00-46/x86_64/x86_64.pc.sel4.usb_hid_raw.log:314:[2021-05-15 06:29:12] [init -> usb_drv] Warning: schedule() called without runnable tasks
autopilot.checker.2021-05-15-00-46/x86_64/x86_64.pc.nova.vbox5_genode_usb_hid_raw.log:414:[2021-05-15 06:33:00] [init -> usb_drv] Warning: schedule() called without runnable tasks
autopilot.checker.2021-05-15-00-46/x86_64/x86_64.pc.nova.vbox5_genode_usb_hid_raw.log:436:[2021-05-15 06:33:00] [init -> log_terminal] [init -> usb_drv] Warning: schedule() called without runnable tasks
autopilot.checker.2021-05-15-00-46/x86_32/x86_32.pc.sel4.usb_hid_raw.log:281:[2021-05-15 07:22:18] [init -> usb_drv] Warning: schedule() called without runnable tasks
autopilot.checker.2021-05-16-03-34/x86_64/x86_64.pc.nova.vbox5_genode_usb_hid_raw.log:414:[2021-05-16 09:19:08] [init -> usb_drv] Warning: schedule() called without runnable tasks
autopilot.checker.2021-05-16-03-34/x86_64/x86_64.pc.nova.vbox5_genode_usb_hid_raw.log:436:[2021-05-16 09:19:09] [init -> log_terminal] [init -> usb_drv] Warning: schedule() called without runnable tasks
autopilot.checker.2021-05-17-00-47/x86_64/x86_64.pc.nova.vbox5_genode_usb_hid_raw.log:414:[2021-05-17 06:17:17] [init -> usb_drv] Warning: schedule() called without runnable tasks
autopilot.checker.2021-05-17-00-47/x86_64/x86_64.pc.nova.vbox5_genode_usb_hid_raw.log:436:[2021-05-17 06:17:17] [init -> log_terminal] [init -> usb_drv] Warning: schedule() called without runnable tasks
autopilot.checker.2021-05-18-00-46/x86_64/x86_64.pc.nova.vbox5_genode_usb_hid_raw.log:414:[2021-05-18 06:07:37] [init -> usb_drv] Warning: schedule() called without runnable tasks
autopilot.checker.2021-05-18-00-46/x86_64/x86_64.pc.nova.vbox5_genode_usb_hid_raw.log:436:[2021-05-18 06:07:37] [init -> log_terminal] [init -> usb_drv] Warning: schedule() called without runnable tasks
autopilot.checker.2021-05-19-00-47/x86_32/x86_32.pc.sel4.usb_hid_raw.log:281:[2021-05-19 07:13:22] [init -> usb_drv] Warning: schedule() called without runnable tasks
```

ssumpf, 2021-05-19:
Ok, who would have thought this draws so great circles. Will look into it.

ssumpf, 2021-05-20:
dba7187 fixes various issues during USB device construction/destruction.

chelmuth, 2021-05-27:
@ssumpf last night there were failing usb_hid_raw tests with the following log messages.

```
[2021-05-27 06:24:17] [init -> usb_hid_drv] Error: complete: got NO_ERROR code in error path
[2021-05-27 06:24:17] [init -> usb_hid_drv] Error: complete: got NO_ERROR code in error path
```

I suspect the following code location causes the diagnostic message and wonder if we should add `p.error = Packet_descriptor::NO_DEVICE_ERROR` before the packet acknowledgement.

https://github.com/genodelabs/genode/blob/291a453f0fc14eaee80c7a703346b00c537ca79d/repos/dde_linux/src/drivers/usb_host/raw.cc#L876-L881

ssumpf, 2021-05-27:
@chelmuth: Yes this was on purpose, we can add the `NO_DEVICE_ERROR` if you like.

chelmuth, 2021-05-27:
> @chelmuth: Yes this was on purpose, we can add the `NO_DEVICE_ERROR` if you like.

Now I wonder what the purpose was...

ssumpf, 2021-05-27:
@chelmuth: 32d3598 sets `NO_DEVICE_ERROR` upon packet drainage. Tested with Qemu plug/unplug loop.

closed 2022-02-17 by skalk