genodelabs/genode #4150: sculpt: themed_decorator dies with Limit_exceeded exception

alex-ab, 2021-05-12:
From time to time the themed_decorator dies with the following message:

```
[runtime -> themed_decorator] Error: Uncaught exception of type 'Genode::Quota_guard<Genode::Cap_quota>::Limit_exceeded'
[runtime -> themed_decorator] Warning: abort called - thread: ep
```

Is a patch, similar to the [decorator](https://github.com/genodelabs/genode/blob/master/repos/gems/src/app/decorator/main.cc#L183) handling, the way to go also for the themed_decorator ?

nfeske, 2021-05-12:
Since the themed decorator maintains multiple GUI sessions per window, I first suspected that the problem may be triggered at https://github.com/genodelabs/genode/blob/master/repos/gems/include/gems/gui_buffer.h#L55. If that is the case, it would be best to add the session-upgrade handling there. Could you give this a try?

Should my suspicion be wrong and we keep your fix, could you please remove the `Genode::` prefixed from your patch? The `Genode` namespace is already imported.

alex-ab, 2021-05-12:
Thanks, I will test drive the commit some working days to see whether it was the cause.

alex-ab, 2021-05-21:
The exception around gems/gui_buffer.h L55 (commit 893ff30f6905b164b97be38993c3d208008daa59) did not trigger in the last days. The only output and upgrade I saw steamed from 893ff30f6905b164b97be38993c3d208008daa59.

nfeske, 2021-05-25:
I merged your fix to staging now.

alex-ab, 2021-05-28:
Entered master.

closed 2021-05-28 by alex-ab