genodelabs/genode #4151: base-hw/run/cpu_scheduler: fails

m-stein, 2021-05-12:
The test of the base-hw CPU scheduler breaks on current master. Btw., I'd opt for testing this nightly.

skalk, 2022-02-17:
@m-stein when looking at the test, I doubt anybody except you can address this issue. I'm overstrained when trying the test's semantic. I have no objections in testing it nightly, as soon as it works in a robust fashion.

m-stein, 2022-02-19:
Commit 2ecf1d887b44c8d5874cf97cd1dbe78886030a73 seems to be the cause of this issue.

m-stein, 2022-02-19:
Commit https://github.com/genodelabs/genode/commit/2ecf1d887b44c8d5874cf97cd1dbe78886030a73 raises some questions in my head (and I definitely should have checked on this way back when the commit was merged) . As it seems that the scheduler test would have to be widely adapted in order to match the new semantics I would like to clear about these questions before going on.

The semantics of ``ready_check`` were changed: The method result previously was true only if setting the given share "ready" makes re-evaluating the scheduling choice necessary (say, if the head pointer becomes outdated by the method call itself). Now, however, it is true if an update _might_ be necessary (because, for instance, method ``unready`` always sets ``_need_to_schedule`` and then ``ready_check`` would always be true as well). Even more, ``ready_check`` is _always_ true now and most of its body is never reached because ``ready(s1)`` causes the following condition to be always true.

skalk, 2022-02-21:
@m-stein : you're right. This is pretty strange. The original commit did not checked for `_need_to_schedule` in the beginning of `ready_check`, see https://github.com/skalk/genode/commit/7b7c9bc0b8743ff5daa79b6ecfbac6e8cac9ad02 but during the resolution of some staging/nightly-build issues it slipped in. I can't explain with good reasons why this line 154 exists. The semantic was: instead of always checking the time, only check it when certain event have occurred. Those events are marking `_need_to_schedule`, and only if it is true the "deep checking" of time, claim and fills gets checked.

m-stein, 2022-09-02:
@skalk I sense that it would be a good idea to resolve the uncertainty you describe in the near future because, IMO, it has the potential of creating subtle misbehavior in the user land that is hard to relate top-downward. Admittedly, I'm not that eager to dive into the topic currently. Would you be willing to give it a try?

skalk, 2022-09-07:
@m-stein I've re-evaluated the changes from the time when "scheduling on demand" was introduced. In fact it renders `ready_check` useless, because the semantic got simplified in the sense that: only when a job becomes active or unactive the scheduler has to update itself (looking at the clock, update remaining time etc.) - all of that is part of the `update` function.
As a further optimization, one can transfer the nowaday dead path of `ready_check` into `ready` as a condition to decide whether `need_to_schedule` has to be set or not. But by removing `ready_check` and replacing the one `ready_check` call by `ready`, we're already functional correct and still perform better than previously when the time update was performed at each kernel entry. Anyway, I've prepared a commit that removes `ready_to_check`, and implements a fine-grained form of detection whether the scheduler has to update itself or not.

Moreover, the parameter semantic of `update` has changed at that time from relative time (elapsed) to an absolute time value. That is the main reason for the cpu scheduler test's fail. I've prepared a solution that somehow modernizes the test a bit, but does not change the expected results. With that changes the test runs a while, but at some places (9) it detects the need for an scheduler update differently that the test expects it. Here I'm lost, because of the overall complexity of the unit test. However, I've prepared an additional commit, which updates the expected unit test values according to the nowadays behaviour.

@m-stein can you please have a deep look at the commits, and especially evaluate the changes of the unit tests metrics?

ssumpf, 2022-09-20:
@skalk, @m-stein: With [eb23b4e](https://github.com/skalk/genode/commit/eb23b4eb7fc67cc3fc9bc94feb1eb24a599fdd13) applied, I see the CPU utilization drop to about 5% when running import on genodians:

without commit:
```
[init -> genodians -> import -> generate -> vfs] generate html/jklmnn/index ...
[init -> top] cpu=0.0   0     0  31.24%   0.00% thread='entrypoint'              label='core'
[init -> top] cpu=0.0   0     0  20.77%   0.00% thread='pager_ep'                label='core'
[init -> top] cpu=0.0   1     0  13.54%   0.00% thread='fork_ep'                 label='init -> genodians -> import -> generate -> /bin/make'
[init -> top] cpu=0.0   1     0  10.30%   0.00% thread='ep'                      label='init -> genodians -> import -> generate -> vfs'
[init -> top] cpu=0.0   1     0   5.55%   0.00% thread='ep'                      label='init -> genodians -> import -> generate -> /bin/make'
[init -> top] cpu=0.0   1     0   3.59%   0.00% thread='ep'                      label='init -> genodians -> import -> generate'
[init -> top] cpu=1.0   0     0 100.00%   0.00% thread='idle'                    label='kernel'
[init -> top] cpu=2.0   0     0 100.00%   0.00% thread='idle'                    label='kernel'
[init -> top] cpu=3.0   0     0 100.00%   0.00% thread='idle'                    label='kernel'
```

with commit (eb23b4e):
```
[init -> genodians -> import -> generate -> vfs] generate html/jklmnn/index ...
[init -> top] cpu=0.0   0     0  95.75%   0.00% thread='idle'                    label='kernel'
[init -> top] cpu=0.0   0     0   1.93%   0.00% thread='entrypoint'              label='core'
[init -> top] cpu=0.0   0     0   0.81%   0.00% thread='pager_ep'                label='core'
[init -> top] cpu=0.0   1     0   0.27%   0.00% thread='ep'                      label='init -> genodians'
[init -> top] cpu=0.0   1     0   0.25%   0.00% thread='fork_ep'                 label='init -> genodians -> import -> generate -> /bin/make'
[init -> top] cpu=0.0   1     0   0.20%   0.00% thread='ep'                      label='init -> genodians -> import -> generate -> /bin/make'
[init -> top] cpu=1.0   0     0 100.00%   0.00% thread='idle'                    label='kernel'
[init -> top] cpu=2.0   0     0 100.00%   0.00% thread='idle'                    label='kernel'
[init -> top] cpu=3.0   0     0 100.00%   0.00% thread='idle'                    label='kernel'
```

m-stein, 2022-09-20:
@ssumpf Thanks for sharing these impressive results :-)

skalk, 2022-09-21:
@ssumpf  thank you for testing my solution with the more complex genodians scenario, and for the investigation! I think I found the root cause for not scheduling correctly when idle is active, which led to the strange top results. After applying a fix, I see normal results with top when executing the genodians run-script. If you like, you might give it a try. @m-stein the commit to review has changed therefore. It is now rebased to staging an has id e03e84a

m-stein, 2022-09-22:
@skalk I'll review it tomorrow if it's fine.

m-stein, 2022-09-26:
@skalk Thanks for your rework of the on-demand scheduling! I'll have a look into the remaining problems with the test soon.

skalk, 2022-11-15:
@m-stein are you still into it, or what shall we do with this issue?

m-stein, 2022-12-01:
I started renaming and restructuring the scheduler without changing semantics in order to get a better grasp of what is going on. In my mind, a list of things came up that I'd like to implement over time and I'd just like to write it down here, so, I don't forget it and, so, that others have a chance to make suggestions:

* Either the test or the semantic of the scheduler must be fixed, so that the test succeeds again.
* The test should be part of our nightly tests.
* The style (especially of the test) should be adapted to be readable and conform to Genode style
  * No precompiler macros
  * No one letter names
  * No excessively long lines
  * ...
* Wording should be adapted to use common langue
  * CPU share -> scheduling context
  * Claim -> quota
  * ...
* The number of priority bands is currently significantly restricted (4) because this parameter's overhead doesn't scale well. I hope to find a solution that allows for the efficient use of a broad range of priorities.
* Currently, there is no check that the total amount of granted quota in a system doesn't exceed the schedulers superperiod. It should be checked somewhere or be made unnecessary through semantic changes.
* Head updates that transition to a new superperiod are currently way more expensive (the scheduler has to modify each scheduling context with quota) than head updates somewhere inside a superperiod. One solution to this would be to have two datastructures current_quota_schedule and next_quota_schedule. A context in the first datastructure whose quota gets depleted would be moved to the latter datastructure and reset directly. Transitioning to the next superperiod would merely consist in swapping the two datastructures.

jschlatow, 2022-12-01:
@m-stein I have a few spontaneous thoughts re. the priority bands: A typical way to deal with priorities that I've seen in real-time schedulers is to store the ready state of all contexts in a single bit vector. The highest-priority context to be scheduled is then determined by the least/most significant bit set in this bit vector (e.g. via a lookup table). I think this approach could be applicable to the base-hw scheduler as well. We might need two vectors though (one for the ready state, one for the quota-available state). The conjunction of both vectors can be used for determining the next context to be scheduled. After the super period, the reset is conducted by restoring the quota-available state vector. The actual quota update could probably be postponed to the point in time when a particular context is scheduled. Does this sound reasonable to you?

skalk, 2022-12-01:
I do not argue against the suggestions both of you made. But I think at the moment the highest priority would be to first analyze whether there are semantic issues within either the test, or scheduler code on the ground of the current implementation. Before changing naming, and trying to optimize things. Respectively, where optimizations in the scheduler algorithm are necessary should be first investigated in more depth. Everything else should be part of a separate issue in my eyes.

m-stein, 2022-12-02:
@jschlatow Thanks for this insight! I had a different approach in mind but am open for your idea. We should discuss it in person once you're in the office again.

m-stein, 2022-12-08:
This commit series should fix the core issue of this thread:

3c3df42615 base-hw scheduler test: add to autopilot
2d1779af65 base-hw scheduler test: fix wrong assumptions
81b6a6fdde base-hw scheduler test: clean up
0e6e958db5 base-hw scheduler: print method
59efa69c3c base-hw scheduler: fix ready method

m-stein, 2022-12-08:
@cnuke This commit could be of help in the audio/pinephone issue. Could you give it a try?
59efa69c3c base-hw scheduler: fix ready method

m-stein, 2022-12-08:
@cnuke This commit would be nice for debugging the scheduler in the audio scenario if we find a sensible value for the new ID member of shares: 
0e6e958db5 base-hw scheduler: print method

cnuke, 2022-12-09:
@m-stein thanks for your commits, I'll give https://github.com/genodelabs/genode/commit/59efa69c3c663944b04acb7e6918e25ad271fd4e a try. Unfortunately I'm still caught up in the integration work and did not manage to futher instrument the scheduler.

nfeske, 2023-01-09:
Merged 53e611dfff2a2f4d91da6d47b67e97e06ad0ea66 to staging now.

m-stein, 2023-01-10:
I created it with the intention to propose it for staging. In combination with the upcoming tracing support for kernel threads, it becomes a powerful tool for understanding scheduling-related effects, for instance, for performance analysis. Furthermore, I tried to keep the implementation well separated from the rest of the code and conformant to common practice. However, I just noticed, that I forgot to replace the ``unsigned _id`` member with ``char const* _label`` in order to make it suitable for inspecting real kernel threads. What is your opinion, would you be willing to merge it with te mentioned fixup?

nfeske, 2023-01-10:
I'd leave the print commit at a debug branch because (1) it introduces a new syntax (JSON) that is not used anywhere else throughout Genode, and (2) the code is not called by the regular compiled kernel (strictly speaking, that would make it dead code).

If you want to ease future development/debugging of the scheduler, I would recommend the following:

1. Find a way to supplement the printing feature local to the unit test (leaving it outside the kernel).
2. Investigating a textual format that is optimized for human consumption, e.g., printing the information in a clearly arranged table. You may take the revived trace logger for inspiration (https://github.com/genodelabs/genode/commit/3394f97f861bacfd252ef4c2cbd127bdbdc68878).

m-stein, 2023-01-10:
I've updated my branch accordingly.

m-stein, 2023-01-10:
@nfeske I've updated my branch to not contain any json'ness.

nfeske, 2023-01-11:
@m-stein are you planning to update your merge_to_staging branch? Right now, the branch seems to contain consecutive commits 2cbfaffa40c5f92f7bf18cb14a5e97776102374d and 3cd2db91945f6cfa9b8535685838a787d415200d (and potentially bc50d0e0f1ac4b92fbaeb2af3ed2799b4f05fb01) that might better be squashed into f53883bf1232dad860522b350d42f515f3c096e6. This would ease the later review of the commit history (e.g., by me when aggregating the material for the release notes).

m-stein, 2023-01-11:
@nfeske Updated my branch. Is that what you were aiming for? I'd keep the print method as extra commit as it is a feature while the other commits are about cleaning up and bugs fixes.

nfeske, 2023-01-11:
Thanks for the additional curation. I'm fine with keeping the cleanup and bug fixes separate.

While going through the commits on your merge_to_staging branch, I had a glance at https://github.com/m-stein/genode/commit/4c51e570c4bc71508fb47ad03c18d4041cb8f036. I very much appreciate that you implemented this idea. Before merging the commit though, would you consider a few coding-style tweaks (as annotated in the commit)?

nfeske, 2023-01-11:
Merged all commits except for https://github.com/m-stein/genode/commit/4c51e570c4bc71508fb47ad03c18d4041cb8f036 to staging.

nfeske, 2023-01-12:
> could you please share your thoughts regarding my statement [...]

I acknowledge the theoretical issue. However, for all practical purposes, I cannot foresee a situation where this could bite us. Three thoughts:

1. Once the rampaging component drops back to round-robin mode, it can be interrupted by important activities like device interrupts or other components that still have a budget in the current super period. If starvation of a particular component is a concern, I would assign a quantum/priority to this component.

2. In the concrete case of the USB driver with 10% quantum assigned, it could monopolize an additional 1% (assuming 10 ms time slice, 1 s super period) over other round-robin activities. I don't perceive this as a problem.

3. If my optimism turns out to be wrong, we can change the implementation. I see the current state of the scheduler as still in flux. We need much more real-world stress to validate it.

commit e6a15699 (2023-01-11): base-hw scheduler test: add to autopilot

Fix #4151
Fix #4710
commit 4617d10c (2023-01-12): base-hw scheduler test: add to autopilot

Fix #4151
Fix #4710
closed 2023-01-24 by chelmuth